respeaker/seeed-voicecard #275: May be a typo?

ray-leiter, 2020-12-25:
At the end -- Under title "uninstall seeed-voicecard";
The document has:   "pi@raspberrypi:~/seeed-voicecard $ sudo ./install.sh"
This doesn't look correct. Is something missing?
Is it 2 separate commands?

HinTak, 2020-12-25:
Reading wrong place? https://github.com/respeaker/seeed-voicecard/blob/master/README.md

Says:
```
pi@raspberrypi:~/seeed-voicecard $ sudo ./install.sh
```

ray-leiter, 2020-12-25:
You have me confused!
I am asking you if this line; pi@raspberrypi:~/seeed-voicecard $ sudo ./uninstall.sh 
is an error?

Are you saying it is NOT an error?

OR, are you saying it IS an error?

Regards--  Ray Leiter

From: HinTak
Sent: Friday, December 25, 2020 4:08 PM
To: respeaker/seeed-voicecard
Cc: ray-leiter; Author
Subject: Re: [respeaker/seeed-voicecard] May be a typo? (#275)

Reading wrong place? https://github.com/respeaker/seeed-voicecard/blob/master/README.md
Says:
pi@raspberrypi:~/seeed-voicecard $ sudo ./install.sh

HinTak, 2020-12-25:
My initial reply was copy-pasting your initial input and have since been corrected - anyway, the current front page says uninstall procedure is `sudo ./uninstall.sh` .That is correct. Where did you see "..../install.sh" being mentioned for uninstallation?

HinTak, 2020-12-25:
I don't work for Seeed Studio btw, just another user.

ray-leiter, 2020-12-26:
Nowhere did I see nor did I mention I saw "..../install.sh" (your words).

If you look at the very last part of the page at this URL:
https://github.com/respeaker/seeed-voicecard/blob/master/README.md

you’ll see the confusion; which is this (literally):

“pi@raspberrypi:~/seeed-voicecard $ sudo ./uninstall.sh”

Do you see the “pi@raspberrypi:~/seeed-voicecard “ preceding the dollar sign ($)?

That can’t be correct – can it?

OK, I realize you don’t work for seeed, but I’d like to ask you a question about the ReSpeaker 4-mic board.

Here it is: I am using the demo.py program provided by SNOWBOY to test the ReSpeaker and see if it recognizes my Hotword.
If I use my Logitech USB microphone/headset – it works!!!
BUT, it won’t work with the ReSpeaker.
It has something to do with the default microphone that is configured (I think).
I don’t know how to make sure the ReSpeaker 4-mic array is the default microphone.
Do you?

Regards--  Ray Leiter

From: HinTak
Sent: Friday, December 25, 2020 6:22 PM
To: respeaker/seeed-voicecard
Cc: ray-leiter; Author
Subject: Re: [respeaker/seeed-voicecard] May be a typo? (#275)

My initial reply was copy-pasting your initial input and have since been corrected - anyway, the current front page says uninstall procedure is sudo ./uninstall.sh .That is correct. Where did you see "..../install.sh" being mentioned for uninstallation?

HinTak, 2020-12-26:
Everything before the "$" sign is part of the command prompt (which the machine is showing you - it is usually of the form `account@machine_name:directory_location`. In this case, account is "pi", machine name is "raspberrypi" (both of these are raspbian defaults - every raspberry pi shows these), the directory is just the source code directory! The machine is telling you that you are login as user pi, on a machine called "raspberrypi", in the directory "~/seeed-voicecard", and shows a "$" and you type what follows the "$". Everything before the "$" and including the "$" itself is what the machine shows you, not what you type!

ray-leiter, 2020-12-26:
Thank you very much for the clarification. I should have seen it (I’m not accustomed to seeing such long CLI prompt strings).

What about my question regarding the default microphone?
Do you know how I can configure things so the default is the ReSpeaker 4-mic array rather than the USB mic?

Regards -- Ray

From: HinTak
Sent: Friday, December 25, 2020 11:11 PM
To: respeaker/seeed-voicecard
Cc: ray-leiter; Author
Subject: Re: [respeaker/seeed-voicecard] May be a typo? (#275)

Everything before the "$" sign is part of the command prompt (which the machine is showing you - it is usually of the form account@machine_name:directory_location. In this case, account is "pi", machine name is "raspberrypi" (both of these are raspbian defaults - every raspberry pi shows these), the directory is just the source code directory! The machine is telling you that you are login as user pi, on a machine called "raspberrypi", in the directory "~/seeed-voicecard", and shows a "$" and you type what follows the "$". Everything before the "$" and including the "$" itself is what the machine shows you, not what you type!

HinTak, 2020-12-26:
You probably should ask the snowboy people about your microphone question. That said, it could be that your recording volume is too low - use `amixer` to increase the recording volume.

ray-leiter, 2020-12-26:
OK, Thanks. I’ll try it.

I haven’t yet figured out how to ask the snowboy people questions.

There is probably a forum of some kind—I’ll check it out.

When I first installed all this stuff, it worked without me doing anything with mixer.

Then for some reason, it stopped working so I decide to start from scratch and I was never able to duplicate what I first did.

Funny thing is, it works with my Headset!

Regards--  Ray Leiter

From: HinTak
Sent: Saturday, December 26, 2020 12:13 PM
To: respeaker/seeed-voicecard
Cc: ray-leiter; Author
Subject: Re: [respeaker/seeed-voicecard] May be a typo? (#275)

You probably should ask the snowboy people about your microphone question. That said, it could be that your recording volume is too low - use amixer to increase the recording volume.

AIWintermuteAI, 2021-06-29:
Issue resolved, 
README was updated, snowboy now depreciated. 
Will close.

Hope you enjoyed the pseudo haiku. On more serious note, check Wiki - it will be gradually updated and already includes the instruction how to use Mycroft Precise instead of Snowboy for keyword spotting.
https://wiki.seeedstudio.com/ReSpeaker_2_Mics_Pi_HAT_Raspberry/#keyword-spotting-with-respeaker-2-mic-pi-hat-and-mycroft-precise

closed 2021-06-29 by AIWintermuteAI