respeaker/seeed-voicecard #277: Not working with kernel 5.4.79-v7l+

solidssss, 2020-12-29:
Hi. I cannot get the seed-voicecard 4mic to work with my raspberry 4. It worked great on my old 3b+ but not on the pi 4.
Installed using **sudo ./install.sh 4mic**

**Kernel: 5.4.79-v7l+**

**arecord -l:**  **** List of CAPTURE Hardware Devices ****
card 0: seeed4micvoicec [seeed-4mic-voicecard], device 0: bcm2835-i2s-ac10x-codec0 ac10x-codec.1-003b-0 [bcm2835-i2s-ac10x-codec0 ac10x-codec.1-003b-0]
  Subdevices: 0/1
  Subdevice #0: subdevice #0

**arecord -L:** null
    Discard all samples (playback) or generate zero samples (capture)
jack
    JACK Audio Connection Kit
pulse
    PulseAudio Sound Server
playback
ac108
default
output
sysdefault:CARD=seeed4micvoicec
    seeed-4mic-voicecard, bcm2835-i2s-ac10x-codec0 ac10x-codec.1-003b-0
    Default Audio Device
dmix:CARD=seeed4micvoicec,DEV=0
    seeed-4mic-voicecard, bcm2835-i2s-ac10x-codec0 ac10x-codec.1-003b-0
    Direct sample mixing device
dsnoop:CARD=seeed4micvoicec,DEV=0
    seeed-4mic-voicecard, bcm2835-i2s-ac10x-codec0 ac10x-codec.1-003b-0
    Direct sample snooping device
hw:CARD=seeed4micvoicec,DEV=0
    seeed-4mic-voicecard, bcm2835-i2s-ac10x-codec0 ac10x-codec.1-003b-0
    Direct hardware device without any conversions
plughw:CARD=seeed4micvoicec,DEV=0
    seeed-4mic-voicecard, bcm2835-i2s-ac10x-codec0 ac10x-codec.1-003b-0
    Hardware device with all software conversions
usbstream:CARD=seeed4micvoicec
    seeed-4mic-voicecard
    USB Stream Output

log: 
```
 cat /var/log/seeed-voicecard.log
+ export PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/opt/vc/bin
+ PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/opt/vc/bin
+ OVERLAYS=/boot/overlays
+ '[' -d /boot/firmware/overlays ']'
+ dtparam -d /boot/overlays i2c_arm=on
+ modprobe i2c-dev
+ dtparam -d /boot/overlays spi=on
+ _VER_RUN=
+ CONFIG=/boot/config.txt
+ '[' -f /boot/firmware/usercfg.txt ']'
++ egrep '(1a|UU)'
++ i2cdetect -y 1 0x1a 0x1a
++ awk '{print $2}'
+ is_1a=
++ i2cdetect -y 1 0x35 0x35
++ egrep '(35|UU)'
++ awk '{print $2}'
+ is_35=
++ i2cdetect -y 1 0x3b 0x3b
++ egrep '(3b|UU)'
++ awk '{print $2}'
+ is_3b=UU
+ RPI_HATS='seeed-2mic-voicecard seeed-4mic-voicecard seeed-8mic-voicecard'
+ overlay=
+ '[' x '!=' x ']'
+ '[' xUU '!=' x ']'
+ '[' x == x ']'
+ echo 'install 4mic'
install 4mic
+ overlay=seeed-4mic-voicecard
+ asound_conf=/etc/voicecard/asound_4mic.conf
+ asound_state=/etc/voicecard/ac108_asound.state
+ '[' xUU '!=' x ']'
+ '[' x '!=' x ']'
+ '[' seeed-4mic-voicecard ']'
+ echo Install seeed-4mic-voicecard ...
Install seeed-4mic-voicecard ...
+ rm /etc/asound.conf
+ rm /var/lib/alsa/asound.state
+ :
+ dtoverlay -d /boot/overlays seeed-4mic-voicecard
* Failed to apply overlay '7_seeed-4mic-voicecard' (kernel)
+ true
+ echo 'create seeed-4mic-voicecard asound configure file'
create seeed-4mic-voicecard asound configure file
+ ln -s /etc/voicecard/asound_4mic.conf /etc/asound.conf
+ echo 'create seeed-4mic-voicecard asound status file'
create seeed-4mic-voicecard asound status file
+ ln -s /etc/voicecard/ac108_asound.state /var/lib/alsa/asound.state
+ alsactl restore
+ aplay -l
+ grep -q 'bcm2835 ALSA'
+ exit 0
```

I've tried th diffrent record commands, like:
arecord -Dac108 -f S32_LE -r 16000 -c 4 hello.wav
arecord -Dhw:0,0 -f S16_LE -r 16000 -c 2 hello.wav
arecord -f S16_LE -d 10 -r 16000 --device="plughw:CARD=seeed4micvoicec,DEV=0" /tmp/test-mic.wav
All I get in response is: **arecord: main:828: audio open error: Device or resource busy**

Please help.

closed 2021-03-05 by solidssss

HinTak, 2021-03-05:
What kernel are you currently on?

HinTak, 2021-03-05:
Device / resource busy probably means something else (pulseaudio most likely) is using the audio device...