respeaker/seeed-voicecard #278: Does not work with raspberry pi 4 kernel 5.4.79-v7l+

solidssss, 2020-12-29:
Hi. I cannot get the seed-voicecard 4mic to work with my raspberry 4. It worked great on my old 3b+ but not on the pi 4.
Installed using **sudo ./install.sh 4mic**

**Kernel: 5.4.79-v7l+**

**arecord -l:**  **** List of CAPTURE Hardware Devices ****
card 0: seeed4micvoicec [seeed-4mic-voicecard], device 0: bcm2835-i2s-ac10x-codec0 ac10x-codec.1-003b-0 [bcm2835-i2s-ac10x-codec0 ac10x-codec.1-003b-0]
  Subdevices: 0/1
  Subdevice #0: subdevice #0

**arecord -L:** null
    Discard all samples (playback) or generate zero samples (capture)
jack
    JACK Audio Connection Kit
pulse
    PulseAudio Sound Server
playback
ac108
default
output
sysdefault:CARD=seeed4micvoicec
    seeed-4mic-voicecard, bcm2835-i2s-ac10x-codec0 ac10x-codec.1-003b-0
    Default Audio Device
dmix:CARD=seeed4micvoicec,DEV=0
    seeed-4mic-voicecard, bcm2835-i2s-ac10x-codec0 ac10x-codec.1-003b-0
    Direct sample mixing device
dsnoop:CARD=seeed4micvoicec,DEV=0
    seeed-4mic-voicecard, bcm2835-i2s-ac10x-codec0 ac10x-codec.1-003b-0
    Direct sample snooping device
hw:CARD=seeed4micvoicec,DEV=0
    seeed-4mic-voicecard, bcm2835-i2s-ac10x-codec0 ac10x-codec.1-003b-0
    Direct hardware device without any conversions
plughw:CARD=seeed4micvoicec,DEV=0
    seeed-4mic-voicecard, bcm2835-i2s-ac10x-codec0 ac10x-codec.1-003b-0
    Hardware device with all software conversions
usbstream:CARD=seeed4micvoicec
    seeed-4mic-voicecard
    USB Stream Output

log: 
```
 cat /var/log/seeed-voicecard.log
+ export PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/opt/vc/bin
+ PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/opt/vc/bin
+ OVERLAYS=/boot/overlays
+ '[' -d /boot/firmware/overlays ']'
+ dtparam -d /boot/overlays i2c_arm=on
+ modprobe i2c-dev
+ dtparam -d /boot/overlays spi=on
+ _VER_RUN=
+ CONFIG=/boot/config.txt
+ '[' -f /boot/firmware/usercfg.txt ']'
++ egrep '(1a|UU)'
++ i2cdetect -y 1 0x1a 0x1a
++ awk '{print $2}'
+ is_1a=
++ i2cdetect -y 1 0x35 0x35
++ egrep '(35|UU)'
++ awk '{print $2}'
+ is_35=
++ i2cdetect -y 1 0x3b 0x3b
++ egrep '(3b|UU)'
++ awk '{print $2}'
+ is_3b=UU
+ RPI_HATS='seeed-2mic-voicecard seeed-4mic-voicecard seeed-8mic-voicecard'
+ overlay=
+ '[' x '!=' x ']'
+ '[' xUU '!=' x ']'
+ '[' x == x ']'
+ echo 'install 4mic'
install 4mic
+ overlay=seeed-4mic-voicecard
+ asound_conf=/etc/voicecard/asound_4mic.conf
+ asound_state=/etc/voicecard/ac108_asound.state
+ '[' xUU '!=' x ']'
+ '[' x '!=' x ']'
+ '[' seeed-4mic-voicecard ']'
+ echo Install seeed-4mic-voicecard ...
Install seeed-4mic-voicecard ...
+ rm /etc/asound.conf
+ rm /var/lib/alsa/asound.state
+ :
+ dtoverlay -d /boot/overlays seeed-4mic-voicecard
* Failed to apply overlay '7_seeed-4mic-voicecard' (kernel)
+ true
+ echo 'create seeed-4mic-voicecard asound configure file'
create seeed-4mic-voicecard asound configure file
+ ln -s /etc/voicecard/asound_4mic.conf /etc/asound.conf
+ echo 'create seeed-4mic-voicecard asound status file'
create seeed-4mic-voicecard asound status file
+ ln -s /etc/voicecard/ac108_asound.state /var/lib/alsa/asound.state
+ alsactl restore
+ aplay -l
+ grep -q 'bcm2835 ALSA'
+ exit 0
```

I've tried th diffrent record commands, like:
arecord -Dac108 -f S32_LE -r 16000 -c 4 hello.wav
arecord -Dhw:0,0 -f S16_LE -r 16000 -c 2 hello.wav
arecord -f S16_LE -d 10 -r 16000 --device="plughw:CARD=seeed4micvoicec,DEV=0" /tmp/test-mic.wav
All I get in response is: **arecord: main:828: audio open error: Device or resource busy**

Please help.

HinTak, 2020-12-29:
Afaik the 4-mics captures an 8-channel recording for which the higher 4 is empty, so you should do `-c 8`?... Or one of the 4-mics devices does. Since your arecord output is correct, the overlay failure message does not matter (as it seems that the kernel got the correct overlay anyway somehow). Though I suspect you are not telling us something - such as the earlier part of the dkms message.

I don't work for Seeed Studio - just another user (and I only have the 6-mics device).

solidssss, 2020-12-29:
@HinTak 
I've tried with `-c 8` and got same output.  Not sure what you mean by 'earlier part of the dkms message'.

solidssss, 2020-12-29:
Update: after I rebooting I can run `arecord -d 10 /tmp/test-mic.wav` and then listen to the file. So the mic is working.  But when I run the command again it gives me **arecord: main:828: audio open error: Device or resource busy**.  If I wait a few seconds after boot and type `arecord -d 10 /tmp/test-mic.wav` it does not work.
Something is taking contol of the device after I reboot, but I can't figure out what. Is there any command to show what applications is using arecord?

Docker-J, 2020-12-29:
> Update: If I type `arecord -d 10 /tmp/test-mic.wav` instantly after reboot, it works and I can listen to the file. So the mic is working. But when I run the command again it gives me **arecord: main:828: audio open error: Device or resource busy**.
> 
> If I wait a few seconds after reboot and type `arecord -d 10 /tmp/test-mic.wav` it does not work...
> Something is taking contol of the device after I reboot, but I can't figure out what. Is there any command to show what applications is using arecord?
> I've tried stopping Home assistant, but that did not help.

Same issue.

pi@raspberrypi:~ $ sudo lsof /dev/snd/pcm*
lsof: WARNING: can't stat() fuse.gvfsd-fuse file system /run/user/1000/gvfs
      Output information may be incomplete.
COMMAND    PID USER   FD   TYPE DEVICE SIZE/OFF  NODE NAME
pulseaudi 5324 root  mem    CHR 116,88          18754 /dev/snd/pcmC2D0c
pulseaudi 5324 root  mem    CHR 116,48           9177 /dev/snd/pcmC1D0p
pulseaudi 5324 root  mem    CHR 116,16           9175 /dev/snd/pcmC0D0p
pulseaudi 5324 root   18u   CHR 116,16      0t0  9175 /dev/snd/pcmC0D0p
pulseaudi 5324 root   24u   CHR 116,48      0t0  9177 /dev/snd/pcmC1D0p
pulseaudi 5324 root   30u   CHR 116,88      0t0 18754 /dev/snd/pcmC2D0c

seems like pulseaudio is using some audio devices

solidssss, 2020-12-30:
Running **sudo lsof /dev/snd/pcm*** gives me this output:
**COMMAND    PID USER   FD   TYPE DEVICE SIZE/OFF  NODE NAME
pulseaudi 4941 root  mem    CHR 116,56          15697 /dev/snd/pcmC1D0c
pulseaudi 4941 root  mem    CHR 116,17           1892 /dev/snd/pcmC0D1p
pulseaudi 4941 root   18u   CHR 116,17      0t0  1892 /dev/snd/pcmC0D1p
pulseaudi 4941 root   26u   CHR 116,56      0t0 15697 /dev/snd/pcmC1D0c
**
Not sure what to do now...

solidssss, 2020-12-30:
Alright, so I just tried with a fresh copy of the latest **2020-12-02-raspios-buster-armhf-lite** and I still get the same error...

closed 2021-01-02 by solidssss