respeaker/seeed-voicecard #279: Why seeed-voice-card driver?

shawaj, 2021-01-02:
Is there a reason why in the 2 mic version you are using the `simple-audio-card` driver but in the 4 and 6 mix versions your are using the custom `seeed-voice-card` instead?

I'm trying to add support for your boards to the mainline kernel and want to understand why you have changed this and if it is really necessary?

HinTak, 2021-01-02:
You mean kernel 5.9 ? See https://github.com/HinTak/seeed-voicecard/tree/v5.9 . 5.10 was out a few days ago, I haven't looked at it yet.

The 2-mics device has different hardware compared to the others; but as far as I remember it also uses the custom seeed voice card driver, although it can be used with simple-audio-card -based driver from elsewhere also.

shawaj, 2021-01-02:
What I mean is, in the [device tree overlay](seeed-2mic-voicecard-overlay.dts) for the 2 mic version it uses simple-audio-card.

But in the [4 mic]() and [6 mic]() it uses seeed-voice-card.

I wondered why it wasn't using simple-audio-card and whether it would be compatible?

shawaj, 2021-01-02:
@HinTak basically I'm trying to add the overlay support directly to RasPi OS without needing install script like it is now, to make it quicker and easier for users https://github.com/raspberrypi/linux/pull/4039

HinTak, 2021-01-02:
You cannot - see licensing issue I commented on at https://github.com/respeaker/seeed-voicecard/issues/213

shawaj, 2021-01-02:
https://github.com/respeaker/seeed-voicecard/blob/v5.5/ac101.c

It has details for people from allwinner and seeed.

It doesn't seem to be an issue.

Also I think it can go into Raspberry Pi kernel it doesn't necessarily need to go all the way upstream...

shawaj, 2021-01-02:
Also @HinTak the domain name works and redirects to allwinner www.reuuimllatech.com

HinTak, 2021-01-03:
http://www.reuuimllatech.com/ is somebody's idea of a joke - since allwinner people decided to hid their identity by spelling their company name backwards, the way site also shows their web site backward (with the right browser).

Anyway, you don't have to convince me that the copyright allows upstreaming - you need to convince the upstream people (which ever you mean). And from my experience, good luck. 

And I am speaking as somebody who have successfully upstream code into the Linux kernel in the past, as well as a person who reviewed (and would have certainly rejected such copyright-dubious...) submitted code.

shawaj, 2021-01-03:
@HinTak it is a 302 redirect straight to allwinner ```
Result
http://www.reuuimllatech.com/
301 Moved Permanently
https://www.reuuimllatech.com/
302 Found
https://www.allwinnertech.com/
200 OK
```

HinTak, 2021-01-03:
You don't need to convince me. Good luck upstreaming.

shawaj, 2021-01-03:
@HinTak thanks.

And back to my original question, is there a reason they're not just using simple-audio-card and have created their own version?

HinTak, 2021-01-03:
It was based on simple-audio-card, then diverged. You can check the repository's history, back to about 2017, I think.

HinTak, 2021-01-03:
April 2018 - https://github.com/respeaker/seeed-voicecard/commit/d0d9107c912893676405bcd27dba692b886a87cb

shawaj, 2021-01-16:
Thanks @HinTak so I guess in principle it will work with simple-audio-card assuming the codec drivers are available

HinTak, 2021-01-16:
@shawaj I think you likely guess wrong, but you are welcomed to try to fix 2 years of divergence... :)

shawaj, 2021-01-16:
It worked 2 years ago, and still works with those old drivers. So seems like it is fine to me

HinTak, 2021-01-16:
@shawaj against 4.10 or 3.x kernel, two years ago. Good luck doing it for 5.x . 

You seem to think that simple-audio-card has not changed - sorry to disappoint you, but the divergence is largely at the upstream's side: the Linux kernel audio stack / RaspberryPi people have re-arranged the simple-audio-card code about 5-6 times at least. My branches are largely just to keep up with upstream's changes; so my branches are closer to upstream than seeed-studio's. And that's already a lot of hard work.

As I said, good luck trying.

shawaj, 2021-01-16:
Your fork seems to be using the seeed-voice-card not the simple-audio-card? So not really understanding your meaning.

Your work is great, and I think probably many people appreciate it very much (me included!), but I think having it in the mainline pi kernel, and meaning that this is maintained by a wider pool of people is probably long term a better solution especially for quite common chip like wm8960. And makes it easier for users.

For the X power ones I guess not as common but same applies, assuming codec can be included in Pi mainline.

Anyway, just my opinion, sounds like you probably disagree :-)

HinTak, 2021-01-17:
Seeed-voice-card is how simple-audio-card was like back in early 4.x. Upstream simple audio card changes every 2/3 kernel minor versions, so that's about a dozen incompatible changes. For in-tree drivers they get updated when these things happens.

Anyway, it is a chicken-and-egg problem - if seeed-voice-card becomes more **current** simple-audio-card like, it becomes easier to catch-up... But you need to catch up to make it more **current** audio-card like.

And that's aside from the X power licensing issue - it is rather frown upon to submit code you did not write, and of a quality you cannot guarantee... And the technical issue like #251 which destabilizes the whole system... I have spent enough time on it and I prefer not to :-( .

AIWintermuteAI, 2021-07-23:
Hello!
This seems to be a general question not a bug/feature request. It is better to be addressed in Seeed Forum reSpeaker subforum.
https://forum.seeedstudio.com/c/products/respeaker/15

closed 2021-07-23 by AIWintermuteAI