respeaker/seeed-voicecard #281: New wm8960-soundcard overlay

shawaj, 2021-01-08:
The wm8960-soundcard overlay has been merged into the RasPi kernel...

https://github.com/raspberrypi/linux/pull/4039

Once it gets into the firmware repo (a few days probably):

https://github.com/raspberrypi/firmware/tree/master/boot/overlays

You should be able to use the seeed 2 mic array by simply adding `dtoverlay=wm8960-soundcard` to the `/boot/config.txt` file.

shawaj, 2021-01-16:
In the official firmware now - https://github.com/raspberrypi/firmware/blob/master/boot/overlays/wm8960-soundcard.dtbo

yapsr, 2021-01-16:
I have a Respeaker 2-mics pi hat installed on my Raspberry Pi 3B+.

I have booted from a fresh install of Raspbery Pi OS, running `rpi-update` (which updates to kernel 5.10.7-v7+), adding `dtoverlay=wm8960-soundcard`  to `/boot/config.txt`, rebooting and running `aplay -L` does not detect the respeaker 2-mics pi hat.

What did I do wrong?
Do I still need to run any other script to be able to view this device in alsa?

HinTak, 2021-01-16:
@yapsr you most likely need v5.9 branch of my repo : https://github.com/HinTak/seeed-voicecard - if it works for you, please feel free to click the donate link.

shawaj, 2021-01-16:
@yapsr can you do:

```
cd /boot/overlays
ls
```

And see if wm8960-soundcard is in there?

yapsr, 2021-01-16:
Yes, it's there. Like I wrote, it is also loaded, according to `vcdbg log msg`.

yapsr, 2021-01-17:
@shawaj With this overlay, I expect the sound card to be directly available for playback and recording via `aplay` and `arecord`. Is this what is supposed to happen? Or do I need additional steps, like @HinTak suggests?

Did you try it yourself, perhaps?

M0Rf30, 2021-01-22:
@yapsr You have to set `CONFIG_SND_SOC_WM8960=m` in your Kconfig. DTB overlay is useless without a kernel module

HinTak, 2021-01-23:
@M0Rf30 : @yapsr got his kernel (binary) via rpi-update , so your advice is not applicable/appropriate . He needs to get my v5.9 branch.

M0Rf30, 2021-01-23:
@HinTak  or it needs to check his configuration with a `zcat /proc/config.gz | grep -i WM8960`. if the previous command something different from `CONFIG_SND_SOC_WM8960=m`, that's the problem. I'm talking of a more generic approach to do a little bit of troubleshooting. Awareness is much more important than `use my branch` IHMO. Anyway thanks for you efforts.

HinTak, 2021-01-23:
@M0Rf30 I beg to differ. Most people prefer not to be a kernel hacker, and rather have a working system by the quickest route, however it is achieved. `Kconfig` is a source build reference, and definitely not appropriate.

AIWintermuteAI, 2021-07-30:
Hi @shawaj !

We have tried the sound overlay after compiling Raspberry Pi kernel with ```CONFIG_SND_SOC_WM8960=m``` enabled for testing a few month ago. During the testing we have found that there is strong noise on recording, playback was normal. The issue is reported in Raspberry Pi Kernel Github as well:
https://github.com/raspberrypi/linux/issues/4384

As per this issue https://github.com/raspberrypi/linux/issues/4454 kernel driver is compiled by default, so it will be available in future releases of Raspberry Pi. However since Raspberry Pi kernel WM8960 SoC codec is broken (at least it doesn't work well with Seeed studio 2-mic), the users are still strongly advised to install seeed-voicecard driver.

closed 2021-07-30 by AIWintermuteAI