respeaker/seeed-voicecard #283: No soundcards found

greykeys, 2021-01-29:
I am trying to setup a ReSpeaker 2mic Pi-Hat for a Raspberry Pi Zero. I walked through some existing issues on this repo but they didn't help me resolve my issue. I have tried with multiple different Raspbian OS (currently on `2019-09-26-raspbian-buster-lite`) to no avail. I've tried setting up using both this repo and HinTak's (https://github.com/HinTak/seeed-voicecard) and neither seem to be working for me.

```
pi@raspberrypi:~/seeed-voicecard $ aplay -l
aplay: device_list:272: no soundcards found...
pi@raspberrypi:~/seeed-voicecard $ arecord -l
arecord: device_list:272: no soundcards found...
```

dmesg tail output
```
   17.961895] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   17.962486] usbcore: registered new interface driver brcmfmac
[   17.988397] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43430-sdio.raspberrypi,model-zero-w.txt failed with error -2
[   18.266130] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   18.266350] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[   18.267732] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Oct 22 2019 01:59:28 version 7.45.98.94 (r723000 CY) FWID 01-3b33decd
[   26.336227] 8021q: 802.1Q VLAN Support v1.8
[   28.439018] random: crng init done
[   28.439048] random: 7 urandom warning(s) missed due to ratelimiting
[   28.863291] uart-pl011 20201000.serial: no DMA platform data
[   29.022363] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
[   29.164557] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   30.125890] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/i2c@7e804000/status
[   30.893748] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/spi@7e204000/status
[   31.298862] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   33.370956] Bluetooth: Core ver 2.22
[   33.371120] NET: Registered protocol family 31
[   33.371134] Bluetooth: HCI device and connection manager initialized
[   33.371164] Bluetooth: HCI socket layer initialized
[   33.371182] Bluetooth: L2CAP socket layer initialized
[   33.371224] Bluetooth: SCO socket layer initialized
[   33.395081] Bluetooth: HCI UART driver ver 2.3
[   33.395103] Bluetooth: HCI UART protocol H4 registered
[   33.395220] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   33.395528] Bluetooth: HCI UART protocol Broadcom registered
[   34.188074] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   34.188092] Bluetooth: BNEP filters: protocol multicast
[   34.188122] Bluetooth: BNEP socket layer initialized
```

systemctl service status
```
● seeed-voicecard.service - Seeed Voicecard service
   Loaded: loaded (/lib/systemd/system/seeed-voicecard.service; enabled; vendor preset: enabled)
   Active: active (exited) since Fri 2021-01-29 07:05:56 GMT; 3min 14s ago
  Process: 291 ExecStart=/usr/bin/seeed-voicecard (code=exited, status=0/SUCCESS)
 Main PID: 291 (code=exited, status=0/SUCCESS)

Jan 29 07:05:56 raspberrypi seeed-voicecard[291]: + '[' x '!=' x ']'
Jan 29 07:05:56 raspberrypi seeed-voicecard[291]: + '[' x '!=' x ']'
Jan 29 07:05:56 raspberrypi seeed-voicecard[291]: + '[' x '!=' x ']'
Jan 29 07:05:56 raspberrypi seeed-voicecard[291]: + '[' '' ']'
Jan 29 07:05:56 raspberrypi seeed-voicecard[291]: + alsactl restore
Jan 29 07:05:56 raspberrypi seeed-voicecard[291]: alsactl: load_state:1735: No soundcards found...
Jan 29 07:05:56 raspberrypi seeed-voicecard[291]: + grep -q 'bcm2835 ALSA'
Jan 29 07:05:56 raspberrypi seeed-voicecard[291]: + aplay -l
Jan 29 07:05:56 raspberrypi seeed-voicecard[291]: aplay: device_list:272: no soundcards found...
Jan 29 07:05:56 raspberrypi systemd[1]: Started Seeed Voicecard service.
```

I've reseated my Pi-Hat, and that doesn't seem to be making a difference either. The link to the Pi image in the [FAQ section of this repo](https://github.com/respeaker/seeed-voicecard#faq) is broken so I'm not able to test with a properly working image either to localize the issue.

Any help would be greatly appreciated. Thanks!

HinTak, 2021-01-29:
Posting the whole of `dmesg` (not just the last few lines), plus the output of `lsmod` would be good.

Btw, pretty sure that the current code (either seeed studio's or mine) does not work with 2019-09-26-raspbian-buster-lite . Stretch was on 4.19 in march 2020, moving onto 5.4 in summer 2020 . The current code is 5.4+ only. There is a v4.19 branch on mine, which you could try if you want to use such "antique" buster image. The 4.19 branch also works for earlier 4.11(?) I think. But you are talking about 3-year-old raspbian now...

greykeys, 2021-01-29:
@HinTak Thanks for taking a look. I realize that the buster image is old. I should've clarified I've tried setting up with newer versions of Raspbian as well (`2019-09-26-raspbian-buster-lite`, `2020-02-13-raspbian-buster-lite`,...) and I end up running into some issue or the other (Some issues here pointed to the fact that newer kernels were not supported/supported as well). I'm happy to try with a version you think works best for this driver, as I'm not tied to any specific version. Until then, please find the outputs you asked for below:

*dmesg*

```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.83+ (dom@buildbot) (gcc version 8.4.0 (Ubuntu/Linaro 8.4.0-3ubuntu1)) #1379 Mon Dec 14 13:06:05 GMT 2020
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi Zero W Rev 1.1
[    0.000000] Memory policy: Data cache writeback
[    0.000000] Reserved memory: created CMA memory pool at 0x17c00000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] On node 0 totalpages: 114688
[    0.000000]   Normal zone: 1008 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 114688 pages, LIFO batch:31
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 113680
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:4E:17:E9 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  console=ttyS0,115200 console=tty1 root=PARTUUID=6c586e13-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 376180K/458752K available (7816K kernel code, 712K rwdata, 2468K rodata, 508K init, 804K bss, 17036K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] ftrace: allocating 27743 entries in 55 pages
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] random: get_random_bytes called from start_kernel+0x2a8/0x4d8 with crng_init=0
[    0.000022] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000089] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000206] bcm2835: system timer (irq = 27)
[    0.000822] Console: colour dummy device 80x30
[    0.001383] printk: console [tty1] enabled
[    0.001476] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
[    0.050394] pid_max: default: 32768 minimum: 301
[    0.050731] LSM: Security Framework initializing
[    0.051097] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.051167] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.053381] Disabling memory control group subsystem
[    0.053704] CPU: Testing write buffer coherency: ok
[    0.055471] Setting up static identity map for 0x8200 - 0x8238
[    0.056720] devtmpfs: initialized
[    0.069313] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
[    0.069850] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.069942] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
[    0.079040] pinctrl core: initialized pinctrl subsystem
[    0.081193] NET: Registered protocol family 16
[    0.085838] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.092380] audit: initializing netlink subsys (disabled)
[    0.094593] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
[    0.094662] hw-breakpoint: maximum watchpoint size is 4 bytes.
[    0.094859] Serial: AMBA PL011 UART driver
[    0.098652] bcm2835-mbox 2000b880.mailbox: mailbox enabled
[    0.100759] audit: type=2000 audit(0.090:1): state=initialized audit_enabled=0 res=1
[    0.120836] raspberrypi-firmware soc:firmware: Attached to firmware from 2021-01-08 14:33, variant start
[    0.130861] raspberrypi-firmware soc:firmware: Firmware hash is 194a85abd768c7334bbadc3f1911c10a7d18ed14
[    0.194965] bcm2835-dma 20007000.dma: DMA legacy API manager, dmachans=0x1
[    0.198202] SCSI subsystem initialized
[    0.198556] usbcore: registered new interface driver usbfs
[    0.198713] usbcore: registered new interface driver hub
[    0.199023] usbcore: registered new device driver usb
[    0.201923] clocksource: Switched to clocksource timer
[    1.676263] VFS: Disk quotas dquot_6.6.0
[    1.676491] VFS: Dquot-cache hash table entries: 1024 (order 0, 4096 bytes)
[    1.676921] FS-Cache: Loaded
[    1.677375] CacheFiles: Loaded
[    1.698731] thermal_sys: Registered thermal governor 'step_wise'
[    1.699473] NET: Registered protocol family 2
[    1.700948] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 4096 bytes, linear)
[... 231 lines cut ...]
spi_bcm2835            24576  0
[... cut ...]
videodev              225280  6 bcm2835_isp,bcm2835_codec,videobuf2_common,bcm2835_v4l2,v4l2_mem2mem,videobuf2_v4l2
mc                     45056  6 bcm2835_isp,bcm2835_codec,videobuf2_common,videodev,v4l2_mem2mem,videobuf2_v4l2
vc_sm_cma              32768  2 bcm2835_isp,bcm2835_mmal_vchiq
uio_pdrv_genirq        16384  0
fixed                  16384  0
uio                    20480  1 uio_pdrv_genirq
i2c_dev                16384  0
ip_tables              28672  0
x_tables               32768  1 ip_tables
ipv6                  446464  18
```

HinTak, 2021-01-29:
Try 
`sudo modprobe -v snd_soc_wm8960` and `sudo modprobe -v snd_soc_seeed_voicecard`? Don't reboot after that but just go and run the aplay / arecord commands again.

Also what is in your `/boot/config.txt`?

greykeys, 2021-01-29:
Looks like the right modules to successfully run those commands do not exist for my setup:

```
pi@raspberrypi:~ $ sudo modprobe -v snd_soc_wm8960
modprobe: FATAL: Module snd_soc_wm8960 not found in directory /lib/modules/5.4.83+
pi@raspberrypi:~ $ sudo modprobe -v snd_soc_seeed_voicecard
modprobe: FATAL: Module snd_soc_seeed_voicecard not found in directory /lib/modules/5.4.83+
pi@raspberrypi:~ $ aplay -l
aplay: device_list:272: no soundcards found...
pi@raspberrypi:~ $ arecord -l
arecord: device_list:272: no soundcards found...
````

The following are the contents of my `/boot/config.txt`

```
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on
#dtparam=i2s=on
dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d
dtoverlay=i2s-mmap
dtparam=i2s=on
```

If you want me to re-setup a new version of Raspbian and retry I'm happy to do that. In case this is too old to diagnose any issues.

HinTak, 2021-01-29:
No need to start afresh - I want to see the whole output of `./install.sh` from the source directory. It should have created those snd_soc_* and copied them to the correct places, but haven't.

greykeys, 2021-01-29:
Sure, so do you want me to run `uninstall.sh` and then run `install.sh` and paste the output I get?

HinTak, 2021-01-29:
No need, since nothing seems to have been installed...

greykeys, 2021-01-29:
Sure, here's the output from the install:

```
pi@raspberrypi:~/seeed-voicecard $ sudo ./install.sh
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Get:2 http://archive.raspberrypi.org/debian buster InRelease [32.8 kB]
Hit:3 http://packages.microsoft.com/repos/code stable InRelease
Get:4 http://archive.raspberrypi.org/debian buster/main armhf Packages [353 kB]
Fetched 385 kB in 4s (87.0 kB/s)
Reading package lists... Done
Building dependency tree
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree
Reading state information... Done
raspberrypi-kernel is already the newest version (1.20210108-1).
raspberrypi-kernel-headers is already the newest version (1.20210108-1).
The following package was automatically installed and is no longer required:
  rpi-eeprom-images
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package linux-raspi
E: Unable to locate package linux-headers-raspi
E: Unable to locate package linux-image-raspi
Reading package lists... Done
Building dependency tree
Reading state information... Done
dkms is already the newest version (2.6.1-4).
git is already the newest version (1:2.20.1-2+deb10u3).
i2c-tools is already the newest version (4.1-1).
libasound2-plugins is already the newest version (1.1.8-1).
The following package was automatically installed and is no longer required:
  rpi-eeprom-images
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

------------------------------
Deleting module version: 0.3
completely from the DKMS tree.
------------------------------
Done.

Creating symlink /var/lib/dkms/seeed-voicecard/0.3/source ->
                 /usr/src/seeed-voicecard-0.3

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area.....
make -j1 KERNELRELEASE=5.4.83+ -C /lib/modules/5.4.83+/build M=/var/lib/dkms/seeed-voicecard/0.3/build.........................
cleaning build area.....

DKMS: build completed.

snd-soc-wm8960.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.4.83+/kernel/sound/soc/codecs/

snd-soc-ac108.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.4.83+/kernel/sound/soc/codecs/

snd-soc-seeed-voicecard.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.4.83+/kernel/sound/soc/bcm/

depmod....

DKMS: install completed.
mkdir: cannot create directory ‘/etc/voicecard’: File exists
git init
Reinitialized existing Git repository in /etc/voicecard/.git/
git add --all
git commit -m "origin configures"
On branch master
nothing to commit, working tree clean
------------------------------------------------------
Please reboot your raspberry pi to apply all settings
Enjoy!
------------------------------------------------------
```

HinTak, 2021-01-29:
What is inside `/lib/modules/5.4.83+/kernel/sound/soc/codecs/` ?

HinTak, 2021-01-29:
If there are a few snd_soc* files there, try the two modprobe again...

greykeys, 2021-01-29:
There were some `snd_soc*` files in there, so I tried the modprobe commands and checked aplay/arecord. Still no luck

```
pi@raspberrypi:~ $ ls /lib/modules/5.4.83+/kernel/sound/soc/codecs/
snd-soc-ac108.ko         snd-soc-adau1977.ko    snd-soc-cs4271.ko       snd-soc-i-sabre-codec.ko  snd-soc-pcm186x-i2c.ko  snd-soc-sigmadsp-i2c.ko  snd-soc-tlv320aic32x4-i2c.ko  snd-soc-wm8804-i2c.ko
snd-soc-ad193x-i2c.ko    snd-soc-adau7002.ko    snd-soc-cs42xx8-i2c.ko  snd-soc-ma120x0p.ko       snd-soc-pcm186x.ko      snd-soc-sigmadsp.ko      snd-soc-tlv320aic32x4.ko      snd-soc-wm8804.ko
snd-soc-ad193x.ko        snd-soc-ak4554.ko      snd-soc-cs42xx8.ko      snd-soc-max98357a.ko      snd-soc-pcm5102a.ko     snd-soc-spdif-rx.ko      snd-soc-tpa6130a2.ko          snd-soc-wm8960.ko
snd-soc-ad193x-spi.ko    snd-soc-arizona.ko     snd-soc-da7213.ko       snd-soc-pcm1794a.ko       snd-soc-pcm512x-i2c.ko  snd-soc-spdif-tx.ko      snd-soc-wm5102.ko             snd-soc-wm-adsp.ko
snd-soc-adau1701.ko      snd-soc-cs4265.ko      snd-soc-dmic.ko         snd-soc-pcm179x-codec.ko  snd-soc-pcm512x.ko      snd-soc-tas5713.ko       snd-soc-wm8731.ko
snd-soc-adau1977-i2c.ko  snd-soc-cs4271-i2c.ko  snd-soc-ics43432.ko     snd-soc-pcm179x-i2c.ko    snd-soc-sgtl5000.ko     snd-soc-tas571x.ko       snd-soc-wm8741.ko
pi@raspberrypi:~ $ sudo modprobe -v snd_soc_wm8960
pi@raspberrypi:~ $ sudo modprobe -v snd_soc_seeed_voicecard
pi@raspberrypi:~ $ aplay -l
aplay: device_list:272: no soundcards found...
pi@raspberrypi:~ $ arecord -l
arecord: device_list:272: no soundcards found...
pi@raspberrypi:~ $
```

HinTak, 2021-01-29:
The tail end of dmesg (before/without reboot), and what's in your /boot/config.txt ?

greykeys, 2021-01-29:
I rebooted after to test again, so I can only provide the `dmesg` output after reboot if it's of any help:

```
pi@raspberrypi:~ $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.83+ (dom@buildbot) (gcc version 8.4.0 (Ubuntu/Linaro 8.4.0-3ubuntu1)) #1379 Mon Dec 14 13:06:05 GMT 2020
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi Zero W Rev 1.1
[    0.000000] Memory policy: Data cache writeback
[    0.000000] Reserved memory: created CMA memory pool at 0x17c00000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] On node 0 totalpages: 114688
[    0.000000]   Normal zone: 1008 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 114688 pages, LIFO batch:31
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 113680
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:4E:17:E9 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  console=ttyS0,115200 console=tty1 root=PARTUUID=6c586e13-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 376180K/458752K available (7816K kernel code, 712K rwdata, 2468K rodata, 508K init, 804K bss, 17036K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] ftrace: allocating 27743 entries in 55 pages
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] random: get_random_bytes called from start_kernel+0x2a8/0x4d8 with crng_init=0
[    0.000023] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000090] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000209] bcm2835: system timer (irq = 27)
[    0.000826] Console: colour dummy device 80x30
[    0.001387] printk: console [tty1] enabled
[    0.001481] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
[    0.050400] pid_max: default: 32768 minimum: 301
[    0.050737] LSM: Security Framework initializing
[    0.051106] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.051175] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.053390] Disabling memory control group subsystem
[    0.053709] CPU: Testing write buffer coherency: ok
[    0.055489] Setting up static identity map for 0x8200 - 0x8238
[    0.056735] devtmpfs: initialized
[    0.069313] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
[    0.069859] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.069950] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
[    0.079053] pinctrl core: initialized pinctrl subsystem
[    0.081200] NET: Registered protocol family 16
[    0.085843] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.092373] audit: initializing netlink subsys (disabled)
[    0.094580] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
[    0.094648] hw-breakpoint: maximum watchpoint size is 4 bytes.
[    0.094848] Serial: AMBA PL011 UART driver
[    0.098636] bcm2835-mbox 2000b880.mailbox: mailbox enabled
[    0.100764] audit: type=2000 audit(0.090:1): state=initialized audit_enabled=0 res=1
[    0.120846] raspberrypi-firmware soc:firmware: Attached to firmware from 2021-01-08 14:33, variant start
[    0.130869] raspberrypi-firmware soc:firmware: Firmware hash is 194a85abd768c7334bbadc3f1911c10a7d18ed14
[    0.194997] bcm2835-dma 20007000.dma: DMA legacy API manager, dmachans=0x1
[    0.198239] SCSI subsystem initialized
[    0.198592] usbcore: registered new interface driver usbfs
[    0.198747] usbcore: registered new interface driver hub
[    0.199059] usbcore: registered new device driver usb
[    0.201960] clocksource: Switched to clocksource timer
[    1.676277] VFS: Disk quotas dquot_6.6.0
[    1.676503] VFS: Dquot-cache hash table entries: 1024 (order 0, 4096 bytes)
[    1.676933] FS-Cache: Loaded
[    1.677388] CacheFiles: Loaded
[    1.698796] thermal_sys: Registered thermal governor 'step_wise'
[    1.699538] NET: Registered protocol family 2
[    1.701023] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 4096 bytes, linear)
[    1.701152] TCP established hash table entries: 4096 (order: 2, 16384 bytes, linear)
[... 247 lines cut ...]

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on
#dtparam=i2s=on
dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d
dtoverlay=i2s-mmap
dtparam=i2s=on
```

HinTak, 2021-01-29:
What does `i2cdetect -y 1 0x1a 0x1a` says? Actually just plain `i2cdetect -y 1` would be more informative. Post that.

greykeys, 2021-01-29:
Sure, here are the outputs:

```
pi@raspberrypi:~ $ i2cdetect -y 1 0x1a 0x1a
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:
10:                               --
20:
30:
40:
50:
60:
70:
pi@raspberrypi:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

greykeys, 2021-01-29:
Does this indicate a hardware/soldering connection issue? What would the ideal output for this check be?

HinTak, 2021-01-30:
Yes, that looks like a hardware problem. For the 2mic device, you should have a "UU" where the two "--" are. The other seeed studio devices have a few more UU's.

You should swap the pi or the mics - it could be either being faulty. (or dirty in the sockets between them?).

greykeys, 2021-01-30:
I see, I'll try swapping to see if that resolves this issue. Thanks a ton @HinTak for your help in diagnosing this issue and your efforts in keeping these drivers up-to-date. I'd have been breaking my head over this for I don't know how long otherwise.

HinTak, 2021-01-30:
@greykeys you are welcomed. It helps if one has one working device to compare with. These (the pi zero + the 2-mics is probably under 30$US) are relatively cheap toys so it is understandable after-sale support is limited, but nonetheless the lack of seeed studio staff's presence here is disappointing. If you feel generous, you can always buy me a beer with the donate links at https://hintak.github.io . :)

greykeys, 2021-01-30:
Some FAQ/sanity checks for common issues may have been helpful. I was hoping to use the FAQ image link from the README to see if it was a s/w or h/w issue, but looks like its been broken for at least a few months now based on other git issues.

Regardless, thank you for your help. I donated from your page links just a little while ago :)

I'll close this issue as it looks like h/w on my end.

closed 2021-01-30 by greykeys

ladyada, 2021-01-30:
@HinTak thanks for your efforts! i've contributed too :)

HinTak, 2021-01-30:
Cheers!

HinTak, 2021-01-30:
@ladyada argh, you do some cool stuff with your projects!