respeaker/seeed-voicecard #285: 64bit

cl84, 2021-02-03:
is there any way to install ReSpeaker 2-Mics Pi HAT on 64bit rasbian?

HinTak, 2021-02-03:
Does it not work? Have you followed the install instructions?

cl84, 2021-02-03:
no, it doesn't work, compiling the driver fails

HinTak, 2021-02-03:
Depends on what kernel versions you are on. If you are on anything other than 5.4/5.5, grep one of the closest number branches from my repo. And click donate when it works...

HinTak, 2021-02-04:
Oh, apparently raspbian is going onto v5.10 soon. You need my v5.9 branch for that.

cl84, 2021-02-05:
in fact my rasbian is 5.10, how can I install the driver on this kernel?

HinTak, 2021-02-05:
As I already said, clone my repo, https://github.com/HinTak/seeed-voicecard, use the v5.9 branch (it works with 5.10 too), follows the install instruction. When it works, consider clicking the donate link.

hackebike, 2021-02-05:
@HinTak I tried to install the 5.9 seeed-voicecard on a Raspberry PI OS 64 bit Kernel 5.10 and I am getting this error at the very end, any ideas?

DKMS: add completed.

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=5.10.11-v8+ -C /lib/modules/5.10.11-v8+/build M=/var/lib/dkms/seeed-voicecard/0.3/build....(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.10.11-v8+ (aarch64)
Consult /var/lib/dkms/seeed-voicecard/0.3/build/make.log for more information.
setup git config
git init
Initialized empty Git repository in /etc/voicecard/.git/
git add --all
git commit -m "origin configures"
[master (root-commit) 8e32a2e] origin configures
7 files changed, 1476 insertions(+)
create mode 100644 ac108_6mic.state
create mode 100644 ac108_asound.state
create mode 100644 asound_2mic.conf
create mode 100644 asound_4mic.conf
create mode 100644 asound_6mic.conf
create mode 100644 dkms.conf
create mode 100644 wm8960_asound.state

Created symlink /etc/systemd/system/sysinit.target.wants/seeed-voicecard.service → /lib/systemd/system/seeed-voicecard.service.
Please reboot your raspberry pi to apply all settings

Enjoy!

HinTak, 2021-02-05:
"make.log" shows you either did not read - or did not understand. I wrote 3 times already: you need the v5.9 branch of my repo . The default branch is v5.5, and there is a v5.7 branch, a v4.19 branch, etc. After clone my repo, you need to switch to the v5.9 branch with `git checkout -b v5.9 origin/v5.9`.

And I'll write a 3rd time: when it works, consider clicking the donate link. I don't work for seeed studio.

HinTak, 2021-02-05:
I have just switched the default branch on my repo to v5.9 , since the newest raspbian kernel is 5.10.11 ( I checked myself). It was tagged 20210201 so presumably came out earlier this week. This is the first 5.10.x kernel from raspbian. The last 6 months or so were all v5.4.x .

hackebike, 2021-02-05:
I did read and understand everything, but when I selected 5.9 I thought github would switch the git link to that version which actually it does not, so I got the wrong versioning.
I downloaded the 5.9 and everything compiled ok.

HinTak, 2021-02-05:
I guess it is not common knowledge how to switch branch    on the web site. If you choose "download tar ball", you do get the branch you choose; but cloning give you the whole repo with every branch, and on your side, it would still show the default branch, until you switch locally.

HinTak, 2021-02-05:
`git branch` and `git status` tell you what branch you are on.

HinTak, 2021-02-06:
@hackebike btw, the 4-mics and 6-mics devices (not the 2-mics @cl84 has) are affected by the crash on the 64-bit OS - see my comment -
https://github.com/respeaker/seeed-voicecard/issues/251#issuecomment-774431976 - likely you have seen the crash by now : ). You probably want to either switch to ubuntu 64-bit and install my slighly-modified kernel packages, or switch to 32-bit raspbian. I am unlikely to try to build sllightly-tweaked 64-bit raspbian kernel packages.

HinTak, 2021-02-06:
@hackebike if you are finding your pi not booting or not being able to use it because of it is crashing, and if you have access to a linux box, you can backtrack and recover by stopping the seeed driver from loading, doing two things (relative to the bottom of the corresponding partitions):

- edit /boot/config.txt to comment out the 'dtoverlay=seeed..' line towards the end,

- remove the "/etc/systemd/system/sysinit.target.wants/seeed-voicecard.service" symlink.

I wrote "access to a linux box" as the 2nd modification can only be done on a linux box. (that partition is not readable by other OSes). The first change is doing-able on other OSes. On linux, two partitions would be auto-mounted under something like "/run/media/yourusername/boot" and "/run/media/yourusername/rootfs" respectively.

You need to do both steps, as either would try to load the seeed driver.

hackebike, 2021-02-06:
thanks for the info, so if I encounter these issues I know what to do.
From your comments looks like Seeed does not support very well their products!

Anyway for now everything looking fine, compilation ok and the tests are ok too.
I was trying to use Picovoice but looks like they are not supporting Raspberry PI OS 64 bit at this stage.
So I need to find another project that does a similar job, off line voice recognition, if you know any let me know.

HinTak, 2021-02-07:
For a ~$20US (the 4-mics device) or ~$10US (the 2-mics device), understandably after-sale support is limited... but nonetheless, the lack of Seeed Studio staff's mere presence is disappointing.

The matrixio device is a bit more expensive but the software code-quality is better. I have had two pulls and haven't heard from their staffer either... you'll need one of my branches for that device too (one of the pulls), if you want to make it work under 5.10.
To be fair, raspbian puts out a 5.10-based kernel only last week (it was all 5.4-based for most of 2020), so it would take a bit of time for vendors to catch up - I see that the picovoice guy responding to you is not clued up to that either. OTOH, desktop linux has been 5.8/5.9/5.10-based for almost a year now, and it is possible to update the code before raspbian switches (as I did).

The octo cards are very well-maintained, but they are more for people with more serious sound-processsing use, rather than voice-assitant type useage.

There is a donate button at https://hintak.github.io if you are feeling generous.

cl84, 2021-02-08:
Hello, I still have compilation errors. Why? What am I doing wrong? Once the correct branch has been downloaded I launch ./install.sh
[image]
[image]

hackebike, 2021-02-09:
Just  wanted to share something else to fix an error.
I installed another PI with latest OS kernel 5.10, tried to install the 5.9 for the seeedstudio ReSpeaker code from your branch which worked fine with the other PI I have.
I got an error at the end of the installation that it could not find the kernel headers, even though everything was fresh installed.
Of course **arecord -L** did not find the card.

So digging around this command fixed the problem:

**sudo apt-get -y --reinstall install raspberrypi-kernel**

I am not familiar with kernel issues, so can't explain why it fixed, maybe you know better. Just wanted to share since others might encounter this issue.

HinTak, 2021-02-10:
That's likely a red herring. I think it is a quirk of debian / ubuntu apt system: you need to occasionally do "sudo apt update" to update its knowledge about packages available for upgrading (as opposed opposed to "sudo apt upgrade" which does the actual upgrade). The update is done periodically, once a week, I think, so for very freshly installed system, it may not be knowledgeable about packages available. I just checked the install script - it might be useful to include "apt update" just before install/upgrade.

WilliamVJacob, 2021-02-19:
Please follow  my repository to install Respeaker Driver for 64 bit Raspbian OS

https://github.com/WilliamVJacob/seeed-voicecad

b-mq, 2021-02-20:
@WilliamVJacob you missed an "r" in "seed-voiceca**r**d" your link, therefore it's broken
https://github.com/WilliamVJacob/seeed-voicecard

WilliamVJacob, 2021-02-20:
@b-mq  Thank you so much for letting me know . I have updated the link in the comment

HinTak, 2021-02-20:
@WilliamVJacob I have had a look at you repo - I think adding "ccflags-y += -fno-stack-protector" is wrong. The other thing is, it is a variation of the --compat-kernel (downgrading kernel) theme... Mostly I am against that, so I have intentionally removed that option in mine: in my opinion the driver code should work with current, and do not require downgrading. (hence I have different branches for what is "current" when raspbian and ubuntu differ)

Drizzt321, 2021-03-12:
With the 6 mic array, I was having the same build problem, cloned your repo @HinTak, built and installed. Unfortunately I'm getting `bcm2835-i2s fe203000.i2s: I2S SYNC error!` messages every time I try and use `arecord test.wav`. When I try and play any audio, such as `mpg123 test.mp3` which is an mp3 file, and the audio plays just fine in my headphones through the headphone jack. 

Using latest Raspbian kernel, `Linux darkcroft 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l GNU/Linux` from a fresh install. Saw some other tickets, such as #251 #253. Any other thoughts? Possible fixes? Should I just go ahead and downgrade to a previous kernel?

HinTak, 2021-03-12:
@Drizzt321 - I have the 6-mic array, and the card works reasonably okay for recording against either 32-bit raspbian or 64-bit ubuntu (I just swap the SD card - same pi + device), the latter unfortunately needing a slightly modified kernel ( https://github.com/HinTak/RaspberryPi-Dev/releases/ ). Do you actually listen to the recorded files? They come out fine despite the occasional I2S SYNC error messages.

The I2S SYNC error is generic to a somewhat unhappy sound card (see discussion at https://github.com/raspberrypi/linux/issues/3580 ) and not specific to the respeaker ; and they were there at 4.19.x kernels too, as I wrote in https://github.com/raspberrypi/linux/issues/3580 , so downgrading won't help. #251 is serious but does not affect 32-bit raspbian, so you can stay with that; I don't use my 6-mics often enough to be affected by / or notice anything like #253 . For the occasional recordings I make with the 6-mic array, it works despite some syc error messages. (caveate issue #251 affecting 64-bit raspbian and 32-bit/64-bit ubuntu; for the latter I have a workaround).

Drizzt321, 2021-03-12:
@HinTak I'm already using 32-bit Raspbian. Using the 2021-01-11-raspios-buster-armhf-lite.zip image, downloaded from https://www.raspberrypi.org/software/operating-systems/#raspberry-pi-os-32-bit. Unless I'm missing something and turns out that that isn't the proper 32-bit version. 

I tried it with a 2nd Pi 4, and an old Pi 2, but no luck, same issue. Could it be the ribbon cable somehow has a break somewhere that I can't see? I'm going to try the separate i2c headers on the side of the board next, just in case, but I don't hold out much hope. Any idea what could be wrong?

HinTak, 2021-03-13:
I have intentionally removed the `--compat-kernel` option from mine. I don't support/recommend downgrading.

Drizzt321, 2021-03-14:
@HinTak So using your branch with the latest kernel without `--compat-kernel`, I'm having the problem. Nothing gets recorded. Using the official branch with `--compcat-kernel` it works fine.

HinTak, 2021-03-14:
@Drizzt321 raspbian has moved to 5.10 in February so `--compat-kernel` will break the next time you do software upgrades. It just isn't a good option in the long term. (btw, I made the 4.19 to 5.4 change in the "official" branch too - the situation after 5.4 is complicated: raspbian moved to 5.10, while ubuntu lts stays at 5.4 (and will do so for another few years) , while ubuntu rolling is on 5.8 at the moment. Hence I have 4 branches. My "v5.5" branch is basically what seeed studio staff merged into the "official" repo)

Drizzt321, 2021-03-16:
@HinTak I had the 5.10 kernel, but when I used `--compat-kernel` the seeed install script automatically downgraded the kernel. So I'm aware of all that, and find it unfortunate, but I can live it it. For a while at least. 

Unfortunately with your branch, the mic doesn't work for me. Despite the same I2S error, with the official branch and the `--compat-kernel` (with the older kernel) it works.

HinTak, 2021-03-16:
@Drizzt321 "doesn't work..." isn't helping... Was there any error messages? If you have simply downgrade rather than wipe / re-install, you can find old syslogs from booting the newer kernels in the past. Look under /var/log/message* . If it is not too large, consider zipping the whole of it and attach here, with "sudo tar -czpvf alllogs.tgz /var/log"

Drizzt321, 2021-03-16:
The error is the one I gave above in https://github.com/respeaker/seeed-voicecard/issues/285#issuecomment-797331283. 

`bcm2835-i2s fe203000.i2s: I2S SYNC error!`

And then then the output .wav file from `arecord test.wav` always the same size (if I recall it was 44Kb, could recall incorrectly) that never had any audio in it at all. 

Unfortunately that's the only error I saw in messages, dmesg, debug.log.

HinTak, 2021-03-16:
@Drizzt321 also a thought - the mixer setting? There is a recording volume there. There were some change in that recently.

Drizzt321, 2021-03-16:
@HinTak Hmm...good question. I'll try and find some time tonight/tomorrow to use another mSD and redo from scratch with your branch.

HinTak, 2021-03-16:
There is a "make DEBUG=1" setting for enabling some debug code, but the installation process with that is a bit complicated afterwards, and not useful for somebody who is not sitting in front of the machine...

HinTak, 2021-03-16:
Yes, I realize arecord dumps from memory on finish so the file size is zero under afterwards.

AIWintermuteAI, 2021-08-06:
Closing the issue due to the lack of activity. 
We're aware of issues with 64-bit kernel, the discussion of that particular problem will continue in https://github.com/respeaker/seeed-voicecard/issues/251

closed 2021-08-06 by AIWintermuteAI

lxne, 2022-12-20:
> @hackebike btw, the 4-mics and 6-mics devices (not the 2-mics @cl84 has) are affected by the crash on the 64-bit OS - see my comment -

Hi HinTak, thank you so much for your awesome work!
I will donate again. Without you these seeed products would be trash.
One question do you know if the the mentioned crash is happening with both 4-mics versions (the linear and the square one)?

Best wishes,
Alex

thetravellor, 2023-11-08:
The 6.5 branch installs and works with the current 64 bit bullseye kernel for raspbian 11

Linux openvpn 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux