respeaker/seeed-voicecard #286: The seeed-2mic-voicecard card doesn't show in output of "aplay -l"

antoszy, 2021-02-06:
I've installed the card according to the procedure in README.
Running: "sudo ./install.sh" gave me some compilation errors so I installed by running: 
`sudo ./install.sh --compat-kernel`
Than I rebooted RPI.
Running: `aplay -l` shows:
```
**** List of PLAYBACK Hardware Devices ****
card 0: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
```
The card was working correctly before, but all the problems started after running:
`apt upgrade`
and installing wiringpi.
Reinstalling seeed-voicecard does not solve the problem.

My setup configuration is:
- Raspberry PI zero with wi-fi
- ReSpeaker 2-Mics Pi HAT
- System: Raspbian GNU/Linux 10 (buster)
- Kernel version: 5.4.51+

I'm attaching output of:
- dmesg
[dmesg.txt](https://github.com/respeaker/seeed-voicecard/files/5937289/dmesg.txt)
- seeed-voicecard.log
[seeed-voicecard.log](https://github.com/respeaker/seeed-voicecard/files/5937291/seeed-voicecard.log)
- vcdbg log msg
[vcdbg_log_msg.txt](https://github.com/respeaker/seeed-voicecard/files/5937309/vcdbg_log_msg.txt)

Can you suggest further steps to solve the issue?

HinTak, 2021-02-07:
Hmm, you are the 2nd person who seems to have problems hooking up a pi zero with a 2-mics device - see the others
https://github.com/respeaker/seeed-voicecard/issues/283 - your pi zero is simply not seeing the 2-mics connected. but you vcdbg says you haven't tried loading seeed-2mic-voicecard - have you got such a line in /boot/config.txt ?

antoszy, 2021-02-07:
I checked the /boot/config.txt file and there was no seeed-2mic-voicecard entry. 
I added:
`dtoverlay=seeed-2mic-voicecard`
at the end of the file and rebooted the RPI, but the card is still not visible in `aplay -l`.
After running dmesg I see errors regarding wm8960 card:
```
[   14.752487] wm8960 1-001a: Failed to issue reset
[   14.752735] wm8960: probe of 1-001a failed with error -121
```

HinTak, 2021-02-08:
You seems to have a connectivity problem too - try the i2cdetect command in the other issue. I think you'll get all --, according to your earlier seeed-voice service log.

antoszy, 2021-02-08:
Yes there are "--" on all addresses.
How can I verify if it's an issue with soundcard or RPI?

HinTak, 2021-02-08:
Apparently pi zero with badly soldered gpio header is quite a common problem...

antoszy, 2021-02-17:
Using a new soundcard have solved the problem. @HinTak Thanks for all your help.

closed 2021-02-17 by antoszy

HinTak, 2021-02-17:
Good to hear!

Alex-Beh, 2021-06-16:
Hello, I suddenly faced this issue. My ReSpeaker 4-Mic Array is working previously but right now it didn't show in `aplay -l`. 

This is the output of `aplay -l`  :
```
ubuntu@ubuntu:~/codeblue_ws$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7

```
Thi is the output of `arecord -l` :
```
ubuntu@ubuntu:~/codeblue_ws$ arecord -l
**** List of CAPTURE Hardware Devices ****
card 1: seeed4micvoicec [seeed-4mic-voicecard], device 0: bcm2835-i2s-ac10x-codec0 ac10x-codec.1-003b-0 [bcm2835-i2s-ac10x-codec0 ac10x-codec.1-003b-0]
  Subdevices: 0/1
  Subdevice #0: subdevice #0
```

Looking forwards to the advice....

HinTak, 2021-06-16:
@Alex-Beh it is a different device, and you should open a new issue. That said, the 4-mics device does not have a phone jack, so I cannot see why it should be listed as a  playback device (unlike the 2-mics / 6-mics devices) . So it looks like your 4-mics is behaving as it should, only for  recording?

Alex-Beh, 2021-06-17:
@HinTak Yes you are right. Today the `arecord -L` output is different to the output from yesterday. Now the output is same as what have shown in the README.md . I opened a new issue already. Thanks for the help.

> ubuntu@ubuntu:~/installation$ arecord -L
> null
>     Discard all samples (playback) or generate zero samples (capture)
> samplerate
>     Rate Converter Plugin Using Samplerate Library
> speexrate
>     Rate Converter Plugin Using Speex Resampler
> jack
>     JACK Audio Connection Kit
> oss
>     Open Sound System
> pulse
>     PulseAudio Sound Server
> upmix
>     Plugin for channel upmix (4,6,8)
> vdownmix
>     Plugin for channel downmix (stereo) with a simple spacialization
> default
>     Playback/recording through the PulseAudio sound server
> playback
> ac108
> usbstream:CARD=Headphones
>     bcm2835 Headphones
>     USB Stream Output
> sysdefault:CARD=seeed4micvoicec
>     seeed-4mic-voicecard, bcm2835-i2s-ac10x-codec0 ac10x-codec.1-003b-0
>     Default Audio Device
> dmix:CARD=seeed4micvoicec,DEV=0
>     seeed-4mic-voicecard, bcm2835-i2s-ac10x-codec0 ac10x-codec.1-003b-0
>     Direct sample mixing device
> dsnoop:CARD=seeed4micvoicec,DEV=0
>     seeed-4mic-voicecard, bcm2835-i2s-ac10x-codec0 ac10x-codec.1-003b-0
>     Direct sample snooping device
> hw:CARD=seeed4micvoicec,DEV=0
>     seeed-4mic-voicecard, bcm2835-i2s-ac10x-codec0 ac10x-codec.1-003b-0
>     Direct hardware device without any conversions
> plughw:CARD=seeed4micvoicec,DEV=0
>     seeed-4mic-voicecard, bcm2835-i2s-ac10x-codec0 ac10x-codec.1-003b-0
>     Hardware device with all software conversions
> usbstream:CARD=seeed4micvoicec
>     seeed-4mic-voicecard
>     USB Stream Output
>

HinTak, 2021-06-17:
@Alex-Beh you realize you were doing `-l` in one case and  `-L` in the other, right? Uppercase L and lowercase l are two different options.