respeaker/seeed-voicecard #287: Build Error with Kernel 5.10.11-v7+ (armv7l) on PI4

MaBauMeBad, 2021-02-07:
`root@raspberrypi:/home/pi/seeed-voicecard# ./install.sh 

### will compile with the latest kernel...

### Install required tool packages
Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian buster InRelease                                                                                                                           
Hit:3 http://packages.microsoft.com/repos/code stable InRelease                                                                                                                           
Hit:4 https://download.docker.com/linux/raspbian buster InRelease                                                                 
Reading package lists... Done                                               
Building dependency tree       
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
dkms is already the newest version (2.6.1-4).
git is already the newest version (1:2.20.1-2+deb10u3).
i2c-tools is already the newest version (4.1-1).
libasound2-plugins is already the newest version (1.1.8-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

### Uninstall previous dkms module

------------------------------
Deleting module version: 0.3
completely from the DKMS tree.
------------------------------
Done.

### Install required kernel package
Reading package lists... Done
Building dependency tree       
Reading state information... Done
raspberrypi-kernel is already the newest version (1.20210201-1).
raspberrypi-kernel-headers is already the newest version (1.20210201-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
W: --force-yes is deprecated, use one of the options starting with --allow instead.

### Install sound card driver

Creating symlink /var/lib/dkms/seeed-voicecard/0.3/source ->
                 /usr/src/seeed-voicecard-0.3

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=5.10.11-v7+ -C /lib/modules/5.10.11-v7+/build M=/var/lib/dkms/seeed-voicecard/0.3/build.....(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.10.11-v7+ (armv7l)
Consult /var/lib/dkms/seeed-voicecard/0.3/build/make.log for more information.
Can't compile with this kernel, aborting
Please try to compile with the option --compat-kernel
`

make.log
`DKMS make.log for seeed-voicecard-0.3 for kernel 5.10.11-v7+ (armv7l)
Sun  7 Feb 16:01:37 GMT 2021
make: Entering directory '/usr/src/linux-headers-5.10.11-v7+'
  CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/wm8960.o
  CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/ac108.o
  CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/ac101.o
  CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.o
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c: In function ‘seeed_voice_card_startup’:
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c:99:9: error: ‘struct snd_soc_pcm_runtime’ has no member named ‘cpu_dai’
  if (rtd->cpu_dai->driver->playback.channels_min) {
         ^~
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c:100:40: error: ‘struct snd_soc_pcm_runtime’ has no member named ‘cpu_dai’
[... 114 lines cut ...]
 #define snd_soc_read               snd_soc_component_read32
                                    ^~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/seeed-voicecard/0.3/build/wm8960.c:619:15: note: in expansion of macro ‘snd_soc_read’
  u16 iface1 = snd_soc_read(codec, WM8960_IFACE1);
               ^~~~~~~~~~~~
cc1: some warnings being treated as errors
/var/lib/dkms/seeed-voicecard/0.3/build/wm8960.c: At top level:
/var/lib/dkms/seeed-voicecard/0.3/build/wm8960.c:1239:3: error: ‘const struct snd_soc_dai_ops’ has no member named ‘digital_mute’
  .digital_mute = wm8960_mute,
   ^~~~~~~~~~~~
/var/lib/dkms/seeed-voicecard/0.3/build/wm8960.c:1239:18: error: initialization of ‘int (*)(struct snd_pcm_substream *, struct snd_soc_dai *)’ from incompatible pointer type ‘int (*)(struct snd_soc_dai *, int)’ [-Werror=incompatible-pointer-types]
  .digital_mute = wm8960_mute,
                  ^~~~~~~~~~~
/var/lib/dkms/seeed-voicecard/0.3/build/wm8960.c:1239:18: note: (near initialization for ‘wm8960_dai_ops.prepare’)
cc1: some warnings being treated as errors
make[1]: *** [scripts/Makefile.build:279: /var/lib/dkms/seeed-voicecard/0.3/build/wm8960.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make[1]: *** [scripts/Makefile.build:279: /var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.o] Error 1
make[1]: *** [scripts/Makefile.build:279: /var/lib/dkms/seeed-voicecard/0.3/build/ac101.o] Error 1
make[1]: *** [scripts/Makefile.build:279: /var/lib/dkms/seeed-voicecard/0.3/build/ac108.o] Error 1
make: *** [Makefile:1808: /var/lib/dkms/seeed-voicecard/0.3/build] Error 2
make: Leaving directory '/usr/src/linux-headers-5.10.11-v7+'
`

System got just new Updates. Did also a fresh clone of the repo - still failing

HinTak, 2021-02-07:
See the other issue https://github.com/respeaker/seeed-voicecard/issues/285 about v5.10 support . Basically you need my repo. Click donate if you feel generous.

uselessgames, 2021-02-12:
I was having the same issue and running the install with `--compat-kernel` fixed it all.

HinTak, 2021-02-12:
`--compat-kernel` is dependent on what /when you downloaded your pi image, and should be now broken for most cases. Count yourself lucky if it works for you, for now.

MaBauMeBad, 2021-02-12:
Hi,

Thanks will try #285 . Maybe tomorrow, will see if i got the time

janiluuk, 2021-04-21:
So the official company does not seem to care much about their products working on new kernels?

Napsty, 2021-06-15:
Can confirm, this isn't working anymore. Brand new Raspberry Pi Zero with Raspbian GNU/Linux 10 (buster) (Kernel 5.10.17+).

```
Building module:
cleaning build area.....
make -j1 KERNELRELEASE=5.10.17+ -C /lib/modules/5.10.17+/build M=/var/lib/dkms/seeed-voicecard/0.3/build...............(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.10.17+ (armv6l)
Consult /var/lib/dkms/seeed-voicecard/0.3/build/make.log for more information.
Can't compile with this kernel, aborting
Please try to compile with the option --compat-kernel
```

HinTak, 2021-06-15:
You need https://github.com/HinTak/seeed-voicecard . Please click donate if it works for you.

HinTak, 2021-06-15:
@Napsty Btw, since you mentioned pi zero, be aware that there is a power supply issue using the pi zero w with some respeaker devices: https://github.com/respeaker/seeed-voicecard/issues/292

ericfont, 2021-06-16:
I've got the same issue on RPi4 building on patchbox 5.10.17-v7l+ armv7l

I will try HinTak's repo.

Napsty, 2021-06-16:
It works with @HinTak 's repo. Thanks for this. 
At ReSpeaker: You should really act on this, people are buying your hardware and expecting it to work on current Raspberry Pi's.

uselessgames, 2021-06-16:
The company i work for is in contact with the engineers at Seeed as we are going to be purchasing a large amount of their microphones. With the condition that they fix the kernel issue. 

From what I know they stated to use HinTak's repo, as the official code to use (also reference [seeed wiki](https://wiki.seeedstudio.com/ReSpeaker_2_Mics_Pi_HAT_Raspberry/#driver-installation-and-configuration)). The question remains why hasnt seeed officially merged the code. 

It seems to me that HinTak wants to be compensated (based on all the responses asking for donations). @HinTak care to weigh in on this? why haven't you created a PR to merge your fix for current kernels? Having it from the source would cure everyone's problems.

Napsty, 2021-06-16:
> It seems to me that HinTak wants to be compensated

If he's not related or involved in Seeed, then I understand this fully. Why do all the work a company is supposed to do and not get compensated? I totally get it.

But yes, a PR would make most sense.

HinTak, 2021-06-16:
@Napsty @uselessgames I wrote the 5.10 update (in fact I did the 4.19->5.4 change too, which have indeed been merged) and spent about 2 weeks solid on it; I don't work for Seeed Studio, just another regular user. The code is there for others to use. It is not my responsibility to do anything more than I am happy to. I am still waiting on seeed studio staffer to fix #251 myself. Yes, being compensated would be good, but the code is posted in the open and free for others to use.

HinTak, 2021-06-16:
@Napsty @uselessgames you should scroll back the commit history
https://github.com/respeaker/seeed-voicecard/commits/master and see two pages of updates from me the last time I did 4.19->5.4... To be honest seeed studio staff are quite absent .

HinTak, 2021-06-16:
@Napsty @uselessgames , to be fair, after the last bunch of 4.19->5.4 update, I got a $50 US dollar voucher from Seeed Studio as a "thank you" - which I used to get two other seeed studio devices I don't have. (hence they all work well with my branch). But what would be really helpful would be to just send me the documentation for the ac101/ac108 privately "off-the-record" too, besides actually compensating for my time properly.

The ac101/ac108 documentation requires registration and a declaration statement of what product it would be used, etc at their suppliers' web site to obtain.

I'd rather **give seeed studio staff $50 US** to get at the official supplier's documents/code samples instead.

HinTak, 2021-06-16:
See also the closed #213 and #203 about the chip supplier's.

HinTak, 2021-06-16:
@uselessgames btw, the seeed studio folks say they were going to have a v2 hardware with different chips (I think it was mentioned in #251 ?), so you might want to take that into consideration for large purchase and also if your company wants to depend on it...

I didn't know my repo is referenced in the wiki :) . The effort after 5.4 was quite substantial, and included some filing upstream at the raspberrypi folks and alsa-devel... Anyway, it now works okay across the different devices against current 32-bit raspbian. If you are using 64-bit or ubuntu, you need to read #251 . (I gave a workaround for ubuntu there too)

AIWintermuteAI, 2021-06-29:
After merging commits from Hintak's branch, driver installation can be successfully completed on latest Raspberry Pi OS image:
5.10.17-v7l+
[image]
Closing the issue. Feel free to re-open if issue persists.

closed 2021-06-29 by AIWintermuteAI

farnoosh22, 2023-10-02:
build error with kernel 5.10.103+ armv7l on pi4 :(
How can I fix that? Please help.