respeaker/seeed-voicecard #291: Difficulties recording with pulseaudio

baileyji, 2021-04-08:
Hi Folks,
   I'm seeing a lot of instability with what I understand to be to most current branch (@HinTak v5.9) on the RPi 4 stock image from  (2021-03-04 kernel 5.10.17-v7l+) and looking for advice on how to debug and figure out where/if there is an issue.

I've installed per the readme, 

> cd seeed-voicecard
> sudo ./install.sh
> sudo reboot

and not done any of the steps in the pulseaudio readme as I don't see it mentioned more broadly and pulse seems to be autodiscovering ok.

For instance capture with ffmpeg causes pulseaudio to die after a short time (even shorter with 48kHz)
> `ffmpeg -v 1000 -f pulse -sample_rate 44100 -channels 4 -c:a pcm_s32le -i alsa_input.platform-soc_sound.multichannel-input  -ac 1 -f pulse -channels 1 -device 4mic.sink "reseeed"`

The journal shows: 

> Apr 08 10:56:43 cloudlight pulseaudio[16457]: W: [alsa-source-bcm2835-i2s-ac10x-codec0 ac10x-codec.1-003b-0] ratelimit.c: 364 events suppressed
Apr 08 10:56:43 cloudlight pulseaudio[16457]: W: [alsa-source-bcm2835-i2s-ac10x-codec0 ac10x-codec.1-003b-0] asyncq.c: q overrun, queuing locally
Apr 08 10:56:43 cloudlight pulseaudio[16457]: W: [alsa-source-bcm2835-i2s-ac10x-codec0 ac10x-codec.1-003b-0] asyncq.c: q overrun, queuing locally
Apr 08 10:57:06 cloudlight pulseaudio[16457]: W: [pulseaudio] pstream.c: Received invalid frame size: 0
Apr 08 10:57:06 cloudlight pulseaudio[16457]: E: [pulseaudio] pstream.c: Assertion 're->data || re->memblock' failed at pulsecore/pstream.c:862, function do_read(). Aborting.   

Using arecord to pipe data to ffmpeg works well and both 
> `arecord -D hw:CARD=seeed4micvoicec,DEV=0 --channels=4 --format=S32_LE --rate=48000 -v > \dev\null`

results in occasional buffer overruns

> overrun!!! (at least 1321.361 ms long)
> Status:
>   state       : XRUN
>   trigger_time: 690281.230847
>   tstamp      : 0.000000
>   delay       : 0
>   avail       : 22052
>   avail_max   : 22052

> `parec -v --rate=48000 --format=s32le --channels=4 -d alsa_input.platform-soc_sound.multichannel-input > /dev/null` 
will run forever and doesn't complain

From this I suspect that the problem lies outside of the driver and either in the configuration of pulseaudio or in ffmpeg's use of pulse but I'm not certain. Happy to run debugging commands.  

> ➜  ~ pactl list sources
> Source #0
> 	State: SUSPENDED
> 	Name: alsa_input.platform-soc_sound.multichannel-input
> 	Description: Built-in Audio Multichannel
> 	Driver: module-alsa-card.c
> 	Sample Specification: s32le 4ch 48000Hz
> 	Channel Map: front-left,front-right,rear-left,rear-right
> 	Owner Module: 8
> 	Mute: no
> 	Volume: front-left: 65536 / 100% / 0.00 dB,   front-right: 65536 / 100% / 0.00 dB,   rear-left: 65536 / 100% / 0.00 dB,   rear-right: 65536 / 100% / 0.00 dB
> 	        balance 0.00
> 	Base Volume: 65536 / 100% / 0.00 dB
> 	Monitor of Sink: n/a
> 	Latency: 0 usec, configured 0 usec
> 	Flags: HARDWARE DECIBEL_VOLUME LATENCY 
> 	Properties:
> 		alsa.resolution_bits = "32"
> 		device.api = "alsa"
> 		device.class = "sound"
> 		alsa.class = "generic"
> 		alsa.subclass = "generic-mix"
> 		alsa.name = "bcm2835-i2s-ac10x-codec0 ac10x-codec.1-003b-0"
> 		alsa.id = "bcm2835-i2s-ac10x-codec0 ac10x-codec.1-003b-0"
> 		alsa.subdevice = "0"
> 		alsa.subdevice_name = "subdevice #0"
> 		alsa.device = "0"
> 		alsa.card = "2"
> 		alsa.card_name = "seeed-4mic-voicecard"
> 		alsa.long_card_name = "seeed-4mic-voicecard"
> 		alsa.driver_name = "snd_soc_seeed_voicecard"
> 		device.bus_path = "platform-soc:sound"
> 		sysfs.path = "/devices/platform/soc/soc:sound/sound/card2"
> 		device.form_factor = "internal"
> 		device.string = "hw:2"
> 		device.buffering.buffer_size = "42304"
> 		device.buffering.fragment_size = "10576"
> 		device.access_mode = "mmap"
> 		device.profile.name = "multichannel-input"
> 		device.profile.description = "Multichannel"
> 		device.description = "Built-in Audio Multichannel"
> 		module-udev-detect.discovered = "1"
> 		device.icon_name = "audio-card"
> 	Ports:
> 		multichannel-input: Multichannel Input (priority: 0)
> 	Active Port: multichannel-input
> 	Formats:
> 		pcm

> ➜  ~ arecord -L                                                                                       
> null
>     Discard all samples (playback) or generate zero samples (capture)
> jack
>     JACK Audio Connection Kit
> pulse
>     PulseAudio Sound Server
> default
>     Playback/recording through the PulseAudio sound server
> playback
> ac108
> usbstream:CARD=b1
>     bcm2835 HDMI 1
>     USB Stream Output
> usbstream:CARD=Headphones
>     bcm2835 Headphones
>     USB Stream Output
> sysdefault:CARD=seeed4micvoicec
>     seeed-4mic-voicecard, bcm2835-i2s-ac10x-codec0 ac10x-codec.1-003b-0
>     Default Audio Device
> dmix:CARD=seeed4micvoicec,DEV=0
>     seeed-4mic-voicecard, bcm2835-i2s-ac10x-codec0 ac10x-codec.1-003b-0
>     Direct sample mixing device
> dsnoop:CARD=seeed4micvoicec,DEV=0
>     seeed-4mic-voicecard, bcm2835-i2s-ac10x-codec0 ac10x-codec.1-003b-0
>     Direct sample snooping device
> hw:CARD=seeed4micvoicec,DEV=0
>     seeed-4mic-voicecard, bcm2835-i2s-ac10x-codec0 ac10x-codec.1-003b-0
>     Direct hardware device without any conversions
> plughw:CARD=seeed4micvoicec,DEV=0
>     seeed-4mic-voicecard, bcm2835-i2s-ac10x-codec0 ac10x-codec.1-003b-0
>     Hardware device with all software conversions
> usbstream:CARD=seeed4micvoicec
>     seeed-4mic-voicecard
>     USB Stream Output

> ➜  ~ arecord --dump-hw-params -D hw:CARD=seeed4micvoicec,DEV=0
> Recording WAVE 'stdin' : Unsigned 8 bit, Rate 8000 Hz, Mono
> HW Params of device "hw:CARD=seeed4micvoicec,DEV=0":
> ACCESS:  MMAP_INTERLEAVED RW_INTERLEAVED
> FORMAT:  S32_LE
> SUBFORMAT:  STD
> SAMPLE_BITS: 32
> FRAME_BITS: 128
> CHANNELS: 4
> RATE: [8000 48000]
> PERIOD_TIME: (333 2048000]
> PERIOD_SIZE: [16 16384]
> PERIOD_BYTES: [256 262144]
> PERIODS: [2 2048]
> BUFFER_TIME: (666 4096000]
> BUFFER_SIZE: [32 32768]
> BUFFER_BYTES: [512 524288]
> TICK_TIME: ALL
> arecord: set_params:1339: Sample format non available
> Available formats:
> - S32_LE

AIWintermuteAI, 2021-08-02:
Hi @baileyji !
I tested the commands you listed on fresh install of Raspbian Pi OS (Desktop). 

My setup:
Raspberry Pi 4 1 Gb
5.10.17-v7l+ kernel
https://github.com/respeaker/seeed-voicecard/commit/41a09a3b67dc476b9ae2111d319e6941c3d2f689
reSpeaker 4-mic hat for raspberry pi

Results:
```
pi@raspberrypi:~ $ ffmpeg -v 1000 -f pulse -sample_rate 44100 -channels 4 -c:a pcm_s32le -i alsa_input.platform-soc_sound.multichannel-input -ac 1 -f pulse -channels 1 -device 4mic.sink "reseeed"
ffmpeg version 4.1.6-1~deb10u1+rpt1 Copyright (c) 2000-2020 the FFmpeg developers
  built with gcc 8 (Raspbian 8.3.0-6+rpi1)
  configuration: --prefix=/usr --extra-version='1~deb10u1+rpt1' --toolchain=hardened --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-omx-rpi --enable-mmal --enable-neon --enable-rpi --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared --libdir=/usr/lib/arm-linux-gnueabihf --cpu=arm1176jzf-s --arch=arm
  WARNING: library configuration mismatch
  avutil      configuration: --prefix=/usr --extra-version='1~deb10u1+rpt1' --toolchain=hardened --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-omx-rpi --enable-mmal --enable-neon --enable-rpi --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --libdir=/usr/lib/arm-linux-gnueabihf/neon/vfp --cpu=cortex-a7 --arch=armv6t2 --disable-thumb --enable-shared --disable-doc --disable-programs
  avcodec     configuration: --prefix=/usr --extra-version='1~deb10u1+rpt1' --toolchain=hardened --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-omx-rpi --enable-mmal --enable-neon --enable-rpi --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --libdir=/usr/lib/arm-linux-gnueabihf/neon/vfp --cpu=cortex-a7 --arch=armv6t2 --disable-thumb --enable-shared --disable-doc --disable-programs
  avformat    configuration: --prefix=/usr --extra-version='1~deb10u1+rpt1' --toolchain=hardened --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libv [... cut ...]
Connection established.
Stream successfully created.
Buffer metrics: maxlength=4194304, fragsize=1536000
Using sample spec 's32le 4ch 48000Hz', channel map 'front-left,front-center,front-right,rear-center'.
Connected to device alsa_input.platform-soc_sound.multichannel-input (index: 1, suspended: no).
^CGot signal, exiting.ncy: 4943 usec. 
```
**the above command ran normally until interrupted**

Conclusion:
Cannot reproduce the behavior you described. Perhaps you can share more information about your setup?

AIWintermuteAI, 2021-08-20:
I'll be closing the issue to lack of activity. Feel free to re-open if issue persists.

closed 2021-08-20 by AIWintermuteAI