respeaker/seeed-voicecard #294: Respeaker in Docker Container

piisku78, 2021-04-21:
It would be useful if Seeed could provice a sample Docker image for Raspberry which includes ReSpeaker. This could enable usage of ReSpeaker in Cloud IoT Platforms and developers could build intelligent, scalable IoT solutions. 

I would like to build Azure Edge module which contains 4 array ReSpeaker. For that I need a container module but I cannot get ReSpeaker installation running inside a Docker container. 

---------- in Dockerfile I do like this --------

FROM arm32v7/python:3.7-slim-buster
…
…
RUN apt-get -y install portaudio19-dev python-pyaudio python3-pyaudio
RUN git clone https://github.com/respeaker/seeed-voicecard.git
RUN chmod 777 seeed-voicecard/install.sh
RUN apt-get install -y alsa-utils
dkms
i2c-tools
libasound2-plugins

WORKDIR /app/seeed-voicecard
RUN ./install.sh

------ ERROR while building the Docker container ------

[15/15] RUN ./install.sh:
#19 0.426 Warning: /boot volume not found …
#19 0.430 /boot/overlays not found or not a directory
#19 0.463 dtparam not found
#19 0.463 You may need to run ./ubuntu-prerequisite.sh
#19 0.492 dtoverlay not found
#19 0.492 You may need to run ./ubuntu-prerequisite.sh
#19 0.494 Errors found, exiting.

executor failed running [/bin/bash -c ./install.sh]: exit code: 1

Why install.sh does not work inside a container? What should I do to get this running?

HinTak, 2021-04-21:
Do you expect any kind of hardware device driver to work inside docker?

Disclaimer: I don't work for Seeed Studio.

piisku78, 2021-04-21:
well if it is not possible to run install.sh (I dont know what it exactly does) then I can run the install.sh script directly on the device command line while I do initial setup for the device. But then there must be away to get the audio from ReSpeaker to Docker application. How to do it?

HinTak, 2021-04-21:
Why do you think it should be possible to run install.sh inside a docker container? A container naturally separates and isolates the user from accessing devices. The isolation is what many people use docker for.

It is possible to pass the audio over a "network" link between a container and its physical host. You search for "network audio" on Google if you want to do that. You still need to put the driver on the host, **not** on the container.

Disclaimer: I don't work for Seeed Studio.

closed 2021-04-21 by piisku78

HinTak, 2021-04-21:
@piisku78 if you know of such a docker image, you should post the url. You should be able to run `apt update; apt upgrade` and save, to update the docker image to current!