respeaker/seeed-voicecard #295: headphones after reboot muted

FacemanBy, 2021-04-29:
I have just installed the Raspberry.
After that I installed the driver according to the wiki. Had to specify --compat-kernal for this.
Theoretically, the sound card works. But the headphones are muted after every reboot.

I've already run the sudo alsactl --file = / etc / voicecard / wm8960_asound.state store command, but to no avail. 

What else could I do to ensure that the headphones reach the desired volume after a restart?

FacemanBy, 2021-05-01:
Forgot to mention the technical details: 

Raspberry Zero
Buster OS
Kernal 5.4.51+
2mic hat

AIWintermuteAI, 2021-07-26:
Hello, @FacemanBy !

I did some experiments with the following setup:

Raspberry Pi Zero W
5.10.17+ kernel
https://github.com/respeaker/seeed-voicecard/commit/41a09a3b67dc476b9ae2111d319e6941c3d2f689
```
arecord -f cd -Dhw:1 test.wav
aplay -Dhw:0 test.wav
```
reSpeaker 2-Mics Pi Hat for Raspberry Pi

Experiment:
Install https://github.com/respeaker/seeed-voicecard/commit/41a09a3b67dc476b9ae2111d319e6941c3d2f689 commit of seeed-voicecard on fresh Raspberry Pi OS Lite. Record voice and play it, then reboot. Listen to recorded voice with headphones connected to reSpeaker 2-Mics Pi Hat for Raspberry Pi.

Hypothesis:
If problem exists, I wouldn't be able to hear sound after reboot.

Results:
Sound is played at normal volume after reboots, the alsamixer settings are as in the following screenshot(persist between reboots).
[image]

Conclusion:
Cannot reproduce the behavior you described. However, I did run into similar problem on the first try - what was different that time is that 1) desktop image was used 2) originally soundcard drivers were installed and configured for Linear Mic Kit 3) I ran into [this issue](https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=293406&p=1772944) and had to mitigate it by stopping piwiz and deleting it from autostart. On that setup headphones output was muted indeed - but it is hard to say what exactly caused the problem. 

Recommend action:
- Flash clean latest image of Raspberry Pi OS and install latest version of seeed-voicecard drivers
- Try using ```sudo alsactl store 0``` after setting the volume with alsamixer

Here are some threads on stackexchange on similar topics:
https://raspberrypi.stackexchange.com/questions/13666/alsa-volume-lower-after-reboot
https://askubuntu.com/questions/50067/howto-save-alsamixer-settings

AIWintermuteAI, 2021-08-06:
Closing the issue due to the lack of activity.
Feel free to re-open if issue persists.

closed 2021-08-06 by AIWintermuteAI