respeaker/seeed-voicecard #298: ReSpeaker 2-Mic Hat not recognized in pyAudio

ochadenat, 2021-06-14:
Hi,

I followed the instructions from this Blog Post: https://store.createlabz.com/blogs/createlabz-tutorials/respeaker-2-mics-pi-hat-with-raspberry-pi. I am using a Raspberry Zero W.

Everything is working fine (I can play & record with aplay & arecord), However, the card does not seem to be recognized by pyAudio. The function get_host_api_info_by_index)0) is returning the following dictionary:
{'index': 0, 'structVersion': 1, 'type': 7, 'name': 'OSS', **'deviceCount': 0**, 'defaultInputDevice': -1, 'defaultOuputDevice': -1}

HinTak, 2021-06-14:
You are almost certainly using pyaudio wrong. There should be another type (11/12?) with name "Alsa" beside type 7 OSS. Trying starting with and using one of pyaudio's example script which retrieves this info.

HinTak, 2021-06-14:
The script I mentioned above is
https://github.com/jleb/pyaudio/blob/master/test/system_info.py - and you should have some type 8, "alsa" entries. BTW, you are definitely using pyaudio wrong - since `index(0)` is OSS, you probably want `index(1)`. Anyway, start from and use that example script, and adapt as you see fit. That's the official pyaudio way of doing what you want. And throw away whatever you are currently using now.

You might want to check out the other example scripts in that directory. That's a set of "official" pyaudio examples.

ochadenat, 2021-06-16:
Thanks for the pyAudio script. However, It looks like the issue is with my PortAudio installation with Alsa. I can see the line (Alsa ----yes) when I run ./configure. But the PortAudio tests such as pa_devs are failing to find devices.

HinTak, 2021-06-16:
Why are you building portaudio yourself? Raspbian provides it as a package, and they know what they are doing and how to build it correctly. You have got to stop reading random blog posts.

closed 2021-06-16 by ochadenat

HinTak, 2021-06-16:
If you do `sudo apt install pyaudio` it would automatically install the portaudio libraries too. You should not need to build portaudio yourself.

ochadenat, 2021-06-16:
Thanks. Issue fixed by doing `sudo apt install portaudio19-dev `