respeaker/seeed-voicecard #299: ReSpeaker 4 Mic Array issue in Pyaudio

Alex-Beh, 2021-06-17:
Hi, I am using ReSpeaker 4 Mic Array with Raspberry pi 4. I run the following simple python script but I didn't see the ReSpeaker 4 Mic Array in the list. It works previously but don't know why suddenly it din't work now. It didn't show in `pacmd list-sources` also.

> #!/usr/bin/env python3
> import pyaudio
> 
> fields = ('index', 'name', 'defaultSampleRate', 'maxInputChannels')
> 
> pa = pyaudio.PyAudio()
> print("------------------- show_audio_devices --------------------")
> for i in range(pa.get_device_count()):
>     info = pa.get_device_info_by_index(i)
>     print(', '.join("'%s': '%s'" % (k, str(info[k])) for k in fields))
> pa.terminate()

`arecord -L` show the same output in the README.md .

HinTak, 2021-06-17:
Against the 6-mics, here is the result:

```
------------------- show_audio_devices --------------------
'index': '0', 'name': 'seeed-8mic-voicecard: bcm2835-i2s-ac10x-codec0 ac10x-codec0-0 (hw:1,0)', 'defaultSampleRate': '44100.0', 'maxInputChannels': '8'
'index': '1', 'name': 'upmix', 'defaultSampleRate': '44100.0', 'maxInputChannels': '0'
'index': '2', 'name': 'vdownmix', 'defaultSampleRate': '44100.0', 'maxInputChannels': '0'
'index': '3', 'name': 'ac108', 'defaultSampleRate': '48000.0', 'maxInputChannels': '128'
'index': '4', 'name': 'dmixer', 'defaultSampleRate': '48000.0', 'maxInputChannels': '0'
'index': '5', 'name': 'ac101', 'defaultSampleRate': '48000.0', 'maxInputChannels': '128'
'index': '6', 'name': 'dmix', 'defaultSampleRate': '48000.0', 'maxInputChannels': '0'
'index': '7', 'name': 'default', 'defaultSampleRate': '48000.0', 'maxInputChannels': '128'
```

BTW, there are a few indentation errors in your code. It should be:

```
#!/usr/bin/env python3
import pyaudio

fields = ('index', 'name', 'defaultSampleRate', 'maxInputChannels')

pa = pyaudio.PyAudio()
print("------------------- show_audio_devices --------------------")
for i in range(pa.get_device_count()):
    info = pa.get_device_info_by_index(i)
    print(', '.join("'%s': '%s'" % (k, str(info[k])) for k in fields))
pa.terminate()
```

also from another thread, it appears that some people have DIY-compiled/installed pyaudio/ portaudio, which can break  when system libraries are upgraded, etc. You should be able to get pyaudio  and portaudio automatically from raspbian. Here is what `sudo apt search pyaudio` shows on mine:

```
python3-pyaudio/stable,now 0.2.11-1+b2 armhf [installed]
  Python3 bindings for PortAudio v19
```

Note the `[installed]` at the end.

and running `sudo apt search portaudio` (the `automatic` means it was installed as a dependency of another package).

```
...
libportaudio2/stable,now 19.6.0-1+deb10u1 armhf [installed,automatic]
  Portable audio I/O - shared library
...
python3-pyaudio/stable,now 0.2.11-1+b2 armhf [installed]
  Python3 bindings for PortAudio v19
...
```

and make sure you are **not* using pip or pip3 to install pyaudio!!

Alex-Beh, 2021-06-18:
Sorry the  few indentation errors happened when I copy and paste the code.

Your output is correct, the `seeed-8mic-voicecard` should appear inside the list. But it didn't appear for my case. I use the two commands that you shared and the output is the same as you. How do I make sure my pyaudio is not installed from pip3?

HinTak, 2021-06-18:
Try running `i2cdetect -y -a 1` ; you should see some UU's.

Alex-Beh, 2021-06-18:
Hi, this is the output:
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- UU -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
```

Alex-Beh, 2021-06-18:
I managed to see the my microphone in show_audio_devices.

```
'index': '0', 'name': 'bcm2835 Headphones: - (hw:0,0)', 'defaultSampleRate': '44100.0', 'maxInputChannels': '0'
'index': '1', 'name': 'seeed-4mic-voicecard: bcm2835-i2s-ac10x-codec0 ac10x-codec.1-003b-0 (hw:1,0)', 'defaultSampleRate': '44100.0', 'maxInputChannels': '4'
```

The issue happened when I start the microphone from a systemctl service. I switch off the service and then everything back to normal. Will update here when I found the solution.

HinTak, 2021-06-18:
What do you mean by "start the microphone from a systemctl service"? The audio device needs to be accessed by either root or one of the audio group members (of which the user "pi" is one), and many GUI / console login also set up the interactive users to have the correct access rights. If you are doing recording via a service, etc (e.g. From a web server script), you need to make sure the service either has root or "audio" group permissions. (the latter is better).

Read up about "suid" - basically you want your script to be owned by the "audio" group and also set the set-group-id part of the file permission.

AIWintermuteAI, 2021-07-23:
Hello, @Alex-Beh !
I wasn't able to reproduce the bug you're describing.

My setup:
reTerminal (Raspberry Pi CM4 based development board)
5.10.17-v7l+ kernel
41a09a3
reSpeaker 4-mic hat for raspberry pi

Experiment:
run https://github.com/respeaker/4mics_hat/blob/master/recording_examples/get_device_index.py

Output:
```
pi@raspberrypi:~/4mics_hat/recording_examples $ python3 get_device_index.py 
ALSA lib pcm_asym.c:106:(_snd_pcm_asym_open) capture slave is not defined
ALSA lib pcm_dmix.c:1108:(snd_pcm_dmix_open) unable to open slave
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib pcm_a52.c:823:(_snd_pcm_a52_open) a52 is only for playback
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib pcm_hw.c:1822:(_snd_pcm_hw_open) Invalid value for card
ALSA lib pcm_hw.c:1822:(_snd_pcm_hw_open) Invalid value for card
ALSA lib pcm_dmix.c:1108:(snd_pcm_dmix_open) unable to open slave
Input Device id  0  -  seeed-4mic-voicecard: bcm2835-i2s-ac10x-codec0 ac10x-codec0-0 (hw:2,0)
Input Device id  1  -  pulse
Input Device id  2  -  ac108
Input Device id  3  -  default
```
Experiment:
run pacmd list-sources

Output:
```
pi@raspberrypi:~/4mics_hat/recording_examples $ pacmd list-sources
2 source(s) available.
  * index: 0
        name: <alsa_input.platform-soc_sound.multichannel-input>
        driver: <module-alsa-card.c>
        flags: HARDWARE DECIBEL_VOLUME LATENCY 
        state: SUSPENDED
        suspend cause: IDLE
        priority: 9000
        volume: front-left: 65536 / 100% / 0.00 dB,   front-right: 65536 / 100% / 0.00 dB,   rear-left: 65536 / 100% / 0.00 dB,   rear-right: 65536 / 100% / 0.00 dB
                balance 0.00
        base volume: 65536 / 100% / 0.00 dB
        volume steps: 65537
        muted: no
        current latency: 0.00 ms
        max rewind: 0 KiB
        sample spec: s32le 4ch 48000Hz
        channel map: front-left,front-right,rear-left,rear-right
                     Surround 4.0
[... 42 lines cut ...]
        suspend cause: IDLE
        priority: 1000
        volume: front-left: 65536 / 100% / 0.00 dB,   front-right: 65536 / 100% / 0.00 dB
                balance 0.00
        base volume: 65536 / 100% / 0.00 dB
        volume steps: 65537
        muted: no
        current latency: 0.00 ms
        max rewind: 344 KiB
        sample spec: s16le 2ch 44100Hz
        channel map: front-left,front-right
                     Stereo
        used by: 0
        linked by: 0
        configured latency: 0.00 ms; range is 0.50 .. 2000.00 ms
        monitor_of: 0
        module: 16
        properties:
                device.description = "Monitor of Dummy Output"
                device.class = "monitor"
                device.icon_name = "audio-input-microphone"
```

Conclusion:
Cannot reproduce the behavior you described. Perhaps you can share more information about your setup?

Alex-Beh, 2021-07-26:
@Alex-Beh Basically the issue happened when I add my python script which using the microphone into systemctl service. I will see what I can share for the systemctl service setup during the weekend.

AIWintermuteAI, 2021-08-06:
Hi @Alex-Beh  !
May I inquiry, has the issue been resolved?

AIWintermuteAI, 2021-08-13:
Closing the issue due to inactivity.

closed 2021-08-13 by AIWintermuteAI