respeaker/seeed-voicecard #300: project dead ?

haresotto, 2021-06-18:
Is this project dead ? Did you abandon or fail your community/customers ?
The last commit is 9 months old, It is time to give birth...

Maybe you should hire HinTak, who seems to be capable and willing to contribute ( and really, REALLY, seems to want to make money ;-) )

HinTak, 2021-06-18:
I'd summarize my position in the 
Comment  https://github.com/respeaker/seeed-voicecard/issues/287#issuecomment-862525015
, and the ones before and after. 

Seeed Studio staff seem to be more busy over at
https://github.com/Seeed-Studio . I have never figured out why this set of repositories is separate - maybe just historical.

haresotto, 2021-06-18:
@HinTak : English is not my mother tong, so I prefer to precise that the last part of my last sentence was a joke, not an attack.
That apart, I am getting frustrated by the lack of support. I bought many products ad waited enough in my opinion.
If  staff is busy this seems fine and sane to because it means some people have work.

But come on, nearly one year without updates, while one person can ( or seems to be able to) do the job, that seems illogical.
I read that this code will hardly if not never be pushed to linux kernel sources, which means that one day this hardaware will not be supported. Has this day already happened :-( ?

HinTak, 2021-06-18:
@haresotto how/whether seeed studio staff spend their time, and on which projects, is their policy, so I'd not try to guess. I checked that the change made to the 2-mic part of their wiki (in the "Seeed-Studio/wiki..." repo) to refer to my repo was made by a staff in May 1 / 30 April as a bulk update, and curiously not to the 4-mic/6-mic, so I am tempted to file an issue over there and ask hem whether they meant it, or was it a mistake... maybe you can go ahead over there and ask. As for the process of pushing to kernel sources, I have gone through that process for a different part of the kernel, so I say with some confidence about this: in general, the kernel requires that person who makes the push to "signed-off" on it explicitly; this, among other consequences, means the person must be willing to take legal responsibility of that code if and when its copyrights, "intellectual property", or "ownership" status is ever challenged or being argued. That generally means the person doing the push must be a major owner/author of the code. As I commented elsewhere, considering that the chip suppliers' are not releasing the docs/samples codes under their own web site freely, it is quite unwise for another person to try to push that upstream on their behalf. I certainly do not want to take that legal responsibility for them.

AIWintermuteAI, 2021-06-29:
Hi, there! Dmitry from Seeed Studio STU here.
We have never abandoned the project - reSpeaker is one of the most-welcome Seeed Studio's projects and we continue to support customers to solve the issues they might be having and successfully integrate it in their products/prototypes.
Having said that, since a lot of customer support was focused on direct interaction with large(r) customers, Wiki and Github repository didn't receive enough attention. We'll be gradually updating Wiki and resolving open issues on Github.

Commits from Hintak's fork were merged and we thank @HinTak for his contribution to open-source software.
Closing this issue now as it was resolved.

closed 2021-06-29 by AIWintermuteAI