respeaker/seeed-voicecard #301: Capture channels disorder

hengsoo, 2021-07-08:
It all works but the captured channels are not in order.
It is inconsistent for each recording. The channels order are 1234, and sometimes 1278 or 5678.

Hardware: ReSpeaker 4-Mic Linear Array Kit for Raspberry Pi
Driver Version: latest master
OS Kernel: 5.10.17-v7l+ Raspi 4
```
print("* recording")
 
frames = []
 
for i in range(0, int(RESPEAKER_RATE / CHUNK * RECORD_SECONDS)):
    data = stream.read(CHUNK)
    frames.append(data)
 
print("* done recording")
```

HinTak, 2021-07-08:
FWIW, this has be reported multiple times, without resolution. Search for "channel swap" or something in the closed issues, and read them. It would be good to have an open issue collecting the back references, so if you find back issues, please collect and link them here. 

Disclaimer: I don't work for Seeed Studio. Just another regular user.

hengsoo, 2021-07-08:
Thank you for your info. Guess I will look into the ALSA overrun issue in #118 and check if I have any problems with that first.

Related issues/comments :
**intermittently incorrect channel order** #145 
**Random noise on one channel after a while** #267  [The related comment](https://github.com/respeaker/seeed-voicecard/issues/267#issuecomment-733275411)

AIWintermuteAI, 2021-07-23:
Hello, @hengsoo !
I wasn't able to reproduce the bug you're describing. 

My setup:
reTerminal (Raspberry Pi CM4 based development board)
5.10.17-v7l+ kernel
https://github.com/respeaker/seeed-voicecard/commit/41a09a3b67dc476b9ae2111d319e6941c3d2f689
https://github.com/respeaker/4mics_hat/blob/master/recording_examples/record.py with channels modified to 4 and length modified to 10 seconds
reSpeaker 4-mic hat for raspberry pi

Experiment:
Play music from a speaker on reTerminal's right, block two microphones on the left. 
[image]

Hypothesis:
If channel order is inconsistent for every recording, I'd see sound and silence on different channels for different recordings when examining sound in audacity. If channel order is consistent for every recording, I'd see sound and silence on the same channels for different recordings.

Results:
For all three recordings, the sound is on the channels 1 and 2, the silence is on channels 3 and 4.
Recordings can be downloaded from shared Google Drive folder below:
https://drive.google.com/drive/folders/1HNH3TyNPhryZSPb7ZsHp6IkxutRJb04p?usp=sharing

Conclusion:
Cannot reproduce the behavior you described. Perhaps you can share more information about your setup?

hengsoo, 2021-07-23:
> Hello, @hengsoo !
> I wasn't able to reproduce the bug you're describing.
> 
> My setup:
> reTerminal (Raspberry Pi CM4 based development board)
> 5.10.17-v7l+ kernel
> [41a09a3](https://github.com/respeaker/seeed-voicecard/commit/41a09a3b67dc476b9ae2111d319e6941c3d2f689)
> https://github.com/respeaker/4mics_hat/blob/master/recording_examples/record.py with channels modified to 4 and length modified to 10 seconds
> reSpeaker 4-mic hat for raspberry pi
> 
> Experiment:
> Play music from a speaker on reTerminal's right, block two microphones on the left.
> [image]
> 
> Hypothesis:
> If channel order is inconsistent for every recording, I'd see sound and silence on different channels for different recordings when examining sound in audacity. If channel order is consistent for every recording, I'd see sound and silence on the same channels for different recordings.
> 
> Results:
> For all three recordings, the sound is on the channels 1 and 2, the silence is on channels 3 and 4.
> Recordings can be downloaded from shared Google Drive folder below:
> https://drive.google.com/drive/folders/1HNH3TyNPhryZSPb7ZsHp6IkxutRJb04p?usp=sharing
> 
> Conclusion:
> Cannot reproduce the behavior you described. Perhaps you can share more information about your setup?

The mic I used is different from the one in your image. Mine is 4-Mic **Linear Array** Kit.
For the setup I just followed the [official guide ](https://wiki.seeedstudio.com/ReSpeaker_4-Mic_Linear_Array_Kit_for_Raspberry_Pi/). 
I have also run `sudo apt-get install libasound2-dev libasound2-plugins` to get the `pyaudio `working.

AIWintermuteAI, 2021-07-23:
>The mic I used is different from the one in your image. Mine is 4-Mic Linear Array Kit.

Oh, right, my bad. I'll test with Linear kit.

>And for the other 4 channels that ain't recording do contain some noises instead of total silence.

Well, the sound isolation of my fingers against mic is not exactly 100% proof. Plus mic might have moved slightly.

AIWintermuteAI, 2021-07-26:
Hello, @hengsoo !

Still not luck reproducing channel disorder with new setup:
Raspberry Pi 4 1 Gb 
5.10.17-v7l+ kernel
41a09a3
https://github.com/respeaker/4mics_hat/blob/master/recording_examples/record.py with channels modified to 8 and length modified to 10 seconds
reSpeaker 4-Mic Linear Array Kit for Raspberry Pi

Here is the picture of setup - the orange circle is the location of the speaker during recording. 
[image]

Same as with experiment before, I blocked two of the microphones on the right of Linear Kit and recorded 3 10 second segments. On all of them, channel 1 and 2 had some little static and channels 3 and 4 had the sound. There is some noise on the recording, possibly has to do with mic location (flat on the table, might be picking vibrations from the environment), but no inconsistencies in channel order.

Here are recording samples:
https://drive.google.com/drive/folders/1HNH3TyNPhryZSPb7ZsHp6IkxutRJb04p?usp=sharing
Under folder "linear kit".

Conclusion:
Cannot reproduce the behavior you described. Perhaps you can share more information about your setup?

AIWintermuteAI, 2021-08-06:
Hi @hengsoo !
May I inquiry, has the issue been resolved?

AIWintermuteAI, 2021-08-13:
Closing the issue due to inactivity.

closed 2021-08-13 by AIWintermuteAI

edwbaker, 2022-10-04:
Still an issue with respeaker 6mic array - latest Raspbian and driver c526066.

zhanglong97, 2023-12-14:
I have the same issue, The channel order changes every time the mic is turned on.
has the issue been resolved?

Hardware: ReSpeaker 4-Mic Linear Array Kit for Raspberry Pi
Driver Version: https://github.com/respeaker/seeed-voicecard/commit/c52606626de050bdad85803d7e427a64cb0cf05c
OS Kernel: 5.10.103-v7+ Raspi 2