respeaker/seeed-voicecard #302: Sounds very tinny

poblabs, 2021-07-10:
I'm unable to control the tone of the sound, like treble and bass. How can I increase the bass so it doesn't sound so tinny?

HinTak, 2021-07-10:
`alsamixer`, turn the volume up...

poblabs, 2021-07-10:
It's up to 100. I need to adjust bass. How?

HinTak, 2021-07-10:
There is a capture volume and a playback volume. For what you are thinking (boosting the bass) you probably want the equalizer part of some some sound app, outside the driver.

Disclaimer: I don't work for Seeed Studio. Just another user.

poblabs, 2021-07-10:
Not sure that's it, I've tried mpc, mpg321 and omxplayer. On identical hardware (not raspberry pi) it sends different. I think what's missing is a little bass or tone control. I don't see that in alsamixer and I think the driver is what controls that ability.

HinTak, 2021-07-11:
Listens to the recordings I made https://github.com/respeaker/seeed-voicecard/issues/289#issuecomment-815313537 with the 4-mics / 6-mics devices. (you have not mentioned which device you are using, btw). If yours is substantially worse than mine, you might have a faulty card as bad as the other person's in that issue.

poblabs, 2021-07-11:
I ended up resolving it using the alsamixer plugin and setting mpd to output to that device.

closed 2021-07-11 by poblabs