respeaker/seeed-voicecard #304: [Bug]: Code doesn’t compile on RetroPie v

StonedEdge, 2021-08-03:
### Describe the bug

Hi guys, 

I’ve tried to compile this via sudo ./install.sh  however I am getting missing kernel header errors that prevent me from compiling (please see relevant log output below). 

I'm using a CM4, running the latest RetroPie image available here, which I believe runs on the kernel 5.4.72-v71+. Has anyone got any tips to get this driver working correctly with the latest version of RetroPie? I’ve created a custom PCB for it, and I’d love to use it with RetroPie if possible. 

I have tried updating to the latest kernel and did see some success with installation, however RetroPie was crashing quite often and not great on the latest kernel. 

### Expected behavior

I expect to be able to install drivers on version 4.5.72. Can install after upgrading via rpi-update and sudo apt upgrade all, however RetroPie crashes. I need to be able to install drivers successfully on 4.5.72. 

### Platform

- Device: Raspberry Pi CM4
- OS: RetroPie - kernel commit 5.4.72

### Relevant log output

```shell
DKMS: add completed.
Error! echo
Your kernel headers for kernel 5.4.51-v7l+ cannot be found at
/lib/modules/5.4.51-v7l+/build or /lib/modules/5.4.51-v7l+/source.
Created symlink /etc/systemd/system/sysinit.target.wants/wm8960-soundcard.service → /lib/systemd/system/wm8960-soundcard.service.
Job for wm8960-soundcard.service failed because the control process exited with error code.
See "systemctl status wm8960-soundcard.service" and "journalctl -xe" for details.
```

AIWintermuteAI, 2021-08-04:
Hi there!
I'm not familiar with RetroPi image, but I'll try finding the time to check this Friday.

StonedEdge, 2021-08-04:
> Hi there!
> I'm not familiar with RetroPi image, but I'll try finding the time to check this Friday.

That’d be great, looking forward to it. I’d love to keep using this DAC, but was considering swapping if it’s too difficult to get it working.

Pillar1989, 2021-08-04:
@StonedEdge  RetroPie use the `5.4.51-v7l+ ` kernel version. This kernel has abandoned by Raspberry pi. Please update to the latest 5.10 version. 
fly, I don't think this issue is even worth discussing. Perhaps it would be more valuable for you to go to RetroPie's GitHub and discuss kernel upgrades than to discuss software support here.

closed 2021-08-04 by Pillar1989

StonedEdge, 2021-08-04:
> @StonedEdge RetroPie use the `5.4.51-v7l+ ` kernel version. This kernel has abandoned by Raspberry pi. Please update to the latest 5.10 version.
> fly, I don't think this issue is even worth [...]

Really? A lot of people use RetroPie and although I agree with you that I should be running it on the 5.10 kernel, it doesn’t work currently and crashes upon updating. I thought it would be worth looking into if it’s such a simple fix.

HinTak, 2021-08-04:
@StonedEdge you might find 
this useful -  https://github.com/HinTak/RaspberryPi-Dev/blob/master/Downgrading-Pi-Kernel.md ; instruction fairly similar for manually switching specific kernel kernels and header packages around. Disclaimer: I don't work for Seeed Studio. Just another user, and the usual "consider clicking donate if you find it useful"...

StonedEdge, 2021-08-04:
> @StonedEdge you might find
> this useful - https://github.com/HinTak/RaspberryPi-Dev/blob/master/Downgrading-Pi-Kernel.md ; instruction fairly similar for manually switching specific kernel kernels [...]

Thank you for your help. I'm a little bit slow with this, but my understanding is that I need to put these lines within the install.sh file? Or can I just execute these before running sudo ./install.sh?

HinTak, 2021-08-04:
@StonedEdge no, you just manually execute the 4 commands, one after the other, **reboot**, then proceed to follow the usual driver install instructions. The 5th command is just for verification.

```
wget -O kernel-headers_armhf.deb http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel-headers_1.20210108-1_armhf.deb
wget -O kernel_armhf.deb http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel_1.20210108-1_armhf.deb
sudo dpkg -i kernel-headers_armhf.deb kernel_armhf.deb
sudo apt-mark hold raspberrypi-kernel-headers raspberrypi-kernel
```

HinTak, 2021-08-04:
@StonedEdge since you are already at 5.4.x, it is not much of a "downgrade" or "upgrade", but just giving you the additional stuff to be able to build the driver.

StonedEdge, 2021-08-04:
> @StonedEdge since you are already at 5.4.x, it is not much of a "downgrade" or "upgrade", but just giving you the additional stuff to be able to build the driver.

Hmm, I just tried executing those commands, then running the script, but same error unfortunately. See below:

`Creating symlink /var/lib/dkms/seeed-voicecard/0.3/source ->
                 /usr/src/seeed-voicecard-0.3

DKMS: add completed.
Error! echo
Your kernel headers for kernel 5.4.72-v7l+ cannot be found at
/lib/modules/5.4.72-v7l+/build or /lib/modules/5.4.72-v7l+/source.
setup git config
git init
Initialized empty Git repository in /etc/voicecard/.git/
git add --all
git commit -m "origin configures"
[master (root-commit) 9a85c24] origin configures
 7 files changed, 1482 insertions(+)
 create mode 100644 ac108_6mic.state
 create mode 100644 ac108_asound.state
 create mode 100644 asound_2mic.conf
 create mode 100644 asound_4mic.conf
 create mode 100644 asound_6mic.conf
 create mode 100644 dkms.conf
 create mode 100644 wm8960_asound.state
Created symlink /etc/systemd/system/sysinit.target.wants/seeed-voicecard.service → /lib/systemd/system/seeed-voicecard.service.`

HinTak, 2021-08-04:
@StonedEdge I need to see what went wrong with the 3rd command (no need to do the first two again, they just download files), it should output piles and piles of stuff. and also the output of `ls -ld /lib/modules/*/build /lib/modules/*/source`.

StonedEdge, 2021-08-04:
Sure, I'm not sure if this is the best way to condense the output of that command since its quite long, but here it is anyway. Btw, the card does install (hooray), but when I try to run "speaker-test" in terminal, I get this response. This seems to be the relevant part (highlighted in bold) but I've put the whole output below anyway to be complete. I've excluded the dtbo adds/removals. 

1) Bus error
speaker-test 1.1.8
Playback device is default
Stream parameters are 48000Hz, S16_LE, 1 channels
Using 16 octaves of pink noise
Rate set to 48000Hz (requested 48000Hz)
Buffer size range from 12000 to 18000
Period size range from 6000 to 6000
Using max buffer size 18000
Periods = 4
Bus error

2) Kernel errors
run-parts: executing /etc/kernel/postinst.d/apt-auto-removal 5.4.83+ /boot/kernel.img
run-parts: executing /etc/kernel/postinst.d/dkms 5.4.83+ /boot/kernel.img
run-parts: executing /etc/kernel/postinst.d/initramfs-tools 5.4.83+ /boot/kernel.img
run-parts: executing /etc/kernel/postinst.d/apt-auto-removal 5.4.83-v7+ /boot/kernel7.img
run-parts: executing /etc/kernel/postinst.d/dkms 5.4.83-v7+ /boot/kernel7.img
run-parts: executing /etc/kernel/postinst.d/initramfs-tools 5.4.83-v7+ /boot/kernel7.img
run-parts: executing /etc/kernel/postinst.d/apt-auto-removal 5.4.83-v7l+ /boot/kernel7l.img
run-parts: executing /etc/kernel/postinst.d/dkms 5.4.83-v7l+ /boot/kernel7l.img
run-parts: executing /etc/kernel/postinst.d/initramfs-tools 5.4.83-v7l+ /boot/kernel7l.img
run-parts: executing /etc/kernel/postinst.d/apt-auto-removal 5.4.83-v8+ /boot/kernel8.img
run-parts: executing /etc/kernel/postinst.d/dkms 5.4.83-v8+ /boot/kernel8.img
/etc/kernel/postinst.d/dkms:
Error! echo
Your kernel headers for kernel 5.4.83-v8+ cannot be found at
/lib/modules/5.4.83-v8+/build or /lib/modules/5.4.83-v8+/source.
Error! echo
Your kernel headers for kernel 5.4.83-v8+ cannot be found at
/lib/modules/5.4.83-v8+/build or /lib/modules/5.4.83-v8+/source.
run-parts: executing /etc/kernel/postinst.d/initramfs-tools 5.4.83-v8+ /boot/kernel8.img

2a)
run-parts: executing /etc/kernel/postrm.d/initramfs-tools 5.4.83+ /boot/kernel.img
run-parts: executing /etc/kernel/postrm.d/initramfs-tools 5.4.83-v7+ /boot/kernel7.img
run-parts: executing /etc/kernel/postrm.d/initramfs-tools 5.4.83-v7l+ /boot/kernel7l.img
run-parts: executing /etc/kernel/postrm.d/initramfs-tools 5.4.83-v8+ /boot/kernel8.img
Setting up raspberrypi-kernel-headers (1.20210108-1) ...
run-parts: executing /etc/kernel/header_postinst.d/dkms 5.4.83+
Job for systemd-modules-load.service failed because the control process exited with error code.
See "systemctl status systemd-modules-load.service" and "journalctl -xe" for details.
Job for systemd-modules-load.service failed because the control process exited with error code.
See "systemctl status systemd-modules-load.service" and "journalctl -xe" for details.
run-parts: executing /etc/kernel/header_postinst.d/dkms 5.4.83-v7+
Job for systemd-modules-load.service failed because the control process exited with error code.
See "systemctl status systemd-modules-load.service" and "journalctl -xe" for details.
Job for systemd-modules-load.service failed because the control process exited with error code.
See "systemctl status systemd-modules-load.service" and "journalctl -xe" for details.
run-parts: executing /etc/kernel/header_postinst.d/dkms 5.4.83-v7l+
run-parts: executing /etc/kernel/header_postinst.d/dkms 5.4.83-v8+
Error! echo
Your kernel headers for kernel 5.4.83-v8+ cannot be found at
/lib/modules/5.4.83-v8+/build or /lib/modules/5.4.83-v8+/source.
Error! echo
Your kernel headers for kernel 5.4.83-v8+ cannot be found at
/lib/modules/5.4.83-v8+/build or /lib/modules/5.4.83-v8+/source.

3) sudo dkms status 
seeed-voicecard, 0.3, 5.4.83+, armv7l: installed
seeed-voicecard, 0.3, 5.4.83-v7+, armv7l: installed
seeed-voicecard, 0.3, 5.4.83-v7l+, armv7l: installed
xpad, 0.4, 5.4.83+, armv7l: installed (original_module exists)
xpad, 0.4, 5.4.83-v7+, armv7l: installed (original_module exists)
xpad, 0.4, 5.4.83-v7l+, armv7l: installed (original_module exists)

HinTak, 2021-08-04:
@StonedEdge according to my notes ( https://github.com/HinTak/RaspberryPi-Dev/blob/master/Raspbian-Kernel-Releases.md ) you want 20201022 if you want specifically 5.4.72 . Modify the first two lines of 4 to do that. Error messages with "v8+" can be ignored (they are for 64-bit kernels). So indeed the driver seems to have installed okay. The bus error I have never seen - likely hardware issue - make sure you have enough power supplied to the module, things are connected up properly for example. Sorry I think that's the limit of what I can offer.

HinTak, 2021-08-04:
@StonedEdge on 2nd thought, I don't know what speaker-test does. If it is not part of seeed studio instructions, maybe you should use something like the "aplay ..." command for testing in the readme instead .

StonedEdge, 2021-08-04:
> @StonedEdge according to my notes ( https://github.com/HinTak/RaspberryPi-Dev/blob/master/Raspbian-Kernel-Releases.md ) you want 20201022 if you want specifically 5.4.72 . Modify the first two lines [...]

I've managed to install it and get some sound out now using aplay Test.wav. However, it only worked once. Then after that I'm getting the following errors:

pi@retropie:~ $ aplay Test.wav
Playing raw data 'Test.wav' : Unsigned 8 bit, Rate 8000 Hz, Mono
ALSA lib dlmisc.c:287:(snd1_dlobj_cache_get) Cannot open shared library /usr/lib/arm-linux-gnueabihf/alsa-lib/libasound_module_rate_samplerate.so ((null): /usr/lib/arm-linux-gnueabihf/alsa-lib/libasound_module_rate_samplerate.so: cannot open shared object file: No such file or directory)
ALSA lib pcm_rate.c:1450:(snd_pcm_rate_open) Cannot find rate converter
aplay: set_params:1403: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  U8
SUBFORMAT:  STD
SAMPLE_BITS: 8
FRAME_BITS: 8
CHANNELS: 1
RATE: 8000
PERIOD_TIME: 125000
PERIOD_SIZE: 1000
PERIOD_BYTES: 1000
PERIODS: 3
BUFFER_TIME: 375000
BUFFER_SIZE: 3000
BUFFER_BYTES: 3000
TICK_TIME: 0

Any idea what would cause this @HinTak?

StonedEdge, 2021-08-04:
Seems to be working so far! Thank you so much!

Just one thing I noticed, does the volume reset upon reboot for this driver? Should alsamixer volume states be saved or do they default to maximum volume on boot? I'm probably going to write a script to fix this anyway, but was just wondering as some other drivers I've used seem to retain the settings upon restarts.

HinTak, 2021-08-04:
@StonedEdge cool. I was going to write that strictly speaking, the asound plugins missing issue is not the driver's responsibility - audio devices (not just Seeed Studio's) typically accepts only certain sampling rates, and if you try to play sound of a different rate to what the device likes, it is up to the middle-man (the generic alsa libs) to convert; also not sure about CM4, but the other Pi's all have on-board audio so the default playback device is not the hat, but the on-pi-board phone jack. Hence the test command in the readme has both rate and devices specified.

As for save/restore volume settings across reboot, supposely yes, it should. I have seen reports here about it not working though... Have a look at the closed reports - I don't recall details.

And ...I don't work for Seeed Studio... So if you find this helpful, please feel free to click on the donate links eg bottom of https://hintak.github.io .

StonedEdge, 2021-08-05:
> @StonedEdge cool. I was going to write that strictly speaking, the asound plugins missing issue is not the driver's responsibility - audio devices (not just Seeed Studio's) typically accepts only [...]

Will be donating shortly! Thanks so much for your help, you made my week and glad that this project can continue forward. Cheers again.

StonedEdge, 2021-10-07:
Hi there,

I'm back as I'm having issues again following the above instructions. Has there been any push requests that may have broken the above steps? Wondering how to fix my issue... sorry to bother you. 

Kind regards

HinTak, 2021-10-07:
The error message about package holds seems harmless and everything looks alright as far as I see. What does `dmesg` says after the reboot?

StonedEdge, 2021-10-08:
Hmm, so it seems to be working overall, but the Pi is lagging quite significantly after installing the driver. 

After I try to play audio from the splashscreen on RetroPie by altering this line in the shell script attached, it basically seems to not want to configure the bluetooth service and I end up getting crashes. Going a bit of topic here, but definitely strange... ugh

https://images-ext-1.discordapp.net/external/lH7C3iNmyZl1AMEYQ4XyN6dzJ87sxF0Ctvy8GWrQvbM/https/media.discordapp.net/attachments/651918229308178468/895692684184920064/unknown.png

HinTak, 2021-10-08:
I am a little confused - does the driver works or not? If you are trying to play sound with unsual player, such as omxplayer as in your screenshot, you can have strange consequences. In particular, omxplayer is pi-specific and use the raspberrypi's video core and firmware to interact with the video and audio hardware, and it goes below the driver, alsa and perhaps even the linux kernel... if I were you, I would avoid using it for unsual usage, and perhaps not for welcome sound at login...

And since it interacts with the hardware more directly, if you experiences crashes from its usage, you should report it to the omxplayer maintainer. Who incidentally, is also one of the core rasberrypi engineers in the rasberrypi foundation!

StonedEdge, 2021-10-09:
The driver installs fine, works well. But for my application (with RetroPie) - it doesn't work at all in practice. 

I can click around in the menus (yes the audio works there) but as soon as I load a game it will just revert back to the emulation station home menu (i.e ROM does not launch correctly). 

I will try to reach out to someone on RetroPie forums to help with my issue as I understand this is not your responsibility and probably out of scope to help me much further - sorry..

HinTak, 2021-10-09:
The current maintaining place is https://github.com/popcornmix/omxplayer (it is forked from another which is no longer maintained) , and as I wrote earlier, popcornmix is one of the pi people; but even  that page says you should be using vlc instead!

I am mostly interested in others' problems with the respeaker because I also own one (and I have no special relationship with Seeed Studio, except they have given me some gift vouchers for contributing bug fixes) so I am okay to try, up to a point.

HinTak, 2021-10-10:
@StonedEdge As far as I understand it (from the Seeed Studio people), upstream (as included in recent kernels) wm8960 is buggy and it is better to use the seeed-studio version of the wm8960 driver code. What may help is actually the dtbo line in config.txt - that was a change I made some time during the last year - before that, the installation process both try to load the dtbo and driver at boot time, and also dynamically probe for and determine which device it has and load on demand. Because load-at-boot cannot be unloaded, I changed it to make it easier to unload, and reload a modified version, to make it easier for trying things out and debugging without rebooting.

If your changes work for you, maybe RetroPie needs to use the driver so early (as you seems to want it at the welcome screen), you can just add that line in config.txt by hand. It is a balance - putting it in config.txt means the driver is loaded earlier, but less flexible (and bad things happens if the driver does not match the device actually connected); probing for which device is connected and loading the corresponding driver, means the driver gets loaded a little later after boot; and as it seems to you in your usage, probably trying to play sound very early after boot does not work. You should be able to tell from `dmesg`, when the driver is loaded - there is a timestamp before it. I did ask for dmesg output earlier...

StonedEdge, 2021-11-05:
Hi @HinTak 

Copying the below dmesg as still getting some errors here, not sure if this helps you debug or not but if you do have some time please feel free to take a look. Cheers.

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.10.63-v7l+ (dom@buildbot) (arm-linux-gnueabihf-gc                                                                             c-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34)                                                                              #1459 SMP Wed Oct 6 16:41:57 BST 2021
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi Compute Module 4 Rev 1.0
[    0.000000] random: fast init done
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001ac00000, s                                                                             ize 320 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id share                                                                             d-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000002fffffff]
[    0.000000]   Normal   empty
[    0.000000]   HighMem  [mem 0x0000000030000000-0x000000007fffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000007fffffff]
[    0.000000] On node 0 totalpages: 504832
[    0.000000]   DMA zone: 1728 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 196608 pages, LIFO batch:63
[    0.000000]   HighMem zone: 308224 pages, LIFO batch:63
[    0.000000] percpu: Embedded 20 pages/cpu s50828 r8192 d22900 u81920
[    0.000000] pcpu-alloc: s50828 r8192 d22900 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 503104
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835                                                                             .enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=HDMI-A-1:800x480M@59 smsc9                                                                             5xx.macaddr=E4:5F:01:3E:B2:F8 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000                                                                             000  console=ttyS0,115200 console=tty1 root=PARTUUID=68da54e0-02 rootfstype=ext4                                                                              elevator=deadline fsck.repair=yes rootwait loglevel=3 consoleblank=0 plymouth.e                                                                             nable=0
[    0.000000] Kernel parameter elevator= does not have any effect anymore.
               Please use sysfs to set IO scheduler for individual devices.
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes,                                                                              linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, li                                                                             near)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x0000000015a00000-0x0000000019a0000                                                                             0] (64MB)
[    0.000000] Memory: 1587372K/2019328K available (10240K kernel code, 1364K rw                                                                             data, 3192K rodata, 2048K init, 891K bss, 104276K reserved, 327680K cma-reserved                                                                             , 1232896K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 34315 entries in 68 pages
[    0.000000] ftrace: allocated 68 pages with 2 groups
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000]  Rude variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jif                                                                             fies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] random: get_random_bytes called from start_kernel+0x3c8/0x59c wit                                                                             h crng_init=1
[    0.000008] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2                                                                             147483647500ns
[    0.000028] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_                                                                             idle_ns: 1911260446275 ns
[    0.000088] bcm2835: system timer (irq = 25)
[    0.000739] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[... cut ...]
[   14.029031] Bluetooth: BNEP filters: protocol multicast
[   14.029043] Bluetooth: BNEP socket layer initialized
[   38.974999] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   40.709100] ICMPv6: process `dhcpcd' is using deprecated sysctl (syscall) net                                                                             .ipv6.neigh.wlan0.retrans_time - use net.ipv6.neigh.wlan0.retrans_time_ms instea                                                                             d

HinTak, 2021-11-05:
The "memory leak" messages are harmless - yes, there is a genuine leak if you load and unload the driver multiple times, but it only affect developers who are modifying the driver code, and need to unload and re-load (a modified versions of it ) multiple times. As in they are still marked as "used", even after you unload the driver, so you lose them and if you do that too often, your available memory decreases. But mostly people only load the driver once, at boot, so you lose track of a few bytes once, and that's the end of the story.

Is there any other problems?

.