respeaker/seeed-voicecard #306: @Pillar1989 I don't think this is the same error as #166 This bug is already fixed and so should not show up any more.

amirush786, 2021-08-05:
@Pillar1989 I don't think this is the same error as #166 This bug is already fixed and so should not show up any more.

The RPi4 (new hardware!) exhibits a different bug, and I have seen this myself (but currently I am out of my home so I cannot investigate it further).

The compilation etc all works well, but somehow the module isn't loaded properly.

When I am back I will send a debug log and everything I can find.

So I suggest for now to un-mark this as duplicated and reopen

_Originally posted by @norbusan in https://github.com/respeaker/seeed-voicecard/issues/168#issuecomment-509863804_

amirush786, 2021-08-05:
I am working with 6 mic tool to collect data for a pose estimation project.
I am not able to
 record any sound from the 6 microphone tool. I tried fixing it but I don't know what's wrong. I have attached the expected and actual output when I am running. Can you please help me with it. website for 6 mic tool
https://wiki.seeedstudio.com/ReSpeaker_6-Mic_Circular_Array_kit_for_Raspberry_Pi/

[image]
[image]
[image]
[image]
[image]
[image]

Can someone please help me fix the issue?

AIWintermuteAI, 2021-08-05:
Hi there, @amirush786 !

To expedite the resolution of problems, we have started using issue forms in this repository. 
Please click on New issue, choose Bug report and then fill in the details mentioned on the form.

Thank you!

closed 2021-08-05 by AIWintermuteAI

HinTak, 2021-08-05:
FWIW, the message looks like the driver is loaded correctly but not binding to the device ie you may have a faulty unit. For that you need to post the boot up message ie do `dmesg > output.txt` and post output.txt.

amirush786, 2021-08-06:
> FWIW, the message looks like the driver is loaded correctly but not binding to the device ie you may have a faulty unit. For that you need to post the boot up message ie do `dmesg > output.txt` and post output.txt.

I will paste that

HinTak, 2021-08-06:
@amirush786 just so we are clear - you are not using the antique 4.19.x kernel, are you? . Anyway I have the Pi4 + 6-mics combo, and it was working with any of the last 5.10.x (current)/5.4.x/4.19.x kernels in May 2021, if you use the latest uptodate raspbian and driver code from here.

HinTak, 2021-08-06:
Each of the last of the 3 kernel series mentioned in https://github.com/HinTak/RaspberryPi-Dev/blob/master/Downgrading-Pi-Kernel.md was explicitly tested around May 2021, only a few months ago. There is no driver code change in the last 3 months and I doubt raspbian has broken since.

Disclaimer: I don't work for Seeed Studio - just another user.

amirush786, 2021-08-06:
> @amirush786 just so we are clear - you are not using the antique 4.19.x kernel, are you? . Anyway I have the Pi4 + 6-mics combo, and it was working with any of the last 5.10.x (current)/5.4.x/4.19.x kernels in May 2021, if you use the latest uptodate raspbian and driver code from here.

Where can I find the driver code and how would I know the version of pi which I am using. 
I am working on a project which is pose estimation. I have to play modulated sound from the speaker which is connected to pi and also record it at the same time from the 6mic tool. I don't know how it should be done. Any leads would be helpful.

HinTak, 2021-08-06:
The version of pi you are using , and the kernel version, are both in the `dmesg` output I asked earlier . And it contains other info too. The latest driver code is https://github.com/respeaker/seeed-voicecard . You *must* update past May 2021 with recent raspbian. If you have not update your os since 2019 (in your post above), you should.

HinTak, 2021-08-07:
@amirush786 if you are not even sure what version of OS you are running, and you have not updated your driver code for a while (since 2019...) , there is one other possibility besides hardware /electrical-connectivity fault: you have had a partial OS upgrade moving off your original installation. Since the older /2019 driver code does not work with current (2020-) kernel, half of the driver (the device tree part) would be loaded but not the other (c/compiled) part. This would looks like the device is present but not working. In any case, `dmesg > output.txt` and posting the output here, as well as `lsmod` (most of the info from `lsmod` is also in dmesg, just more convenient) would be useful to see what's wrong with your setup. As I said , if everything is up to date (may 2021 onwards) pi4 + 6 mics definitely works.

amirush786, 2021-08-07:
I will try doing it with the latest github drivers for the seeed 6 mic tool. Don't know what's the wront let you know with the updates.

amirush786, 2021-08-09:
> @amirush786 if you are not even sure what version of OS you are running, and you have not updated your driver code for a while (since 2019...) , there is one other possibility besides hardware [...]

hello I tried  doing that but the problem is that its not allowing me to play the wave files that I record.
[image]

HinTak, 2021-08-09:
You still need to post the result of `dmesg > output.txt` and `lsmod`. As for the audio file, try copying it to a different computer to play, and/or look at it with audacity.

amirush786, 2021-08-09:
> You still need to post the result of `dmesg > output.txt` and `lsmod`. As for the audio file, try copying it to a different computer to play, and/or look at it with audacity.

How do we post the result to output.txt means I know I syntax but not sure about it. I have to play a FMCW sound from the pi on the speaker example alexa and also record the sound as well at the same time when the sound is playing on speaker.

HinTak, 2021-08-09:
There is a button you can press just below the github comment for attaching files - the process is the same as, or similar to, attaching images, which you already did.

As for general audio question not specific to the respeaker, please ask elsewhere. The original readme (before the recent has a tips about playing while recording, it is something like

arecord .... & ; aplay ...

"command1 & ; command2" is generic for running command1 in background while continue to start running command2. See any unix introductory book. I hate to say this, but if you have trouble understanding the "... & ; " construct, or have never seen this before, you need to go and read some introductory books on unix systems before you try any driver/compilation-related tasks. It is quite basic unix daily-usage stuff.

Disclaimer: I don't work for Seeed Studio...

amirush786, 2021-08-09:
Yeh got it. I know the command but the problem is that it gives the error for ac101 For playing the sound. I tried just playing the wav file which I re order but it doesn't work and gives me ac101 pcm unknown.

HinTak, 2021-08-09:
You still have not posted the saved output from `dmesg > output.txt` (it is quite a large output so I asked you to save to a file and posted that) , and `lsmod`, despite it being requested 3 or 4 times already. You cannot complain that things don't work for you, when you cannot supply the information needed to diagnose the problem... So stop saying things don't work, until you supply both.

Repeating that you cannot play sound is just not going to help, when you cannot supply the information needed.

amirush786, 2021-08-09:
I will post both the things in a day.

amirush786, 2021-08-10:
> You still have not posted the saved output from `dmesg > output.txt` (it is quite a large output so I asked you to save to a file and posted that) , and `lsmod`, despite it being requested 3 or 4 [...]

I have attached the output file.
[lsmod_output.txt](https://github.com/respeaker/seeed-voicecard/files/6958326/lsmod_output.txt)
[output.txt](https://github.com/respeaker/seeed-voicecard/files/6958327/output.txt)

HinTak, 2021-08-10:
According to your dmesg output, you have a "Raspberry Pi 3 Model B Plus Rev 1.3" , i.e. a Pi 3B+, not a Pi 4. So that's the first inaccuracy/problem you have - you don't even know you have a Pi3, and commented wrongly on the Pi4 support thread.

There are some duplicate entries in it, in particular, i2c seems to be registered twice, and some of the audio hardware too. You might check for duplicate entries in your /boot/config.txt , and perhaps start from scratch. The firmware is not up to date, that might help too. And try not to use rpi-update - rpi-update is for getting at "pre-releases". You have enough problem as you are, no need to make your own life more difficult by using pre-releases. "sudo apt update ; sudo apt upgrade" is the command to get your pi up to date to current releases.

You seems to have a lot of usb devices connected - a network device (what's wrong with using the built-in ethernet on the Pi3?), a usb-serial adaptor, and mouses, etc.

Mostly, I think your /boot/config.txt probably have some wrong / duplicated contents, but seeing as you have modified it by hand for your USB serial adaptor, the problem is probably caused by your own editing... I would say check your /boot/config.txt, and if you cannot see what you did wrong, start with a new clean SD card, and install from fresh, without all the extra USB devices.

amirush786, 2021-08-10:
> According to your dmesg output, you have a "Raspberry Pi 3 Model B Plus Rev 1.3" , i.e. a Pi 3B+, not a Pi 4. So that's the first inaccuracy/problem you have - you don't even know you have a Pi3, [...]

Thank you for the help. I will start from scratch. And get u the updates

HinTak, 2021-08-10:
Also, according to your `lsmod`, somehow the 2-mics device kernel module is also loaded too; that's usually not a problem in other context, as unused kernel modules not bound to any device does not have any effect, other than wasting a bit of memory - however, since you also seems to have duplicate entries / boot messages, my suspicion is that you might have conflicting things loaded multiple times via /boot/config.txt

amirush786, 2021-08-14:
> Also, according to your `lsmod`, somehow the 2-mics device kernel module is also loaded too; that's usually not a problem in other context, as unused kernel modules not bound to any device does not [...]

I have done the setup of raspberry pi. I am using the wifi and monitor, mouse and keyboard. I am not using the ethernet cable.
so when I am using the command for playing and recording at the same time which is "#Doing capture && playback the same time
arecord -D hw:1,0 -f S32_LE -r 16000 -c 8 to_be_record.wav &
#mono_to_play.wav is a mono channel wave file to play
aplay -D plughw:1,0 -r 16000 mono_to_play.wav"

from the seed 6 mic tool website. it plays the sound but can only hear the sound when I plug in my earphones. But I want it be played on the speaker which JBL which is connected via Bluetooth. It doesn't record and create a file which is to_be_record.wav. I have attached the lsmod and dmesg text file. let me know where I am going wrong. Thanks for the help in advance.
[Uploading lsmod_output (1).txt…]()
[Uploading dmesg_output.txt…]()

> Also, according to your `lsmod`, somehow the 2-mics device kernel module is also loaded too; that's usually not a problem in other context, as unused kernel modules not bound to any device does not [...]

I have done the setup of raspberry pi. I am using the wifi and monitor, mouse and keyboard. I am not using the ethernet cable.
so when I am using the command for playing and recording at the same time which is "#Doing capture && playback the same time
arecord -D hw:1,0 -f S32_LE -r 16000 -c 8 to_be_record.wav &
#mono_to_play.wav is a mono channel wave file to play
aplay -D plughw:1,0 -r 16000 mono_to_play.wav"

from the seed 6 mic tool website. it plays the sound but can only hear the sound when I plug in my earphones. But I want it be played on the speaker which JBL which is connected via Bluetooth. It doesn't record and create a file which is to_be_record.wav. I have attached the lsmod and dmesg text file. let me know where I am going wrong. Thanks for the help in advance.
[Uploading lsmod_output (1).txt…]()
[Uploading dmesg_output.txt…]()

HinTak, 2021-08-15:
If you want to play sound through another device (bluetooth connected) that you should consult the instruction for that device. The playback instruction for the 6-macs device is for playing sound through the phone jacket on the seeed studio board, which you seems to be able to do.

amirush786, 2021-08-15:
How can I detect my speaker on the pi. Is there any way we can do that. Can we change the default setting to play on the speaker connected via aux cable to the pi.

HinTak, 2021-08-15:
You could just search for 'raspberrypi bluetooth speaker" on Google. This was one of the earlier results: https://www.raspberrypi.org/forums/viewtopic.php?f=35&t=235519

amirush786, 2021-08-15:
> You could just search for 'raspberrypi bluetooth speaker" on Google. This was one of the earlier results: https://www.raspberrypi.org/forums/viewtopic.php?f=35&t=235519

Thank you sir will look into it.

amirush786, 2021-08-18:
> You could just search for 'raspberrypi bluetooth speaker" on Google. This was one of the earlier results: https://www.raspberrypi.org/forums/viewtopic.php?f=35&t=235519

Hello I tried doing the tutorials for connecting the pi via Bluetooth device. The problem u am facing is that when I hit the command of recording and playing at the same time it's not playing anything on the speaker even though it shows me there the wav file is playing and also nothing is recorded as well. 
Also when I try to play just the sound file it throws me error unknown pcm101.
Below I have attached the screen shot of the pi terminal. [image]
Can you please help me fix this.

HinTak, 2021-08-18:
The problem you are facing is that you haven't figured out how to use your bluetooth speaker, just for playing sound through it. If you are not playing sound through it, you cannot record with any device, including respeaker. You not being able to use the bluetooth speaker alone, is a separate and unrelated problem to how you can or cannot use the respeaker for recording and playback.

The higher respeakers (linear 4-mics and 6-mics, not the square 4-mics) have their own phone sockets and can be used for playback. The instructions for simultaneous playback and recording are for those two devices. If you want to use a separate playback device, such as a bluetooth speaker, for playing sound, while recording with respeaker, you need to figure out how to use those unrelated bluetooth speakers first.

amirush786, 2021-08-18:
> The problem you are facing is that you haven't figured out how to use your bluetooth speaker, just for playing sound through it. If you are not playing sound through it, you cannot record with any [...]

So when I connect my jbl speaker and instead of Bluetooth I connected with aux cable. So when I got the command for playing the wav file and recording at the same time, it does play the sound from my speaker but it doesn't record a new file. So in that case we can't do both the things at the same time right?. Just want to know am I on the right thought process.

HinTak, 2021-08-18:
If you are using your speaker via the phone socket on the pi through a cable, you should be using `aplay -D sysdefault ...` or `aplay -D CARD=Headphones ...` . and you record through respeaker via `arecord -D ac108 ...`. You should try to do each of these separately and successfully before you try to do them simultaneously.

If you are unable to record, also check your recording volume (one of the settings under `alsamixer`) is not too low.

HinTak, 2021-08-18:
You should be able to record alone, and record your own voice, by saying a few words in front of the device not too far from it. Check with `alsamixer` the recording volume settings is not too low, if your voice is not in the resulting recording.

amirush786, 2021-08-19:
> You should be able to record alone, and record your own voice, by saying a few words in front of the device not too far from it. Check with `alsamixer` the recording volume settings is not too low, if your voice is not in the resulting recording.

When I am doing a recording as you told the sound gets re order but it seems that it's not recording my voice it's recording sshhhhhhhhhh volume kinda thing not sure why?. Is there any way you can have a look at it via pi tunnel or something like that. That woukd be great help if you can help me figure out this issue that I am facing.

HinTak, 2021-08-19:
Did you check the "Capture" (not "Playback"!) volume setting under "alsamixer" that the setting is high enough? You can attach the sound recording to issues (same as attaching txt files, etc - zip it if does not take wav directly) too.

At this stage I think you best option is really to get somebody local to your location to check what's wrong. Join a local computer club or something...

amirush786, 2021-08-20:
> Did you check the "Capture" (not "Playback"!) volume setting under "alsamixer" that the setting is high enough? You can attach the sound recording to issues (same as attaching txt files, etc - zip [...]

I check the capture in alsamixer it was 100 there. Moreover I have attached the images, zip file (which holds the ls,od dmesg and wave file in it). Can you know where I am going wrong?

[dmesg_output.txt](https://github.com/respeaker/seeed-voicecard/files/7024258/dmesg_output.txt)
[lsmod_output.txt](https://github.com/respeaker/seeed-voicecard/files/7024259/lsmod_output.txt)
[a.zip](https://github.com/respeaker/seeed-voicecard/files/7024263/a.zip)
[image]
[image]

HinTak, 2021-08-20:
@amirush786 alsamixer on the wrong card - use F6 to switch to the correct one. And WTF, according to both dmesg and your screenshot, you have the square 4-mics device, not the 6-mics and not the linear-4-mics device. In that case, the square 4-mics device does not have a phone jack, so you play back through the pi's on-board phone jacket, which is just for example, `aplay -D plughw:CARD=Headphones,DEV=0 ...` .

amirush786, 2021-08-20:
> @amirush786 alsamixer on the wrong card - use F6 to switch to the correct one. And WTF, according to both dmesg and your screenshot, you have the square 4-mics device, not the 6-mics and not the [...]

Will check on it. Thank you very much for the help. I will give the updates.

amirush786, 2021-08-20:
> @amirush786 alsamixer on the wrong card - use F6 to switch to the correct one. And WTF, according to both dmesg and your screenshot, you have the square 4-mics device, not the 6-mics and not the [...]

I didn't change any setting like I got the device from the university stating that its 6 mic and I started doing research on that. I will notify them. As well and give you the uodates of the tings you told me to do.

amirush786, 2021-08-20:
> @amirush786 alsamixer on the wrong card - use F6 to switch to the correct one. And WTF, according to both dmesg and your screenshot, you have the square 4-mics device, not the 6-mics and not the [...]

I have to choose bcm2835 Headphones right?. [image]

[image]

How can I make sure it has chosen the correct one. Because when I play using bcm2835 it shows me this error.

[image]

HinTak, 2021-08-20:
No, for capture you need to looking at the alsamixer setting of seeed-4mics-voicecard, and make sure the capture setting of that card is high.

HinTak, 2021-08-20:
And your command line is wrong - for your hardware combo, you record through the voicecard - `arecord -D ac108 ...` and you play through the headphone. Your alsamixer capture setting for the 4-mics card, and use -D ac108 for recording.

HinTak, 2021-08-20:
Your voicecard is device **2**, so if you want the "-D hw:X,Y" syntax instead of ac108, etc, you should be using hw:2,0 !

HinTak, 2021-08-20:
Since you mentioned you got you card "from the university", surely there are people there who can help and tell you that some of your basic operations with alsa / recording / playback are just looking at the wrong setting panel, passing wrong device numbers, etc?

amirush786, 2021-08-20:
> Since you mentioned you got you card "from the university", surely there are people there who can help and tell you that some of your basic operations with alsa / recording / playback are just looking at the wrong setting panel, passing wrong device numbers, etc?

No they won't help in such cases. You can have access to my pi using pi tunnel. It would be great help sir. I am struggling a lot with the device and I am stuck too badly with it. My main aim is to collect data from the tool which is playing a modulated sound via a speaker which is connected to pi and then at the same time I have to record as well. Didn't knew that it might be so tedious. There's a website from which you can access my pi remotely.

HinTak, 2021-08-20:
This is getting quite crazy - at the beginning, you said you have pi4 + 6-mics . Now it turns out that you have pi 3.+ square 4-mics!

amirush786, 2021-08-20:
> This is getting quite crazy - at the beginning, you said you have pi4 + 6-mics . Now it turns out that you have pi 3.+ square 4-mics!

Thats what I was told by my college. Can you please help me to fix this issue by accessing my terminal sir please

HinTak, 2021-08-20:
I do not and will not access your pi channel! You need to be able to record identifiable sound (your own voice, etc), and that absolutely cannot be done by any kind of remone access. You must solve the problem by having the device in front of you while you record with identiable sound, such as you speaking to the device or you clapping or clicking your fingers. It cannot be fixed by any kind of remote access.

HinTak, 2021-08-20:
Get your teacher to help you, and show you this message.

amirush786, 2021-08-20:
arecord -D hw:2,0 -f S32_LE -r 16000 -c 8 to_be_record.wav & aplay -D plughw:2,0 -r 16000 a1.wav 
arecord: main:828: aplay: main:828: audio open error: Device or resource busy
audio open error: No such file or directory
[1]+  Exit 1                  arecord -D hw:2,0 -f S32_LE -r 16000 -c 8 to_be_record.wav

I am getting this error.

amirush786, 2021-08-20:
> Get your teacher to help you, and show him/her this message, and the threads above.

I will tell my professor regarding this and also in the dmesg where can I find out that its a 4 mic and not the 6 mic. Can you show me that.

HinTak, 2021-08-21:
Same problem over and over! Use ac108 for recording and headphone for playing. You are using the ac108 for playing! Why did you do 2,0 in both? For your hardware you need to use 2,0 for recording but  `aplay -D plughw:CARD=Headphones,DEV=0 ...` .

Make sure your arecord part looks like `arecord -D ac108 ...` and your aplay part looks like
 `aplay -D plughw:CARD=Headphones,DEV=0 ...` . Why is that so hard for you!!!???

HinTak, 2021-08-21:
And you still need to show a alsamixer *capture* setting screen for the *correct* device being high enough.

HinTak, 2021-08-21:
> I have to choose bcm2835 Headphones right?. [image]
> 

No, seeed-4-mic-voicecard.

> [image]
> 

No, press F4 for "capture", and make sure it is high, and it still says "seeed-4mics-voicecard" at the top left!!!

amirush786, 2021-08-23:
> > I have to choose bcm2835 Headphones right?. [image]
> 
> No, seeed-4-mic-voicecard.
> 
> > [image]
> 
> No, press F4 for "capture", and make sure it is high, and it still says "seeed-4mics-voicecard" at the top left!!!

Yes Its recording and playing at the same time thabkyoy very much for that. But now the issue with the Alsa mixer the wav file after doing arecord that file still plays shhhhh sound I don't know why so I will explain what I did. I did Alsamixer command and it's shows me this.

[image]

So after that I typed in F6 to change my card to seeed card.
[image]

Then is scroll down to seed card and hit enter. So I see this. 
[image]
[image]

Then I press f4 and I see this so I don't know where should I see that capture is high.
This is the last thing that messing with me.

amirush786, 2021-08-24:
> 
> > I have to choose bcm2835 Headphones right?. [image]
> > 
> 
> No, seeed-4-mic-voicecard.
> 
> > [image]
> > 
> 
> No, press F4 for "capture", and make sure it is high, and it still says "seeed-4mics-voicecard" at the top left!!!

Just the problem with capture of Alsa mixer is left to do. Please can you help with that.

AIWintermuteAI, 2021-08-24:
@amirush786 this is not Stack Overflow, neither it is repository for alsamixer. Please adress your questions at the appropriate channel - since now you're asking help with general sound help I recommend:
https://www.google.com/
https://www.raspberrypi.org/forums/
https://raspberrypi.stackexchange.com/