respeaker/seeed-voicecard #308: Kernel Version Preventing Successful Download

Tswizly, 2021-08-18:
### Describe the bug

The code obtained from the Respeaker 4-Mic Array [Wiki  ](https://wiki.seeedstudio.com/ReSpeaker_4_Mic_Array_for_Raspberry_Pi/) :
<html>
<body>

  | sudo apt-get update
-- | --
  | git clone https://github.com/respeaker/seeed-voicecard.git
  | cd seeed-voicecard
  | sudo ./install.sh
  | sudo reboot now

</body>
</html>
results in the error message "WARNING Your loaded kernel version is 5.10.52-v7. Not matching the updated version 5.10.52. Kernel was updated, but new kernel was not loaded yet. Please reboot your machine and then run this script again. This warning was added recently. However, upon rebooting and rerunning script, the kernel version remains 5.10.52-v7, and the issue is unresolved. 

### Expected behavior

The expected behavior is a successful installation resulting in the correct voice card name:
<html>
<body>

  | pi@raspberrypi:~ $ arecord -L
-- | --
  | null
  | Discard all samples (playback) or generate zero samples (capture)
  | jack
  | JACK Audio Connection Kit
  | pulse
  | PulseAudio Sound Server
  | default
  | playback
  | ac108
  | sysdefault:CARD=seeed4micvoicec
  | seeed-4mic-voicecard,
  | Default Audio Device
  | dmix:CARD=seeed4micvoicec,DEV=0
  | seeed-4mic-voicecard,
  | Direct sample mixing device
  | dsnoop:CARD=seeed4micvoicec,DEV=0
  | seeed-4mic-voicecard,
  | Direct sample snooping device
  | hw:CARD=seeed4micvoicec,DEV=0
  | seeed-4mic-voicecard,
  | Direct hardware device without any conversions
  | plughw:CARD=seeed4micvoicec,DEV=0
  | seeed-4mic-voicecard,
  | Hardware device with all software conversions
  | usbstream:CARD=seeed4micvoicec
  | seeed-4mic-voicecard
  | USB Stream Output
  | usbstream:CARD=ALSA
  | bcm2835 ALSA
  | USB Stream Output

</body>
</html>

### Platform

- Device: Raspberry Pi 3+
- OS: Raspbian OS 32bit kernel version
- Version/commit number 1fad1d2

HinTak, 2021-08-18:
I think you misunstand the message. There are two kernel releases showing 5.10.52:

 https://github.com/HinTak/RaspberryPi-Dev/blob/master/Raspbian-Kernel-Releases.md

As long as you ended up with the most recent (0805), you are fine.

AIWintermuteAI, 2021-08-19:
Hi @Tswizly 
It appears a commit solving https://github.com/respeaker/seeed-voicecard/issues/307 had a small typo in install.sh, that passed unnoticed when testing on Raspberry Pi Zero W.
I made changes and briefly tested them on both R PI 4 and R Pi Zero W. Can you pull latest code and see if the issue was resolved?

Tswizly, 2021-08-19:
Hi @AIWintermuteAI 
It worked! Yes, your change worked. Rebooting and re-running the sudo ./install.sh code did not work right away, but I had another microSD card with nothing on it so I simply used that to save myself the hassle of determining how to remove the git clone https://.../seeed-voicecard.git. However the code worked like a charm and did not give any error message at all, but rather worked the first time. 
Thank you @AIWintermuteAI  and @HinTak for your time. It is greatly appreciated.

Tswizly, 2021-08-19:
I am going to close the issue as it is resolved for me. Thank you again.

closed 2021-08-19 by Tswizly

AIWintermuteAI, 2021-08-19:
@Tswizly I'm glad that it worked for you! Have you checked if reSpeaker functioning normally?

Tswizly, 2021-08-20:
@AIWintermuteAI Yes it is functioning normally. The quality is very good in Audacity, the LED's work, and the sound recording is working. It is tremendous. Thank you again for the awesome help!