respeaker/seeed-voicecard #311: Failed to apply overlay '3 seeed-2mic-voicecard' (5.10.60+ kernel)

goderro, 2021-09-08:
### Describe the bug

Hi, I really want to get some solutions for my PI zero w board.

After installing 'seeed-voicecard' and reboot, it didn't work.
'Error! There are no instances of module: seeed-voicecard' 
I saw above messages during installing. I thought it gonna be some problems.

I can't find 'card 0' when 'aplay -l'. 
And I can't use dtoverlay after rebooting.

I could test for LED, It was worked.

I also installed bcm2835-1.68 at PI zero w board.

I tried uninstall and reinstall many time but it didn't work for me.

Is anybody help for me?

I also tried 'sudo ./install.sh --compat-kernel' option. But same result.

### Expected behavior

aplay -l

**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: seeed2micvoicec [seeed-2mic-voicecard], device 0: bcm2835-i2s-wm8960-hifi wm8960-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0

### Platform

- Device: [Raspberry zero W]
- OS: [e.g. Raspbian OS 32bit kernel 5.10.60+], Lite version
- Version/commit number [lastest version]

### Relevant log output

```shell
==================

pi@raspberrypi:~/seeed-voicecard $ sudo ./install.sh
Hit:1 http://ftp.kaist.ac.kr/raspbian/raspbian buster InRelease
Hit:2 http://archive.raspberrypi.org/debian buster InRelease
Reading package lists... Done
Building dependency tree
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree
Reading state information... Done
raspberrypi-kernel is already the newest version (1:1.20210831-1).
raspberrypi-kernel-headers is already the newest version (1:1.20210831-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package linux-raspi
E: Unable to locate package linux-headers-raspi
E: Unable to locate package linux-image-raspi
Reading package lists... Done
Building dependency tree
Reading state information... Done
dkms is already the newest version (2.6.1-4).
git is already the newest version (1:2.20.1-2+deb10u3).
i2c-tools is already the newest version (4.1-1).
libasound2-plugins is already the newest version (1.1.8-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Error! There are no instances of module: seeed-voicecard
0.3 located in the DKMS tree.

Creating symlink /var/lib/dkms/seeed-voicecard/0.3/source ->
                 /usr/src/seeed-voicecard-0.3

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
make -j1 KERNELRELEASE=5.10.60+ -C /lib/modules/5.10.60+/build M=/var/lib/dkms/seeed-voicecard/0.3/build.............................
cleaning build area....

DKMS: build completed.

snd-soc-wm8960.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.10.60+/kernel/sound/soc/codecs/

snd-soc-ac108.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.10.60+/kernel/sound/soc/codecs/

snd-soc-seeed-voicecard.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.10.60+/kernel/sound/soc/bcm/

depmod.....

DKMS: install completed.
git init
Initialized empty Git repository in /etc/voicecard/.git/
git add --all
git commit -m "origin configures"
[master (root-commit) 7055ea6] origin configures
 7 files changed, 1482 insertions(+)
 create mode 100644 ac108_6mic.state
 create mode 100644 ac108_asound.state
 create mode 100644 asound_2mic.conf
 create mode 100644 asound_4mic.conf
 create mode 100644 asound_6mic.conf
 create mode 100644 dkms.conf
 create mode 100644 wm8960_asound.state
Created symlink /etc/systemd/system/sysinit.target.wants/seeed-voicecard.service → /lib/systemd/system/seeed-voicecard.service.
------------------------------------------------------
Please reboot your raspberry pi to apply all settings
Enjoy!
------------------------------------------------------

========================

pi@raspberrypi:~ $ sudo dtoverlay seeed-2mic-voicecard
* Failed to apply overlay '3_seeed-2mic-voicecard' (kernel)

========================

pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: b1 [bcm2835 HDMI 1], device 0: bcm2835 HDMI 1 [bcm2835 HDMI 1]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 1: seeed2micvoicec [seeed-2mic-voicecard], device 0: bcm2835-i2s-wm8960-hifi wm8960-hifi-0 [bcm2835-i2s-wm8960-hifi wm8960-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
pi@raspberrypi:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 1: seeed2micvoicec [seeed-2mic-voicecard], device 0: bcm2835-i2s-wm8960-hifi wm8960-hifi-0 [bcm2835-i2s-wm8960-hifi wm8960-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

========================
```

AIWintermuteAI, 2021-09-13:
Hi there!
Where did you find this command?
```
sudo dtoverlay seeed-2mic-voicecard
```
To my knowledge it is not necessary at all.
Additionally, ```Error! There are no instances of module: seeed-voicecard``` is not an actual error - it merely says that you have no such module at the moment, then it proceeds to building it.

goderro, 2021-09-13:
> Hi there!
> Where did you find this command?
>
> ```
> sudo dtoverlay seeed-2mic-voicecard
> ```
>
> To my knowledge it is not necessary at all.
> Additionally, `Error! There are no instances of module: [...]

You can ignore about 'dtoverlay'. You already commented it is not necessary.
I just want to how to resolve this issue. 
I would take downgrade kernel version if you let me know what kernel version would be best stable version for us.

Thank you.