respeaker/seeed-voicecard #312: [Bug]: Kernel panic on master v5.9 and others with 4mic Circular array

baileyji, 2021-09-08:
### Describe the bug

I'm seeing kernel panics and improper operation arecord -L does seem to work, but using of the device via paman, Mumble, or other programs brings the system to a stop. 

### Expected behavior

Successfully use the microphone via pulse input in e.g. Audacity or Mumble.

### Platform

- Device: Raspberry Pi 4 4GB
- OS:  Raspbian OS 32bit kernel version with all updates: 5.10.60-v7l+ #1449 SMP Wed Aug 25 15:00:44 BST 2021 armv7l GNU/Linux
- Version/commit number master (dd9391fb78), HinTak v5.9 (48dbb3f8), and a commit I'd has success on last year (50dbe24b).

### Relevant log output

```shell
Sep 08 17:12:32 cloudlight kernel: ------------[ cut here ]------------
Sep 08 17:12:32 cloudlight kernel: WARNING: CPU: 3 PID: 325 at drivers/firmware/raspberrypi.c:64 rpi_firmware_transaction+0xec/0x128
Sep 08 17:12:32 cloudlight kernel: Firmware transaction timeout
Sep 08 17:12:32 cloudlight kernel: Modules linked in: uinput cmac fuse rfcomm bnep hci_uart btbcm bluetooth ecdh_generic ecc nft_counter nft_compat nf_tables nfnetlink spidev spi_bcm2835 8021q garp stp llc brcmfmac brcmutil v3d bcm2835_codec(C) gpu_sched bcm2835_isp(C) v4l2_mem2mem sha256_generic i2c_bcm2835 raspberrypi_hwmon bcm2835_v4l2(C) videobuf2_dma_contig bcm2835_mmal_vchiq(C) videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 cfg80211 videobuf2_common vc4 rfkill cec drm_kms_helper vc_sm_cma(C) w1_therm snd_soc_bcm2835_i2s videodev drm mc pwm_bcm2835 snd_bcm2835(C) w1_gpio rpivid_mem wire cn drm_panel_orientation_quirks syscopyarea sysfillrect uio_pdrv_genirq sysimgblt uio fb_sys_fops backlight nvmem_rmem sg snd_soc_wm8960(O) snd_soc_ac108(O) regmap_i2c snd_soc_seeed_voicecard(O) snd_soc_simple_card_utils snd_soc_core snd_compress snd_pcm_dmaengine snd_pcm snd_timer snd i2c_dev ip_tables x_tables ipv6
Sep 08 17:12:32 cloudlight kernel: CPU: 3 PID: 325 Comm: kworker/3:4 Tainted: G         C O      5.10.60-v7l+ #1449
Sep 08 17:12:32 cloudlight kernel: Hardware name: BCM2711
Sep 08 17:12:32 cloudlight kernel: Workqueue: events get_values_poll [raspberrypi_hwmon]
Sep 08 17:12:32 cloudlight kernel: Backtrace: 
Sep 08 17:12:32 cloudlight kernel: [<c0b828fc>] (dump_backtrace) from [<c0b82c90>] (show_stack+0x20/0x24)
Sep 08 17:12:32 cloudlight kernel:  r7:ffffffff r6:00000000 r5:60000013 r4:c12e6b7c
Sep 08 17:12:32 cloudlight kernel: [<c0b82c70>] (show_stack) from [<c0b8707c>] (dump_stack+0xcc/0xf8)
Sep 08 17:12:32 cloudlight kernel: [<c0b86fb0>] (dump_stack) from [<c0220b28>] (__warn+0xfc/0x114)
Sep 08 17:12:32 cloudlight kernel:  r10:dac06008 r9:00000009 r8:c09bc1a4 r7:00000040 r6:00000009 r5:c09bc1a4
Sep 08 17:12:32 cloudlight kernel:  r4:c0ea1428 r3:c1205094
Sep 08 17:12:32 cloudlight kernel: [<c0220a2c>] (__warn) from [<c0b83434>] (warn_slowpath_fmt+0xa4/0xd8)
Sep 08 17:12:32 cloudlight kernel:  r7:00000040 r6:c0ea1428 r5:c1205048 r4:c0ea1448
Sep 08 17:12:32 cloudlight kernel: [<c0b83394>] (warn_slowpath_fmt) from [<c09bc1a4>] (rpi_firmware_transaction+0xec/0x128)
Sep 08 17:12:32 cloudlight kernel:  r9:c1a7d9c0 r8:00000010 r7:00000000 r6:ffffff92 r5:c1a7d9c0 r4:c1205048
Sep 08 17:12:32 cloudlight kernel: [<c09bc0b8>] (rpi_firmware_transaction) from [<c09bc29c>] (rpi_firmware_property_list+0xbc/0x170)
Sep 08 17:12:32 cloudlight kernel:  r7:c1205048 r6:dac06000 r5:00001000 r4:dac0601c
Sep 08 17:12:32 cloudlight kernel: [<c09bc1e0>] (rpi_firmware_property_list) from [<c09bc3c0>] (rpi_firmware_property+0x70/0x118)
Sep 08 17:12:32 cloudlight kernel:  r10:c77d2c4c r9:00030046 r8:00000010 r7:c1a7d9c0 r6:c3b01ed8 r5:00000004
Sep 08 17:12:32 cloudlight kernel:  r4:c77d2c40
Sep 08 17:12:32 cloudlight kernel: [<c09bc350>] (rpi_firmware_property) from [<bf1850c0>] (get_values_poll+0x4c/0x150 [raspberrypi_hwmon])
Sep 08 17:12:32 cloudlight kernel:  r10:00000000 r9:00000000 r8:000000c0 r7:eff30a00 r6:eff2d680 r5:c37245cc
Sep 08 17:12:32 cloudlight kernel:  r4:c1205048 r3:00000004
Sep 08 17:12:32 cloudlight kernel: [<bf185074>] (get_values_poll [raspberrypi_hwmon]) from [<c023dbb0>] (process_one_work+0x250/0x5a0)
Sep 08 17:12:32 cloudlight kernel:  r5:c3724a80 r4:c37245cc
Sep 08 17:12:32 cloudlight kernel: [<c023d960>] (process_one_work) from [<c023df60>] (worker_thread+0x60/0x5c4)
Sep 08 17:12:32 cloudlight kernel:  r10:eff2d680 r9:c1203d00 r8:eff2d698 r7:00000008 r6:eff2d680 r5:c3724a94
Sep 08 17:12:32 cloudlight kernel:  r4:c3724a80
Sep 08 17:12:32 cloudlight kernel: [<c023df00>] (worker_thread) from [<c0245a34>] (kthread+0x170/0x174)
Sep 08 17:12:32 cloudlight kernel:  r10:c3aade74 r9:c3724a80 r8:c023df00 r7:c3b00000 r6:00000000 r5:c3b6a0c0
Sep 08 17:12:32 cloudlight kernel:  r4:c38ad480
Sep 08 17:12:32 cloudlight kernel: [<c02458c4>] (kthread) from [<c02000ec>] (ret_from_fork+0x14/0x28)
Sep 08 17:12:32 cloudlight kernel: Exception stack(0xc3b01fb0 to 0xc3b01ff8)
Sep 08 17:12:32 cloudlight kernel: 1fa0:                                     00000000 00000000 00000000 00000000
Sep 08 17:12:32 cloudlight kernel: 1fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Sep 08 17:12:32 cloudlight kernel: 1fe0: 00000000 00000000 00000000 00000000 00000013 00000000
Sep 08 17:12:32 cloudlight kernel:  r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:c02458c4
Sep 08 17:12:32 cloudlight kernel:  r4:c3b6a0c0
Sep 08 17:12:32 cloudlight kernel: ---[ end trace 91a5eed15c0afa4c ]---
Sep 08 17:12:32 cloudlight kernel: hwmon hwmon2: Failed to get throttled (-110)
Sep 08 17:12:34 cloudlight rtkit-daemon[809]: The canary thread is apparently starving. Taking action.
Sep 08 17:12:34 cloudlight rtkit-daemon[809]: Demoting known real-time threads.
Sep 08 17:12:34 cloudlight rtkit-daemon[809]: Successfully demoted thread 1474 of process 791 (n/a).
Sep 08 17:12:34 cloudlight rtkit-daemon[809]: Successfully demoted thread 798 of process 798 (n/a).
Sep 08 17:12:34 cloudlight rtkit-daemon[809]: Demoted 2 threads.
Sep 08 17:12:37 cloudlight fcserver-rpi[570]: Error submitting USB transfer: No such device (it may have been disconnected)
Sep 08 17:12:37 cloudlight fcserver-rpi[570]: Error submitting USB transfer: No such device (it may have been disconnected)
Sep 08 17:12:37 cloudlight fcserver-rpi[570]: Error submitting USB transfer: No such device (it may have been disconnected)
Sep 08 17:12:37 cloudlight fcserver-rpi[570]: Error submitting USB transfer: No such device (it may have been disconnected)
Sep 08 17:12:37 cloudlight kernel: xhci_hcd 0000:01:00.0: xHCI host not responding to stop endpoint command.
Sep 08 17:12:37 cloudlight kernel: xhci_hcd 0000:01:00.0: USBSTS: EINT
Sep 08 17:12:37 cloudlight kernel: xhci_hcd 0000:01:00.0: xHCI host controller not responding, assume dead
Sep 08 17:12:37 cloudlight kernel: xhci_hcd 0000:01:00.0: HC died; cleaning up
Sep 08 17:12:37 cloudlight kernel: usb 1-1: USB disconnect, device number 2
Sep 08 17:12:37 cloudlight kernel: usb 1-1.3: USB disconnect, device number 3
Sep 08 17:12:37 cloudlight kernel: usb 2-1: USB disconnect, device number 2
Sep 08 17:12:37 cloudlight fcserver-rpi[570]: Error submitting USB transfer: No such device (it may have been disconnected)
Sep 08 17:12:37 cloudlight fcserver-rpi[570]: Error submitting USB transfer: No such device (it may have been disconnected)
Sep 08 17:12:37 cloudlight fcserver-rpi[570]: Error submitting USB transfer: No such device (it may have been disconnected)
Sep 08 17:12:37 cloudlight fcserver-rpi[570]: Error submitting USB transfer: No such device (it may have been disconnected)
Sep 08 17:12:37 cloudlight fcserver-rpi[570]: Error submitting USB transfer: No such device (it may have been disconnected)
Sep 08 17:12:37 cloudlight fcserver-rpi[570]: Error submitting USB transfer: No such device (it may have been disconnected)
Sep 08 17:12:37 cloudlight fcserver-rpi[570]: USB device Fadecandy (Serial# HZOMJZDJMWMLDXSN, Version 1.07) removed.
Sep 08 17:12:37 cloudlight kernel: sched: RT throttling activated
Sep 08 17:12:37 cloudlight kernel: sd 0:0:0:0: [sda] Synchronizing SCSI cache
Sep 08 17:12:37 cloudlight kernel: sd 0:0:0:0: [sda] Synchronize Cache(10) failed: Result: hostbyte=0x01 driverbyte=0x00
Sep 08 17:12:37 cloudlight udisksd[385]: Cleaning up mount point /media/pi/SamsungUSB (device 8:1 no longer exists)
Sep 08 17:12:37 cloudlight systemd[773]: media-pi-SamsungUSB.mount: Succeeded.
Sep 08 17:12:37 cloudlight systemd[1]: media-pi-SamsungUSB.mount: Succeeded.
Sep 08 17:12:37 cloudlight mount.exfat[1402]: failed to write super block
Sep 08 17:12:37 cloudlight systemd[1153]: media-pi-SamsungUSB.mount: Succeeded.
```

HinTak, 2021-09-09:
From the log you posted it looks like you have a USB device fadecandy which has stopped responding, and that in turn has caused the raspberry pi firmware to time out. Nothing to do with audio at all. You may need to review which part of your applications are using fadecandy, and perhaps contact the Adafruit people (who sell these, and presumably wrote the drivers too) for help.

baileyji, 2021-09-09:
Yes, I saw that as well and did find it interesting. I'm sure there is an issue with the 4mic array driver though there may be some interplay with USB. I think what is happening there though is that something related to the seeed driver is dominating things in the kernel and causing a hose of other quasi-intensive low level resources to randomely fail sometimes. As long as I don't access the 4mic array though the system is stable on a week+ timescale. I've found it very hard to capture any sort of debugging data as generally the system becomes completely unresponsive and connections drop before I get any info in my tail of the journal.

For example if I open paman, elect the multichannel input of the mic and try to show its levels pulse crashes instantly. If I try to open any of the various seed options with also in e.g. mumble the system dies, If I open the input in mumble via pulse pulse dies.   

Some i2c timeouts:

> Sep 08 14:18:35 cloudlight kernel: i2c-bcm2835 fe804000.i2c: i2c transfer timed out
> Sep 08 14:18:35 cloudlight kernel: ac10x_update_bits() error->[REG-0x30,val-0x00]
> Sep 08 14:18:35 cloudlight kernel: ac10x_update_bits() error->[REG-0x10,val-0x00]
> Sep 08 14:18:36 cloudlight kernel: i2c-bcm2835 fe804000.i2c: i2c transfer timed out
> Sep 08 14:18:36 cloudlight kernel: ac10x_update_bits() error->[REG-0x30,val-0x00]
> Sep 08 14:18:36 cloudlight kernel: ac10x_write error->[REG-0x21,val-0x00]
> Sep 08 14:18:38 cloudlight kernel: i2c-bcm2835 fe804000.i2c: i2c transfer timed out
> Sep 08 14:18:38 cloudlight kernel: ac10x_update_bits() error->[REG-0x30,val-0x00]
> Sep 08 14:18:38 cloudlight kernel: ac10x_update_bits() error->[REG-0x10,val-0x00]
> Sep 08 14:18:40 cloudlight kernel: i2c-bcm2835 fe804000.i2c: i2c transfer timed out
> Sep 08 14:18:40 cloudlight kernel: ac10x_update_bits() error->[REG-0x30,val-0x00]
> Sep 08 14:18:40 cloudlight kernel: ac10x_update_bits() error->[REG-0x10,val-0x00]
> Sep 08 14:18:41 cloudlight kernel: i2c-bcm2835 fe804000.i2c: i2c transfer timed out
> Sep 08 14:18:41 cloudlight kernel: ac10x_update_bits() error->[REG-0x30,val-0x00]
> Sep 08 14:18:41 cloudlight kernel: ac10x_update_bits() error->[REG-0x10,val-0x00]
 

Another kernel log I did get (related to the GPU perhaps, which seems odd):
 
> Sep 08 14:18:34 cloudlight kernel: ------------[ cut here ]------------
> Sep 08 14:18:34 cloudlight kernel: WARNING: CPU: 1 PID: 636 at drivers/gpu/drm/vc4/vc4_firmware_kms.c:403 vc4_plane_set_blank+0x15c/0x168 [vc4]
> Sep 08 14:18:34 cloudlight kernel: vc4_plane_set_blank: firmware call failed. Please update your firmware
> Sep 08 14:18:34 cloudlight kernel: Modules linked in: uinput cmac rfcomm fuse bnep hci_uart btbcm bluetooth ecdh_generic ecc nft_counter nft_compat nf_tables nfnetlink spidev spi_bcm2835 8021q garp stp llc brcmfmac brcmutil sg v3d gpu_sched sha256_generic cfg80211 rfkill raspberrypi_hwmon bcm2835_codec(C) i2c_bcm2835 bcm2835_isp(C) v4l2_mem2mem bcm2835_v4l2(C) videobuf2_dma_contig bcm2835_mmal_vchiq(C) videobuf2_vmalloc videobuf2_memops vc4 videobuf2_v4l2 videobuf2_common cec videodev snd_bcm2835(C) mc pwm_bcm2835 vc_sm_cma(C) snd_soc_bcm2835_i2s drm_kms_helper w1_therm drm rpivid_mem drm_panel_orientation_quirks nvmem_rmem w1_gpio syscopyarea sysfillrect wire sysimgblt cn fb_sys_fops backlight uio_pdrv_genirq uio snd_soc_wm8960(O) snd_soc_ac108(O) regmap_i2c snd_soc_seeed_voicecard(O) snd_soc_simple_card_utils snd_soc_core snd_compress snd_pcm_dmaengine snd_pcm snd_timer snd i2c_dev ip_tables x_tables ipv6
> Sep 08 14:18:34 cloudlight kernel: CPU: 1 PID: 636 Comm: Xorg Tainted: G        WC O      5.10.60-v7l+ #1449
> Sep 08 14:18:34 cloudlight kernel: Hardware name: BCM2711
> Sep 08 14:18:34 cloudlight kernel: Backtrace: 
> Sep 08 14:18:34 cloudlight kernel: [<c0b828fc>] (dump_backtrace) from [<c0b82c90>] (show_stack+0x20/0x24)
> Sep 08 14:18:34 cloudlight kernel:  r7:ffffffff r6:00000000 r5:60000013 r4:c12e6b7c
> Sep 08 14:18:34 cloudlight kernel: [<c0b82c70>] (show_stack) from [<c0b8707c>] (dump_stack+0xcc/0xf8)
> Sep 08 14:18:34 cloudlight kernel: [<c0b86fb0>] (dump_stack) from [<c0220b28>] (__warn+0xfc/0x114)
> Sep 08 14:18:34 cloudlight kernel:  r10:c7500c00 r9:00000009 r8:bf331810 r7:00000193 r6:00000009 r5:bf331810
> Sep 08 14:18:34 cloudlight kernel:  r4:bf358544 r3:c1205094
> Sep 08 14:18:34 cloudlight kernel: [<c0220a2c>] (__warn) from [<c0b83434>] (warn_slowpath_fmt+0xa4/0xd8)
> Sep 08 14:18:34 cloudlight kernel:  r7:00000193 r6:bf358544 r5:c1205048 r4:bf35850c
> Sep 08 14:18:34 cloudlight kernel: [<c0b83394>] (warn_slowpath_fmt) from [<bf331810>] (vc4_plane_set_blank+0x15c/0x168 [vc4])
> Sep 08 14:18:34 cloudlight kernel:  r9:00000000 r8:00000000 r7:c34d5800 r6:c3500220 r5:c1205048 r4:ffffff92
> Sep 08 14:18:34 cloudlight kernel: [<bf3316b4>] (vc4_plane_set_blank [vc4]) from [<bf3319a4>] (vc4_plane_atomic_update+0x34/0x38 [vc4])
> Sep 08 14:18:34 cloudlight kernel:  r8:c7501e00 r7:bf353b38 r6:c3500040 r5:00000007 r4:c74f26c0
> Sep 08 14:18:34 cloudlight kernel: [<bf331970>] (vc4_plane_atomic_update [vc4]) from [<bf272138>] (drm_atomic_helper_commit_planes+0xe8/0x28c [drm_kms_helper])
> Sep 08 14:18:34 cloudlight kernel: [<bf272050>] (drm_atomic_helper_commit_planes [drm_kms_helper]) from [<bf334174>] (vc4_atomic_complete_commit+0x1c8/0x714 [vc4])
> Sep 08 14:18:34 cloudlight kernel:  r10:c69cf9c0 r9:00000000 r8:c74f28c0 r7:00000000 r6:00000001 r5:c34d5800
> Sep 08 14:18:34 cloudlight kernel:  r4:c74f26c0
[... cut ...]
> Sep 08 14:43:26 cloudlight pulseaudio[2760]: E: [alsa-source-bcm2835-i2s-ac10x-codec0 ac10x-codec0-0] alsa-util.c:   silence_size : 0
> Sep 08 14:43:26 cloudlight pulseaudio[2760]: E: [alsa-source-bcm2835-i2s-ac10x-codec0 ac10x-codec0-0] alsa-util.c:   boundary     : 1386217472
> Sep 08 14:43:26 cloudlight pulseaudio[2760]: E: [alsa-source-bcm2835-i2s-ac10x-codec0 ac10x-codec0-0] alsa-util.c:   appl_ptr     : 12221854
> Sep 08 14:43:26 cloudlight pulseaudio[2760]: E: [alsa-source-bcm2835-i2s-ac10x-codec0 ac10x-codec0-0] alsa-util.c:   hw_ptr       : 12248365

HinTak, 2021-09-09:
"I think what is happening there though is that something related to the seeed driver is dominating things in the kernel and causing a hose of other quasi-intensive low level resources to randomely fail sometimes." is issue #251 ! I thought it is disastrous on 64-bit kernel, didn't know it is observable on 32-bit kernel too.

And it certainly looks like it is the same section of code in ac10x_update_bits() as in #251 .

baileyji, 2021-09-09:
Sigh. Reading through these other tickets has me suspecting its time to invest in a different mic hat... I'm willing to offer some help debugging/and thinking, but I've just delivered the hardware to someone so I'm a bit limited for the moment.