respeaker/seeed-voicecard #313: [Bug]: seeed-2mic-voicecard driver installed, but hardware is not working

IgorPelevanyuk, 2021-09-11:
### Describe the bug

Hello! I have bought ReSpeaker 2mic and plugged in Raspbery Pi Zero.

Driver installation passed well as far as I see:
`Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian buster InRelease
Reading package lists... Done              
Building dependency tree       
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
raspberrypi-kernel is already the newest version (1:1.20210831-1).
raspberrypi-kernel-headers is already the newest version (1:1.20210831-1).
The following package was automatically installed and is no longer required:
  python-colorzero
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package linux-raspi
E: Unable to locate package linux-headers-raspi
E: Unable to locate package linux-image-raspi
Reading package lists... Done
Building dependency tree       
Reading state information... Done
dkms is already the newest version (2.6.1-4).
git is already the newest version (1:2.20.1-2+deb10u3).
i2c-tools is already the newest version (4.1-1).
libasound2-plugins is already the newest version (1.1.8-1).
The following package was automatically installed and is no longer required:
  python-colorzero
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

-------- Uninstall Beginning --------
Module:  seeed-voicecard
Version: 0.3
Kernel:  5.10.60+ (armv6l)
-------------------------------------

Status: Before uninstall, this module version was ACTIVE on this kernel.

snd-soc-wm8960.ko:
 - Uninstallation
   - Deleting from: /lib/modules/5.10.60+/kernel/sound/soc/codecs/
rmdir: failed to remove 'kernel/sound/soc/codecs': Directory not empty
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

snd-soc-ac108.ko:
 - Uninstallation
   - Deleting from: /lib/modules/5.10.60+/kernel/sound/soc/codecs/
rmdir: failed to remove 'kernel/sound/soc/codecs': Directory not empty
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

snd-soc-seeed-voicecard.ko:
 - Uninstallation
   - Deleting from: /lib/modules/5.10.60+/kernel/sound/soc/bcm/
rmdir: failed to remove 'kernel/sound/soc/bcm': Directory not empty
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

depmod.....

DKMS: uninstall completed.

------------------------------
Deleting module version: 0.3
completely from the DKMS tree.
------------------------------
Done.

Creating symlink /var/lib/dkms/seeed-voicecard/0.3/source ->
                 /usr/src/seeed-voicecard-0.3

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area.....
make -j1 KERNELRELEASE=5.10.60+ -C /lib/modules/5.10.60+/build M=/var/lib/dkms/seeed-voicecard/0.3/build..............................
cleaning build area.....

DKMS: build completed.

snd-soc-wm8960.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.10.60+/kernel/sound/soc/codecs/

snd-soc-ac108.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.10.60+/kernel/sound/soc/codecs/

snd-soc-seeed-voicecard.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.10.60+/kernel/sound/soc/bcm/

depmod...

DKMS: install completed.
mkdir: cannot create directory ‘/etc/voicecard’: File exists
git init
Reinitialized existing Git repository in /etc/voicecard/.git/
git add --all
git commit -m "origin configures"
On branch master
nothing to commit, working tree clean
------------------------------------------------------
Please reboot your raspberry pi to apply all settings
Enjoy!
------------------------------------------------------
`

But recording records silence. So I tried to play something with headphones. 
`aplay -D hw:1 work.wav` also muted although I know that the file is not empty and when I play it on hw:0, which is HDMI to my TV, I can hear the record. 

### Expected behavior

Sound in headphones, signal in records.

### Platform

- Device: Raspberry Pi Zero
- OS: Raspbian OS 32bit kernel version Linux raspberrypi 5.10.60+ #1449 Wed Aug 25 14:59:20 BST 2021 armv6l GNU/Linux
- Version/commit number dd9391fb78e6ec0ead495e5246e32a781d056ab2

IgorPelevanyuk, 2021-09-11:
Thats may be my issue, please do not react on this issue

IgorPelevanyuk, 2021-09-12:
I have to mention that I have bought my respeaker from aliexpress. And it looks like it was produced by Keyestudio. On official wiki page it is written that seeed drivers should work. Do you really support it?

AIWintermuteAI, 2021-09-13:
Hi there!
No we do not support third party knock-offs.

closed 2021-09-13 by AIWintermuteAI