respeaker/seeed-voicecard #314: [Bug]: Soundcard not available/ shown in aplay after installation

Baggerfahrer, 2021-09-12:
### Describe the bug

Hey all

i got an issue during installation process of the soundcard. It seems to be, that the installation process is not run properly, as the soundcard is not available after the installation.

I dont know if this is relevant, but the following errors message during installation process occurs:
```
System has not been booted with systemd as init system (PID 1). Can't operate.
Failed to connect to bus: Host is down
```

After the installation and a reboot, i got the following output of `aplay -a` and `dkms`
```
sudo dkms status
seeed-voicecard, 0.3, 5.10.60-v7+, armv7l: installed
```

```
aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 H                                                                                                                  eadphones]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
```

The soundcard is as well not available in alsamixer and cannot be used...

Any ideas?

### Expected behavior

Working soundcard :) But unfortnately, not...

### Platform

Raspberry Pi 3B

uname -a
Linux raspberrypi-lulu 5.10.60-v7+ #1449 SMP Wed Aug 25 15:00:01 BST 2021 armv7l                                                                                                            GNU/Linux

cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

### Relevant log output

```shell
Installation process log is as follows:

sudo ./install.sh
Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Ign:2 https://apt.mopidy.com buster InRelease
Get:3 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Err:4 https://apt.mopidy.com buster Release
  Certificate verification failed: The certificate is NOT trusted. The certificate chain uses not yet valid certificate.  Could not handshake: Error in the certificate verification. [IP:                                                  185.199.108.153 443]
Reading package lists... Done
E: The repository 'https://apt.mopidy.com buster Release' no longer has a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: Release file for http://raspbian.raspberrypi.org/raspbian/dists/buster/InRelease is not valid yet (invalid for another 8h 45min 28s). Updates for this repository will not be applied.
Reading package lists... Done
Building dependency tree
Reading state information... Done
raspberrypi-kernel is already the newest version (1:1.20210831-1).
raspberrypi-kernel-headers is already the newest version (1:1.20210831-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package linux-raspi
E: Unable to locate package linux-headers-raspi
E: Unable to locate package linux-image-raspi
Reading package lists... Done
Building dependency tree
Reading state information... Done
dkms is already the newest version (2.6.1-4).
git is already the newest version (1:2.20.1-2+deb10u3).
i2c-tools is already the newest version (4.1-1).
libasound2-plugins is already the newest version (1.1.8-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Error! There are no instances of module: seeed-voicecard
0.3 located in the DKMS tree.

Creating symlink /var/lib/dkms/seeed-voicecard/0.3/source ->
                 /usr/src/seeed-voicecard-0.3

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=5.10.60-v7+ -C /lib/modules/5.10.60-v7+/build M=/var/lib/dkms/seeed-voicecard/0.3/build.................
cleaning build area...

DKMS: build completed.

snd-soc-wm8960.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.10.60-v7+/kernel/sound/soc/codecs/

snd-soc-ac108.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.10.60-v7+/kernel/sound/soc/codecs/

snd-soc-seeed-voicecard.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.10.60-v7+/kernel/sound/soc/bcm/

depmod....
System has not been booted with systemd as init system (PID 1). Can't operate.
Failed to connect to bus: Host is down

DKMS: install completed.
git init
Initialized empty Git repository in /etc/voicecard/.git/
git add --all
git commit -m "origin configures"
[master (root-commit) cead452] origin configures
 7 files changed, 1482 insertions(+)
 create mode 100644 ac108_6mic.state
 create mode 100644 ac108_asound.state
 create mode 100644 asound_2mic.conf
 create mode 100644 asound_4mic.conf
 create mode 100644 asound_6mic.conf
 create mode 100644 dkms.conf
 create mode 100644 wm8960_asound.state
Created symlink /etc/systemd/system/sysinit.target.wants/seeed-voicecard.service → /lib/systemd/system/seeed-voicecard.service.
System has not been booted with systemd as init system (PID 1). Can't operate.
Failed to connect to bus: Host is down
------------------------------------------------------
Please reboot your raspberry pi to apply all settings
Enjoy!
------------------------------------------------------
```

AIWintermuteAI, 2021-09-13:
Hi there!
It seems that this particular issue is not related to reSpeaker installation procedure, but rather a problem with system set up (I found some results [here](https://www.google.com/search?q=System+has+not+been+booted+with+systemd+as+init+system+%28PID+1%29.+Can%27t+operate.+raspberry+pi&newwindow=1&client=ubuntu&hs=dBK&channel=fs&ei=Uq4-YeSxCJHE8gK_taOAAQ&oq=System+has+not+been+booted+with+systemd+as+init+system+%28PID+1%29.+Can%27t+operate.+raspberry+pi&gs_lcp=Cgdnd3Mtd2l6EAMyBwgAEEcQsAMyBwgAEEcQsAMyBwgAEEcQsAMyBwgAEEcQsAMyBwgAEEcQsAMyBwgAEEcQsAMyBwgAEEcQsAMyBwgAEEcQsANKBAhBGABQp78DWOXQA2Di1QNoAXACeACAAQCIAQCSAQCYAQCgAQHIAQjAAQE&sclient=gws-wiz&ved=0ahUKEwjkorP76vryAhURolwKHb_aCBAQ4dUDCA0&uact=5) . Could you please try with fresh installation of latest Raspberry Pi OS 32-bit and report back? I'm sure that will fix the problem.
I'll keep the issue open for some time and await for your reply.

Baggerfahrer, 2021-09-15:
thanks for the reply!
I will re-set-up the pi (potentially on the weekend) and report back.

Many thanks

Baggerfahrer, 2021-09-16:
issue can be closed.
After a clear new set-up of rpi with an sudo apt-get update before the installation of this driver, everything went well....

closed 2021-09-16 by Baggerfahrer