respeaker/seeed-voicecard #317: [Bug]: ./install.sh on 64bit Raspberry OS

diphtong, 2021-12-11:
### Describe the bug

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

### Expected behavior

install the package

### Platform

- Device: [e.g. Raspberry Pi 4]
- OS: [e.g. Raspbian OS 32bit kernel version ...]
- Version/commit number [e.g. d1816f5]

closed 2021-12-11 by diphtong

StuartIanNaylor, 2022-02-22:
```
 WARNING Your loaded kernel version is 5.10.92
 Not matching the updated version .
 Kernel was updated, but new kernel was not loaded yet
 Please reboot your machine AND THEN run this script again
```
```
pi@raspberrypi:~/seeed-voicecard $ uname -a
Linux raspberrypi 5.10.92-v8+ #1514 SMP PREEMPT Mon Jan 17 17:39:38 GMT 2022 aarch64 GNU/Linux
```

StuartIanNaylor, 2022-02-22:
sudo ./install_arm64.sh not  ./install.sh