respeaker/seeed-voicecard #318: [Bug]: Crashing when adjusting sound above 20%

StonedEdge, 2022-01-07:
### Describe the bug

1. Open a standalone emulator in RetroPie, such as DraStic or PPSSPP 
2. Adjust volume above 20% in Alsamixer
3. Emulator begins to lag - eventually crashes and file system becomes read-only
4. Doesn't happen with the headphone drivers, only speaker drivers 

### Expected behavior

The speaker drivers should allow me to drive the volume of the speakers. I have them connected to 5V on my PCB as per the datasheet. 

### Platform

- Device: [e.g. Raspberry Pi CM4]
- OS: [e.g. RetroPie 32bit kernel version 5.10.63-v71+]

closed 2022-01-07 by StonedEdge

HinTak, 2022-01-07:
why did you close this? Even you realize it is an unrelated issue, it would be nice to write it down before closing...

StonedEdge, 2022-01-07:
> why did you close this? Even you realize it is an unrelated issue, it would be nice to write it down before closing...

Hi HinTak.

I closed it as I probably realized that this issue more relates to RetroPie than the seeed development team. However, just for some pre-content, the RetroArch audio drivers seem to be using alsathread (which I have no issue with running games), however as soon as I try any X11 based backend emulator, it seems to crash due when the speaker outputs go above 20%. No problem if you cant tackle the issue, I thought it might have been a simple fix, for example changing the audio driver used on the emulator side...

StonedEdge, 2022-01-08:
I dont particularly think this will help, but here's an image of the crash error: 
[image]

And here's what my .asoundrc file contents looks like:
```
pcm.!default {
  type asym
  playback.pcm {
    type plug
    slave.pcm "output"
  }
  capture.pcm {
    type plug
    slave.pcm "input"
  }
}

pcm.output {
  type hw
  card 0
}

ctl.!default {
  type hw
  card 0
}

```

StonedEdge, 2022-01-11:
I guess no one has any idea then... :(

StonedEdge, 2022-01-11:
I am powering the speakers on 5V rail compared to 3v3 on my PCB. Do I need to change anything in the code to accommodate for the higher voltages?

HinTak, 2022-01-11:
The fat message means your sd card got corrupted. I really suspect a power issue - your power supply is not strong enough?

StonedEdge, 2022-01-12:
> The fat message means your sd card got corrupted. I really suspect a power issue - your power supply is not strong enough?

Hmm, I guess it could be that. The regulator I'm using is the TPS61232, perhaps it cannot handle the 2A+ transient spikes from the speakers. I don't have the inductors/low pass filters on the speaker rails, like the seeed PCB schematic shows. I'll send over my schematic shortly. 

For what it's worth, the PPSSPP emulator SDL and Qt builds of PPSSPP just use SDL audio rather than ALSAthread like the libretro core emulators. Not sure if this would be the source of the issue or not.

StonedEdge, 2022-01-12:
Is anything writing to the SD card in the code? I find that when my SD card becomes a read-only file system, everything speeds up considerably also.

StonedEdge, 2022-01-12:
I will also add that driving the headphones works completely fine with zero lag, however the speakers cause the slowdowns.

StonedEdge, 2022-01-12:
And finally, my schematic, FWIW[image]

HinTak, 2022-01-12:
speakers draw more power than headphone - so it sounds like a power issue?

StonedEdge, 2022-01-13:
Don't think so. Checked the power supply ripple and transient noise near the speakers with my scope and it looks fine. Is there anything that I need to manipulate in the code to drive two speakers? FWIW, muting the "Right Output Mixer PCM" in Alsamixer removes any issues I have.

HinTak, 2022-01-13:
'muting the "Right Output Mixer PCM"' works around the issue, that's interesting. If I could find time would be useful to see what that does at the code level.

StonedEdge, 2022-01-13:
> 'muting the "Right Output Mixer PCM"' works around the issue, that's interesting. If I could find time would be useful to see what that does at the code level.

Here's a video of the issue by the way, just in case: https://youtu.be/XeDUuK9mMEA

HinTak, 2022-01-14:
Thanks for letting me know! Good it is resolved!