respeaker/seeed-voicecard #319: [Bug]: seeed-4mic-voicecard not showing up in 'arecord -L'

AnduriI, 2022-02-07:
### Describe the bug

I have two RPis, which I want to use as microphones with Rhasspy. On is my test setup with a RPi Zero WH working fine. The other is an already running RPi 3B with Raspbian stretch. On the Zero I installed the standard driver and it works just fine. On the full raspi I can not get the respeaker mic to show up. I also tried to uninstall and reinstall HinTak's version with no success.
Using 'arecord -L' on my zero gives this:
```
$ arecord -L
null
    Discard all samples (playback) or generate zero samples (capture)
jack
    JACK Audio Connection Kit
pulse
    PulseAudio Sound Server
default
playback
ac108
sysdefault:CARD=seeed4micvoicec
    seeed-4mic-voicecard, bcm2835-i2s-ac10x-codec0 ac10x-codec0-0
    Default Audio Device
dmix:CARD=seeed4micvoicec,DEV=0
    seeed-4mic-voicecard, bcm2835-i2s-ac10x-codec0 ac10x-codec0-0
    Direct sample mixing device
dsnoop:CARD=seeed4micvoicec,DEV=0
    seeed-4mic-voicecard, bcm2835-i2s-ac10x-codec0 ac10x-codec0-0
    Direct sample snooping device
hw:CARD=seeed4micvoicec,DEV=0
    seeed-4mic-voicecard, bcm2835-i2s-ac10x-codec0 ac10x-codec0-0
    Direct hardware device without any conversions
plughw:CARD=seeed4micvoicec,DEV=0
    seeed-4mic-voicecard, bcm2835-i2s-ac10x-codec0 ac10x-codec0-0
    Hardware device with all software conversions
usbstream:CARD=seeed4micvoicec
    seeed-4mic-voicecard
    USB Stream Output
```
while on the other one it gives
```
$ arecord -L
null
    Discard all samples (playback) or generate zero samples (capture)
default
playback
ac108
sysdefault:CARD=Device
    USB PnP Sound Device, USB Audio
    Default Audio Device
front:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    Front speakers
surround21:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    2.1 Surround output to Front and Subwoofer speakers
surround40:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    4.0 Surround output to Front and Rear speakers
surround41:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    4.1 Surround output to Front, Rear and Subwoofer speakers
surround50:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    5.0 Surround output to Front, Center and Rear speakers
surround51:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    5.1 Surround output to Front, Center, Rear and Subwoofer speakers
surround71:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    7.1 Surround output to Front, Center, Side, Rear and Woofer speakers
iec958:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    IEC958 (S/PDIF) Digital Audio Output
dmix:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    Direct sample mixing device
dsnoop:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    Direct sample snooping device
hw:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    Direct hardware device without any conversions
plughw:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    Hardware device with all software conversions
```
There is an usb soundcard on this one as it is used as media player running squeezelite (just in case it matters).
I'm not very used to this debugging, so please give me a hint to what logs may be helpful.

Thank you in advance.

### Expected behavior

well I think it should apear as record device and also in Rhasspy

### Platform

- Device: Zero and Pi 3B
- OS: Raspbian 32bit
- Version/commit number [e.g. d1816f5]

Horschig, 2022-06-09:
I have a similar issue on my fresh raspi 4. `arecord-L` says

```
null
    Discard all samples (playback) or generate zero samples (capture)
lavrate
    Rate Converter Plugin Using Libav/FFmpeg Library
samplerate
    Rate Converter Plugin Using Samplerate Library
speexrate
    Rate Converter Plugin Using Speex Resampler
jack
    JACK Audio Connection Kit
oss
    Open Sound System
pulse
    PulseAudio Sound Server
upmix
    Plugin for channel upmix (4,6,8)
vdownmix
    Plugin for channel downmix (stereo) with a simple spacialization
playback
ac108
default
output
usbstream:CARD=Headphones
    bcm2835 Headphones
    USB Stream Output
usbstream:CARD=vc4hdmi0
    vc4-hdmi-0
    USB Stream Output
usbstream:CARD=vc4hdmi1
    vc4-hdmi-1
    USB Stream Output
```

Horschig, 2022-06-09:
Thanks @AnduriI for pointing to @HinTak. Using his/her version helped resolve my issue! 

https://github.com/HinTak/seeed-voicecard

Jem-H, 2023-01-04:
Hi @HinTak, I used the v5.9 branch of your fork for my ReSpeaker 4-mic linear array with my Raspberry Pi 3B (kernel version 5.15).  It is now detecting the sound card correctly, which is great, so thank you! However, the menu bar of my Raspberry Pi OS has now disappeared after downloading your fork (which didn't happen with the official respeaker version). I can get the menu bar back by fiddling with Raspberry Pi preferences menu (1-wire option), but then arecord/aplay/alsa can no longer detect the seeed sound card. It's not stopping my project progress to not have the menu bar as I can get around most of it with terminal/keyboard shortcuts, but it's definitely weird (also a bit annoying). Do you know anything about that? Or has anyone experienced something similar?

refractslight, 2023-01-04:
I'm having the same issue as Jem-H. Menu bar is gone after fresh install. :C  I'm using the ReSpeaker 4-mic Array on a RaspberryPi 3B+ with mine, but it seems to be the same issue.

Please let me know if I can provide more info to help!

HinTak, 2023-01-05:
I highly doubt that audio driver affects menu bar (and its disappearance) - unless the menu bar actually probes hardware, and fails and dies when it encounters unusual hardware. Anyway, just saying I always have my pi headless (without GUI) so am unlikely to look at that kind of issue, if it is even related, at all.

Jem-H, 2023-01-05:
Okay, I see, thank you both!

vslobody, 2023-02-15:
[this](https://raspberrypi.stackexchange.com/questions/122579/after-fresh-install-of-raspberry-os-the-menu-bar-is-missing-in-tightvnc-session) worked for me

Jem-H, 2023-06-02:
> Re: disappearing menu bar - [this](https://raspberrypi.stackexchange.com/questions/122579/after-fresh-install-of-raspberry-os-the-menu-bar-is-missing-in-tightvnc-session) worked for me

I can confirm that the first solution suggested by the top answer works for me! Many thanks @vslobody

Silver-Echo, 2024-02-05:
its pulseaudio that interferes with the menu bar.. just remove it and reinstall it.. also might wanna got to /seeed-voicecard/ac108_plugin/ here type make && sudo make install for the ac108 tool.. after it u mught wanna so to /seeed-voicecard/ type "sudo ./builddtbo.sh && sudo ./ubuntu-prerequisite.sh && sudo ./seeed-voicecard" and then "sudo ./install.sh"

Silver-Echo, 2024-02-05:
> I highly doubt that audio driver affects menu bar (and its disappearance) - unless the menu bar actually probes hardware, and fails and dies when it encounters unusual hardware. Anyway, just saying [...]

lets be honest.. we really love the product fysically i actually have all the respeakers haha but.. installing them and really using them more than this little install is kinda sad.. i expect a lot more from such awesome machine.. i would love to ask you to please help us, the user make full use of all potential aspects of this device and i assure you.. more people will be in love with using it .. in the beginning i was so excited.. now i am 3 month stuck on a installation.. its really sad.. u will make more money if you help us the user than just do it for the money.. we would love to buy more and better but help us install it easier and better and help us do more with it.. ur the one who made us love ur work i assume.. haha sure the hintak branch helps, but raspberry removed buster..  many aspects could be easily resolved if you wanted too.. and i would love you if you do..

HinTak, 2024-02-05:
My fork should work against current raspberrypi os. Why the mention of old buster? That said, apparently there is some problem with pi 5-based systems.