respeaker/seeed-voicecard #324: [Bug]: Kernel 5.15.32-v71 "set params:1407: Unable to install hw-params

telefonjoker100, 2022-04-30:
### Describe the bug

1. Install the latest Raspberry Pi 4 Os via Raspberry Pi Imager
2. Follow the instructions of this git and also checking out this: https://wiki.seeedstudio.com/ReSpeaker_2_Mics_Pi_HAT_Raspberry/

Additionally, I tried it with the raspberry pi image you providing in this git. The pi hat 2 worked.

### Expected behavior

Expecting the audio to record

### Platform

- Device: Raspberry Pi 4
- OS: Raspbian OS 32bit kernel 5.15.32-v71
- Pi Hat: ReSpeaker_2_Mics_Pi_HAT_Raspberry
- Version/commit number latest version

### Relevant log output

```shell
pi@raspberrypi:~ $ arecord -f cd -Dhw:3 test.wav
Recording WAVE 'test.wav' : Signed 16 bit Little Endian, Rate 44100 Hz, Stereo
arecord: set_params:1407: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 32
CHANNELS: 2
RATE: 44100
PERIOD_TIME: (125011 125012)
PERIOD_SIZE: 5513
PERIOD_BYTES: 22052
PERIODS: 4
BUFFER_TIME: (500045 500046)
BUFFER_SIZE: 22052
BUFFER_BYTES: 88208
TICK_TIME: 0

pi@raspberrypi:~ $ dmesg
...
[  917.708122] wm8960 1-001a: failed to configure clock
[  917.708148] wm8960 1-001a: ASoC: error at snd_soc_dai_hw_params on wm8960-hifi: -22
[  917.708202]  bcm2835-i2s-wm8960-hifi: ASoC: soc_pcm_hw_params() failed (-22)
```

createcandle, 2022-05-04:
I am also seeing this error, but on kernel 5.10 with  a 32 bit Bullseye OS.

yezzer, 2022-05-04:
> I am also seeing this error, but on kernel 5.10 with a 32 bit Bullseye OS. The recorded audio sounds like hissing.

Try uninstalling, then cloning the fork and installing that. Fixed the same or similar issue https://github.com/respeaker/seeed-voicecard/issues/325

createcandle, 2022-05-04:
I believe I'm using the fork already.

flatsiedatsie, 2022-05-04:
@yezzer I tried a re-install, and it still worked. Perhaps I was wrong about which version I was using.

telefonjoker100, 2022-05-07:
The fork worked perfectly for me. Thank you!