respeaker/seeed-voicecard #326: [Bug]: Respeaker not shown on Linux 5.15.32-v8+ aarch64

pauloesteban, 2022-05-18:
### Describe the bug

**To Reproduce**
Steps to reproduce the behavior:
1. Attach Respeaker Pi Hat onto the RPi
2. Power cycle
3. Install using `install_arm64.sh` script
4. See error
5. Reboot
6. `arecord -l` not showing anything

Note: I've already tried some forks without success

### Expected behavior

Error! Bad return status for module build on kernel: 5.15.32-v8+ (aarch64)
Linux 5.15.32-v8+ aarch64

### Platform

- Raspberry Pi 4B
- OS: Raspberry OS Bullseye arm64
- Version/commit number Linux 5.15.32-v8+ aarch64

### Relevant log output

```shell
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
raspberrypi-kernel is already the newest version (1:1.20220331-1).
raspberrypi-kernel-headers is already the newest version (1:1.20220331-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package linux-raspi
E: Unable to locate package linux-headers-raspi
E: Unable to locate package linux-image-raspi
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libasound2-plugins is already the newest version (1.2.2-2).
dkms is already the newest version (2.8.4-3).
git is already the newest version (1:2.30.2-1).
i2c-tools is already the newest version (4.2-1+b1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
5.15.32-v8+
5.15.32-v8+

-------- Uninstall Beginning --------
Module:  seeed-voicecard
Version: 0.3
Kernel:  5.15.32-v8+ (aarch64)
-------------------------------------

Status: Before uninstall, this module version was ACTIVE on this kernel.

snd-soc-wm8960.ko:
 - Uninstallation
   - Deleting from: /lib/modules/5.15.32-v8+/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

snd-soc-ac108.ko:
 - Uninstallation
   - Deleting from: /lib/modules/5.15.32-v8+/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

snd-soc-seeed-voicecard.ko:
 - Uninstallation
   - Deleting from: /lib/modules/5.15.32-v8+/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

depmod....

DKMS: uninstall completed.

------------------------------
Deleting module version: 0.3
completely from the DKMS tree.
------------------------------
Done.

Creating symlink /var/lib/dkms/seeed-voicecard/0.3/source ->
                 /usr/src/seeed-voicecard-0.3

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=5.15.32-v8+ -C /lib/modules/5.15.32-v8+/build M=/var/lib/dkms/seeed-voicecard/0.3/build....(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.15.32-v8+ (aarch64)
Consult /var/lib/dkms/seeed-voicecard/0.3/build/make.log for more information.
mkdir: cannot create directory '/etc/voicecard': File exists
git init
Reinitialized existing Git repository in /etc/voicecard/.git/
git add --all
git commit -m "origin configures"
On branch master
nothing to commit, working tree clean
------------------------------------------------------
Please reboot your raspberry pi to apply all settings
Enjoy!
------------------------------------------------------
```

HinTak, 2022-05-18:
Pretty sure it is the same issue filed by others and now closed: in my fork https://github.com/HinTak/seeed-voicecard there are a few updates for newer kernels not yet picked up by upstream.

miraz12, 2022-05-23:
I had the same problem trying to install drivers on my rpi4 on kernel 5.15.32-v8+. Can confirm that @HinTak's fork fixed it for me. Thank you!

StuartIanNaylor, 2022-05-23:
[@hintak ](https://github.com/HinTak) @hintak thanks as got be going on 64bit Bullseye but do you have a fix for the random channel order on record?
Every time I record the channels have rotated randomly which sort of kills any use of the 6mic circular Hat for Pi.

pauloesteban, 2022-06-03:
> I had the same problem trying to install drivers on my rpi4 on kernel 5.15.32-v8+. Can confirm that @HinTak's fork fixed it for me. Thank you!

It works

closed 2022-06-03 by pauloesteban