respeaker/seeed-voicecard #327: [Bug]: ReSpeaker 4-Mic Array Driver does not work with latest raspbian kernel

gpgmailencrypt, 2022-06-27:
### Describe the bug

**To Reproduce**
Steps to reproduce the behavior:
1. Have an existing (working) raspberry os
2. sudo apt update
After reboot the respeaker 4-Mic Array is not available any more

Building a functional driver  does also not work on a fresh installed raspberry os.
Also tried HinTak's fork with no success

### Expected behavior

dkms builds the driver without error

### Platform

- Device: Raspberry Pi 4
- OS: Raspberry Pi OS 32bit kernel version 
- Version: latest master branch

### Relevant log output

```shell
Error! Bad return status for module build on kernel: 5.15.32+ (armv7l)
Consult /var/lib/dkms/seeed-voicecard/0.3/build/make.log for more information.
.
run-parts: executing /etc/kernel/header_postinst.d/dkms 5.15.32-v7+
dkms: running auto installation service for kernel 5.15.32-v7+:
Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=5.15.32-v7+ -C /lib/modules/5.15.32-v7+/build M=/var/lib/dkms/seeed-voicecard/0.3/build....(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.15.32-v7+ (armv7l)
Consult /var/lib/dkms/seeed-voicecard/0.3/build/make.log for more information.
.
run-parts: executing /etc/kernel/header_postinst.d/dkms 5.15.32-v7l+
dkms: running auto installation service for kernel 5.15.32-v7l+:
Kernel preparation unnecessary for this kernel.  Skipping...

with make.log:

DKMS make.log for seeed-voicecard-0.3 for kernel 5.15.32-v7l+ (armv7l)
So 26. Jun 21:52:15 CEST 2022
make: Verzeichnis „/usr/src/linux-headers-5.15.32-v7l+“ wird betreten
  CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/wm8960.o
  CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/ac108.o
  CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/ac101.o
  CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.o
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c: In function ‘seeed_voice_card_dai_link_of’:
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c:463:8: error: implicit declaration of function ‘asoc_simple_parse_cpu’; did you mean ‘asoc_simple_parse_clk’? [-Werror=implicit-function-declaration]
  463 |  ret = asoc_simple_parse_cpu(cpu, dai_link, &single_cpu);
      |        ^~~~~~~~~~~~~~~~~~~~~
      |        asoc_simple_parse_clk
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c:468:8: error: implicit declaration of function ‘asoc_simple_parse_codec’; did you mean ‘asoc_simple_parse_dai’? [-Werror=implicit-function-declaration]
  468 |  ret = asoc_simple_parse_codec(codec, dai_link);
      |        ^~~~~~~~~~~~~~~~~~~~~~~
      |        asoc_simple_parse_dai
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c:480:8: error: implicit declaration of function ‘asoc_simple_parse_platform’; did you mean ‘asoc_simple_parse_daifmt’? [-Werror=implicit-function-declaration]
  480 |  ret = asoc_simple_parse_platform(plat, dai_link);
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~~
      |        asoc_simple_parse_daifmt
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c:505:8: error: implicit declaration of function ‘asoc_simple_parse_clk_cpu’; did you mean ‘asoc_simple_parse_clk’? [-Werror=implicit-function-declaration]
  505 |  ret = asoc_simple_parse_clk_cpu(dev, cpu, dai_link, cpu_dai);
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~
      |        asoc_simple_parse_clk
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c:513:8: error: implicit declaration of function ‘asoc_simple_parse_clk_codec’; did you mean ‘asoc_simple_parse_clk’? [-Werror=implicit-function-declaration]
  513 |  ret = asoc_simple_parse_clk_codec(dev, codec, dai_link, codec_dai);
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |        asoc_simple_parse_clk
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c:546:31: error: passing argument 1 of ‘asoc_simple_canonicalize_cpu’ from incompatible pointer type [-Werror=incompatible-pointer-types]
  546 |  asoc_simple_canonicalize_cpu(dai_link, single_cpu);
      |                               ^~~~~~~~
      |                               |
      |                               struct snd_soc_dai_link *
In file included from /var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c:28:
./include/sound/simple_card_utils.h:158:70: note: expected ‘struct snd_soc_dai_link_component *’ but argument is of type ‘struct snd_soc_dai_link *’
  158 | void asoc_simple_canonicalize_cpu(struct snd_soc_dai_link_component *cpus,
      |                                   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c:548:36: error: passing argument 1 of ‘asoc_simple_canonicalize_platform’ from incompatible pointer type [-Werror=incompatible-pointer-types]
  548 |  asoc_simple_canonicalize_platform(dai_link);
      |                                    ^~~~~~~~
      |                                    |
      |                                    struct snd_soc_dai_link *
In file included from /var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c:28:
./include/sound/simple_card_utils.h:156:75: note: expected ‘struct snd_soc_dai_link_component *’ but argument is of type ‘struct snd_soc_dai_link *’
  156 | void asoc_simple_canonicalize_platform(struct snd_soc_dai_link_component *platforms,
      |                                        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c:548:2: error: too few arguments to function ‘asoc_simple_canonicalize_platform’
  548 |  asoc_simple_canonicalize_platform(dai_link);
      |  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c:28:
./include/sound/simple_card_utils.h:156:6: note: declared here
  156 | void asoc_simple_canonicalize_platform(struct snd_soc_dai_link_component *platforms,
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
At top level:
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c:263:12: warning: ‘asoc_simple_parse_dai’ defined but not used [-Wunused-function]
  263 | static int asoc_simple_parse_dai(struct device_node *node,
      |            ^~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[1]: *** [scripts/Makefile.build:277: /var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.o] Fehler 1
make[1]: *** Es wird auf noch nicht beendete Prozesse gewartet....
make: *** [Makefile:1868: /var/lib/dkms/seeed-voicecard/0.3/build] Fehler 2
make: Verzeichnis „/usr/src/linux-headers-5.15.32-v7l+“ wird verlassen
```

HinTak, 2022-06-28:
Fixed months ago in https://github.com/HinTak/seeed-voicecard .

jbroeckling, 2022-07-07:
> Fixed months ago in https://github.com/HinTak/seeed-voicecard .

yea, just tried your repo and it works with newer heades and on 64 Bit Raspi. Thanks :)

luigi-riefolo, 2022-08-05:
same here, tried @HinTak's fork and it's not working. Any idea?

HinTak, 2022-08-05:
Little vague. If you see the same messages above, your are out of date. Otherwise, what `dmesg` says after boot up, and `arecord -l` would be useful.

FluffyKoala12, 2022-08-22:
Hi, I am using a Raspberry Pi Zero W along with the ReSpeaker 4-mic linear array HAT. I have followed all instructions for getting started on Wiki Seeed Studio website (https://wiki.seeedstudio.com/ReSpeaker_4-Mic_Linear_Array_Kit_for_Raspberry_Pi/) but I am getting a lot of warnings/errors when running this file. The output when running  python3 recording_examples/get_device_index.py is as follows:

ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib pcm_a52.c:823:(_snd_pcm_a52_open) a52 is only for playback
ALSA lib conf.c:4871:(parse_args) Unknown parameter AES0
ALSA lib conf.c:5031:(snd_config_expand) Parse arguments error: No such file or                                                                                         directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM iec958:{AES0 0x6 AES1 0x                                                                                        82 AES2 0x0 AES3 0x2  CARD 0}
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib pcm_dmix.c:1043:(snd_pcm_dmix_open) The dmix plugin supports only playb                                                                                        ack stream
Input Device id  0  -  seeed-8mic-voicecard: bcm2835-i2s-ac10x-codec0 ac10x-code                                                                                        c0-0 (hw:0,0)
Input Device id  1  -  sysdefault
Output Device id  2  -  front
Output Device id  3  -  rear
Output Device id  4  -  center_lfe
Output Device id  5  -  side
Input Device id  6  -  surround21
Output Device id  7  -  surround40
Output Device id  8  -  surround41
Output Device id  9  -  surround50
Output Device id  10  -  surround51
Output Device id  11  -  surround71
Output Device id  12  -  iec958
Input Device id  13  -  spdif
Input Device id  14  -  hdmi
Input Device id  15  -  modem
Input Device id  16  -  phoneline
Input Device id  17  -  lavrate
Input Device id  18  -  samplerate
Input Device id  19  -  speexrate
Input Device id  20  -  pulse
Input Device id  21  -  upmix
Input Device id  22  -  vdownmix
Input Device id  23  -  ac108
Output Device id  24  -  dmixer
Input Device id  25  -  ac101
Output Device id  26  -  dmix
Input Device id  27  -  default

I first started using the 32-bit Bullseye OS and after having little success I have also tried with the 32-bit Buster OS. Nothing has changed and I can't find solutions on any forums - a lot of people seem to have the same problem.

Any help would be greatly appreciated !

HinTak, 2022-08-22:
Two things:

- there are two 4-mics devices - a square one which gives 4 channel outputs, and a linear one, which gives  8 channel outputs with 4 silent channels. They are very different. 

- the message above seems to suggests some other audio devices are also plugged in, or was plugged in in the past. Those demo script often assumes the seeedstudio device is the first audio device, which may not be the case if you have/had multiple devices connected in the past or currently.

On the 2nd point, the linux kernel itself has changed slightly to allow multi-seats (two displays two keyboards for two users on the same pi). As a result audio device enumerations changed - back when those scripts etc were written, it was a setting to one logical audio device to direct audio between on-board phone jack, hdmi audio, or the seeed studio device. Now they are 3 separate logical audio devices. Your scripts might need updating.

Disclaimer: I don't work for seeed  studio. Just another user.

Feles-Mala, 2022-09-19:
Hi there, I suspect this is going to be an eternal struggle of driver vs os, which makes this device unsuitable for a long-lifetime project but I'll give it a go:

I have a raspberry pi 4B 4gb and have tried both 32bit and 64bit versions of current raspberry pi os (Bullseye).

Also tried both the official (64 and 32bit) and the HinTak fork (default branch 5.9) of the driver installs.

In all cases in which the installation succeeds:

1. the lxde panel task menu crash/restarts about 4 times and then stays dead
2. only 2 channels appear in audacity

Uninstalling and rebooting resolves the lxpanel issue.

Checking the lxpanel .cache file shows a sequence of messages akin to: 'lxpanel CRITICAL dhcpcd_open inappropriate ioctl for device.

I would actually be ok with two channels for now if I could have the lxpanel working. Any suggestions welcome; I appreciate that the most active contributor doesn't actually work for them :)

squinty204, 2022-09-20:
I too am experiencing the same problem as participant Feles-Mala. We have been chatting about this bug over on reddit and have been experimenting to solve this issue.  Today I decided to try installing the 4mic driver on a brand new installation of Ubuntu 22.04 but while the panel doesn't disappear after the driver installation, Alsamixer still doesn't display the card as being installed.  I also became aware today that a "4mics_hat" directory had been installed but it's external to the seeed-voicecard directory.  Would following the README.md instructions located in the "4mics_hat"  be of any use towards solving this problem?

HinTak, 2022-11-15:
I think upstream supports up to v5.12 . There is adjustments required for the kernel for 5.13, which in my fork was done in a "if kernel >=5.13" manner so my v5.9 branch supports v5.9 to v5.19, actually. (Upstream only up to 5.12, and not beyond). There is a separate v6.0 branch in my repo for v6.0+, because the difference between 5.19 and 6.0 is a bit large and ugly if I have to do a lot of "if kernel >5.19 ... else...".

Summary, 5.12 or below, either mine or upstream seedstudio should work. If you have v5.13 to v5.19, use my default v5.9 branch (I actually have alias to v5.13 etc branches to the same thing) - upstream won't work.if you are using v6.0+, use my 6.0 branch.

HinTak, 2023-03-23:
Just for the record, the lxpanel crash is pulseaudio related and not specific the seeed driver. See for example , 
 https://forums.raspberrypi.com/viewtopic.php?t=295641 and possibly elsewhere too. Apparently quite widely reported.

stjq12, 2024-05-22:
> I think upstream supports up to v5.12 . There is adjustments required for the kernel for 5.13, which in my fork was done in a "if kernel >=5.13" manner so my v5.9 branch supports v5.9 to v5.19, [...]

My rasp OS is bullseye and the kernel version is 5.10.92. But when I run `sudo ./install.sh` in seeed-voicecard, the program updates kernel automatically to 6.1. 

Logout is below:

```
pi@raspberrypi:~/seeed-voicecard $ sudo ./install.sh
Get:1 http://security.debian.org/debian-security bullseye-security InRelease [48.4 kB]
Get:2 http://deb.debian.org/debian bullseye InRelease [116 kB]                 
Get:3 http://archive.raspberrypi.org/debian bullseye InRelease [23.6 kB]       
Get:4 http://security.debian.org/debian-security bullseye-security/main arm64 Packages [269 kB]
Get:5 http://deb.debian.org/debian bullseye-updates InRelease [44.1 kB]
Get:6 http://deb.debian.org/debian bullseye/main arm64 Packages [7957 kB]
Get:7 http://archive.raspberrypi.org/debian bullseye/main arm64 Packages [307 kB]
Get:8 http://archive.raspberrypi.org/debian bullseye/main armhf Packages [313 kB]
Get:9 http://deb.debian.org/debian bullseye/main armhf Packages [7842 kB]      
Get:10 http://deb.debian.org/debian bullseye/main Translation-en [6236 kB]     
Get:11 http://deb.debian.org/debian bullseye/contrib arm64 Packages [40.8 kB]  
Get:12 http://deb.debian.org/debian bullseye/contrib armhf Packages [40.1 kB]  
Get:13 http://deb.debian.org/debian bullseye/contrib Translation-en [46.9 kB]  
Get:14 http://deb.debian.org/debian bullseye/non-free armhf Packages [56.0 kB] 
Get:15 http://deb.debian.org/debian bullseye/non-free arm64 Packages [72.3 kB] 
Get:16 http://deb.debian.org/debian bullseye/non-free Translation-en [92.2 kB] 
Get:17 http://deb.debian.org/debian bullseye-updates/main armhf Packages [16.3 kB]
Get:18 http://deb.debian.org/debian bullseye-updates/main arm64 Packages [16.3 kB]
Get:19 http://deb.debian.org/debian bullseye-updates/main Translation-en [10.9 kB]
Get:20 http://security.debian.org/debian-security bullseye-security/main armhf Packages [270 kB]
Get:21 http://security.debian.org/debian-security bullseye-security/main Translation-en [176 kB]                    
Get:22 http://security.debian.org/debian-security bullseye-security/non-free Translation-en [472 B]                 
Fetched 24.0 MB in 1min 29s (269 kB/s)                                                                              
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
344 packages can be upgraded. Run 'apt list --upgradable' to see them.
N: Repository 'http://security.debian.org/debian-security bullseye-security InRelease' changed its 'Suite' value from 'stable-security' to 'oldstable-security'
N: Repository 'http://deb.debian.org/debian bullseye InRelease' changed its 'Version' value from '11.2' to '11.9'
N: Repository 'http://deb.debian.org/debian bullseye InRelease' changed its 'Suite' value from 'stable' to 'oldstable'
N: Repository 'http://archive.raspberrypi.org/debian bullseye InRelease' changed its 'Suite' value from 'stable' to 'oldstable'
N: Repository 'http://deb.debian.org/debian bullseye-updates InRelease' changed its 'Suite' value from 'stable-updates' to 'oldstable-updates'
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following package was automatically installed and is no longer required:
  libfuse2
Use 'sudo apt autoremove' to remove it.
The following NEW packages will be installed:
  raspberrypi-kernel-headers
The following packages will be upgraded:
  raspberrypi-kernel
1 upgraded, 1 newly installed, 0 to remove and 343 not upgraded.
Need to get 37.3 MB of archives.
After this operation, 5094 kB of additional disk space will be used.
Get:1 http://archive.raspberrypi.org/debian bullseye/main arm64 raspberrypi-kernel arm64 1:1.20230405-1 [27.0 MB]
Get:2 http://archive.raspberrypi.org/debian bullseye/main arm64 raspberrypi-kernel-headers arm64 1:1.20230405-1 [10.2 MB]
Fetched 37.3 MB in 7s (5420 kB/s)                                                                                   
apt-listchanges: Can't set locale; make sure $LC_* and $LANG are correct!
apt-listchanges: Reading changelogs...
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = (unset),
	LC_ALL = (unset),
	LANG = "en_GB.UTF-8"
    are supported and installed on your system.

```

HinTak, 2024-05-22:
It is as I already stated, use the v5.9 branch , https://github.com/HinTak/seeed-voicecard/tree/v5.9 , and just do `sudo make install` instead of `sudo ./install.sh` . `make install` won't try to upgrade your kernel, but you have to install the compiler, kernel headers etc manually. It is one of the steps inside the `install.sh` scripts, and the only important one.

stjq12, 2024-05-23:
Thx! Should I choice branch v6.1 If I wanna use the OS with kernel v6.1?

HinTak, 2024-05-23:
> Thx! Should I choice branch v6.1 If I wanna use the OS with kernel v6.1?

Yes.