respeaker/seeed-voicecard #329: [Bug]:  Respeaker 4 mic linear - channel not in order

souviksaha97, 2022-08-29:
### Describe the bug

**To Reproduce**
Steps to reproduce the behavior:
1. Follow the steps for installation from the README.md, from the master branch
2. Use the 4 mic linear ReSpeaker Microphone
3. Run Audacity software, select input as SeedStudio voice card
4. Choose channel as 8 in Audacity
5. Out of the 8 channels we don't see microphone data in channels 1, 2, 3, 4. Rather it is disordered without specific pattern.

### Expected behavior

Mic channel 1, 2, 3, 4 - output channel 1, 2, 3, 4
5, 6th channel empty
8th, 7th channel playback from DAC

### Platform

- Device: Raspberry Pi 3B
- OS: Buster kernel no - 5.10.103-v7+
- Version/commit number -  master branch (seedvoice-card) c52606626de050bdad85803d7e427a64cb0cf05c

HinTak, 2022-08-29:
Possibly dup of https://github.com/respeaker/seeed-voicecard/issues/309

souviksaha97, 2022-08-30:
What was the resolution for the issue @HinTak ?