respeaker/seeed-voicecard #331: [Bug]: install.sh unable to determine kernel version

mkatliar, 2022-12-06:
### Describe the bug

**To Reproduce**
Steps to reproduce the behavior:
```bash
sudo ./install.sh
```
outputs the message
```
WARNING Your loaded kernel version is 5.15.0
 Not matching the updated version .
 Kernel was updated, but new kernel was not loaded yet
 Please reboot your machine AND THEN run this script again
```
and does not start building the driver. 
Same error after rebooting.

The output "Not matching the updated version ." (empty version string) suggests that the script is unable to determine the kernel version (`get_kernel_version()`).

### Expected behavior

Drive builds and installs.

### Platform

- Device: Raspberry Pi 2
- OS: Ubuntu 22.04.1 LTS 5.15.0-1021-raspi
- Commit number: c526066

### Relevant log output

```shell
pi@raspberry2:~/seeed-voicecard$ sudo ./install.sh 
Hit:1 http://ports.ubuntu.com/ubuntu-ports jammy InRelease
Hit:2 http://ports.ubuntu.com/ubuntu-ports jammy-updates InRelease
Hit:3 http://ports.ubuntu.com/ubuntu-ports jammy-backports InRelease
Hit:4 http://ports.ubuntu.com/ubuntu-ports jammy-security InRelease
Hit:5 http://packages.ros.org/ros2/ubuntu jammy InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
1 package can be upgraded. Run 'apt list --upgradable' to see it.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package raspberrypi-kernel-headers
E: Unable to locate package raspberrypi-kernel
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
linux-headers-raspi is already the newest version (5.15.0.1021.18).
linux-image-raspi is already the newest version (5.15.0.1021.18).
linux-raspi is already the newest version (5.15.0.1021.18).
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libasound2-plugins is already the newest version (1.2.6-1).
i2c-tools is already the newest version (4.3-2build1).
dkms is already the newest version (2.8.7-2ubuntu2.1).
git is already the newest version (1:2.34.1-1ubuntu1.5).
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.

gzip: stdin: not in gzip format
dpkg-query: package 'raspberrypi-kernel-headers' is not installed
Use dpkg --contents (= dpkg-deb --contents) to list archive files contents.

gzip: stdin: not in gzip format
------------------------------------------------------
 WARNING Your loaded kernel version is 5.15.0
 Not matching the updated version .
 Kernel was updated, but new kernel was not loaded yet
 Please reboot your machine AND THEN run this script again
```

HinTak, 2022-12-07:
Read my comment 
https://github.com/respeaker/seeed-voicecard/issues/327#issuecomment-1315641736

mkatliar, 2022-12-07:
> Read my comment [#327 (comment)](https://github.com/respeaker/seeed-voicecard/issues/327#issuecomment-1315641736)

It did work with v6.0 branch from your fork, thank you!

Shouldn't the changes from that branch be merged to the master of the main repo?

HinTak, 2022-12-10:
This is probably an FAQ - the main reason it being an on-going process (to update for current kernel) so there is no fixed point for a pul; the other is that, while I like to keep it up to date as a way of following kernel development, and I like to do with well, clearly and documented properly why each individual change happens and at what kernel version (unlike some of the whole-sale change style upstream does), I don't actually test the change on real hardware. My Pi box is not operating for other reasons atm.

mkatliar, 2022-12-10:
Thank you for the explanation. I close the issue.

closed 2022-12-10 by mkatliar