respeaker/seeed-voicecard #332: [Bug]: ReSpeaker 4-Mic Array Driver does not work with 64bit Raspbian kernel 5.15.84-v8+

vrai, 2023-01-09:
### Describe the bug

**To Reproduce**
Steps to reproduce the behavior:
1. Update Raspbian to latest version (in this case 5.15.84-v8+).
2. Follow the instructions on https://wiki.seeedstudio.com/ReSpeaker_4_Mic_Array_for_Raspberry_Pi/ up to the `sudo ./install.sh` step.
3. Run `sudo ./install.sh`.
4. Build will fail, with the error written to `/var/lib/dkms/seeed-voicecard/0.3/build/make.log`.

### Expected behavior

Driver is built and installed correctly.

### Platform

- Device: Raspberry Pi 3 Model B Plus Rev 1.3
- OS: Debian GNU/Linux 11 (bullseye) aarch64
- Version/commit number: Latest branch from https://github.com/Seeed-Projects/seeed-voicecard.git.

### Relevant log output

```shell
DKMS make.log for seeed-voicecard-0.3 for kernel 5.15.84-v8+ (aarch64)
Mon  9 Jan 18:42:33 GMT 2023
make: Entering directory '/usr/src/linux-headers-5.15.84-v8+'
  CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/wm8960.o
  CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/ac108.o
  CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/ac101.o
  CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.o
/var/lib/dkms/seeed-voicecard/0.3/build/ac108.c:1148:3: error: ‘const struct snd_soc_dai_ops’ has no member named ‘digital_mute’
 1148 |  .digital_mute = ac108_aif_mute,
      |   ^~~~~~~~~~~~
/var/lib/dkms/seeed-voicecard/0.3/build/ac108.c:1148:18: error: initialization of ‘int (*)(struct snd_pcm_substream *, int,  struct snd_soc_dai *)’ from incompatible pointer type ‘int (*)(struct snd_soc_dai *, int)’ [-Werror=incompatible-pointer-types]
 1148 |  .digital_mute = ac108_aif_mute,
      |                  ^~~~~~~~~~~~~~
/var/lib/dkms/seeed-voicecard/0.3/build/ac108.c:1148:18: note: (near initialization for ‘ac108_dai_ops.bespoke_trigger’)
In file included from /var/lib/dkms/seeed-voicecard/0.3/build/wm8960.c:28:
/var/lib/dkms/seeed-voicecard/0.3/build/wm8960.c: In function ‘wm8960_configure_clocking’:
/var/lib/dkms/seeed-voicecard/0.3/build/sound-compatible-4.18.h:34:36: error: implicit declaration of function ‘snd_soc_component_read32’; did you mean ‘snd_soc_component_read’? [-Werror=implicit-function-declaration]
   34 | #define snd_soc_read               snd_soc_component_read32
      |                                    ^~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/seeed-voicecard/0.3/build/wm8960.c:619:15: note: in expansion of macro ‘snd_soc_read’
  619 |  u16 iface1 = snd_soc_read(codec, WM8960_IFACE1);
      |               ^~~~~~~~~~~~
cc1: some warnings being treated as errors
make[1]: *** [scripts/Makefile.build:289: /var/lib/dkms/seeed-voicecard/0.3/build/ac108.o] Error 1
make[1]: *** Waiting for unfinished jobs....
/var/lib/dkms/seeed-voicecard/0.3/build/wm8960.c: At top level:
/var/lib/dkms/seeed-voicecard/0.3/build/wm8960.c:1239:3: error: ‘const struct snd_soc_dai_ops’ has no member named ‘digital_mute’
 1239 |  .digital_mute = wm8960_mute,
      |   ^~~~~~~~~~~~
/var/lib/dkms/seeed-voicecard/0.3/build/wm8960.c:1239:18: error: initialization of ‘int (*)(struct snd_pcm_substream *, struct snd_soc_dai *)’ from incompatible pointer type ‘int (*)(struct snd_soc_dai *, int)’ [-Werror=incompatible-pointer-types]
 1239 |  .digital_mute = wm8960_mute,
      |                  ^~~~~~~~~~~
/var/lib/dkms/seeed-voicecard/0.3/build/wm8960.c:1239:18: note: (near initialization for ‘wm8960_dai_ops.prepare’)
cc1: some warnings being treated as errors
make[1]: *** [scripts/Makefile.build:289: /var/lib/dkms/seeed-voicecard/0.3/build/wm8960.o] Error 1
make: *** [Makefile:1902: /var/lib/dkms/seeed-voicecard/0.3/build] Error 2
make: Leaving directory '/usr/src/linux-headers-5.15.84-v8+'
```

HinTak, 2023-01-10:
Same as #327 - read https://github.com/respeaker/seeed-voicecard/issues/327#issuecomment-1315641736

vrai, 2023-01-11:
> #327 (comment)

Thanks for the response. I tried using the suggested branch (v5.9) but got exactly the same behaviour (as the v5.9 branch is the default I assume I was using that one anyway). I also attempted to use your version (https://github.com/HinTak/seeed-voicecard) at v5.9, and got the same errors.

For reference the full output of install.sh follows. It's complaining about unknown modules and a missing `/boot/kernel.img`, so perhaps the version of Raspbian I'm running (installed from the non-GUI image IIRC) is just missing something required. However until the last time I ran update/upgrade everything works, so that seems a bit unlikely.

```
sudo ./install.sh
Hit:1 http://archive.raspberrypi.org/debian bullseye InRelease
Hit:2 http://deb.debian.org/debian bullseye InRelease
Hit:3 http://deb.debian.org/debian bullseye-updates InRelease
Hit:4 http://security.debian.org/debian-security bullseye-security InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
raspberrypi-kernel is already the newest version (1:1.20230106-1).
raspberrypi-kernel-headers is already the newest version (1:1.20230106-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package linux-raspi
E: Unable to locate package linux-headers-raspi
E: Unable to locate package linux-image-raspi
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libasound2-plugins is already the newest version (1.2.2-2).
dkms is already the newest version (2.8.4-3).
git is already the newest version (1:2.30.2-1).
i2c-tools is already the newest version (4.2-1+b1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
grep: /boot/kernel.img: No such file or directory

gzip: stdin: unexpected end of file
grep: /boot/kernel.img: No such file or directory

gzip: stdin: unexpected end of file

------------------------------
Deleting module version: 0.3
completely from the DKMS tree.
------------------------------
Done.

Creating symlink /var/lib/dkms/seeed-voicecard/0.3/source ->
                 /usr/src/seeed-voicecard-0.3

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...
applying patch back-to-v5.8.diff...patching file ac108.c
patching file sound-compatible-4.18.h
patching file wm8960.c

Building module:
cleaning build area...
make -j4 KERNELRELEASE=5.15.84-v8+ -C /lib/modules/5.15.84-v8+/build M=/var/lib/dkms/seeed-voicecard/0.3/build.....(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.15.84-v8+ (aarch64)
Consult /var/lib/dkms/seeed-voicecard/0.3/build/make.log for more information.
mkdir: cannot create directory ‘/etc/voicecard’: File exists
git init
Reinitialized existing Git repository in /etc/voicecard/.git/
git add --all
git commit -m "origin configures"
On branch master
nothing to commit, working tree clean
------------------------------------------------------
Please reboot your raspberry pi to apply all settings
Enjoy!
------------------------------------------------------
```

vrai, 2023-01-11:
After further playing around I noticed that the install script wasn't using the source from the git repo to build the module, but what was already present in /lib/modules/. So instead I just did a `make`, `sudo make install` and everything sprang back to life.

closed 2023-01-11 by vrai

mikeysklar, 2023-04-11:
I'm seeing this issue as reported by an Adafruit customer and reproduced it myself. The problem is that the current Bullseye release uses kernel 6.1.28-v8+, but the kernel headers are not available. There is no /lib/modules/6.1.28-v8+/build link off to a /usr/src/linux-headers-6.1.28-v8+. The headers are available for the 6.1.21* kernels in /usr/src. 

Would it make sense to rollback to the a 6.1.21 kernel by changing /boot/config.txt or is there easy way to install the src files specific to the 6.1.28-v8 kernel when they are not available via apt?

`$ apt search linux-headers-$(uname -r)
Sorting... Done
Full Text Search... Done
`

HinTak, 2023-04-11:
See https://github.com/HinTak/RaspberryPi-Dev/blob/master/Downgrading-Pi-Kernel.md for manually downgrading the kernel with matching kernel headers, and how to lock the version against the auto- upgrade bits in the seeed studio driver's install script. (And the url for past versions of kernels)

mikeysklar, 2023-04-13:
@HinTak - While rolling back to and pinning to an earlier kernel is a reasonable work around this problem is different than previous kernel compatibility issues. The problem is being caused by Bullseye updating the 32-bit OS to a 64-bit kernel so the headers are not being installed. @alexcorvis84 did most of the heavy lifting over the last few days and came up with some work arounds with @makermelissa: 

https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/issues/247

`Raspberry Pi recently made a breaking change where all 32-bit installs are automatically updated to a 64-bit install once you do an apt update/upgrade. I was planning on testing the BrainCraft HAT more thoroughly soon. Here are some things you may wish to try:

* Try adding arm_64bit=0 to /boot/config.txt
* Install the 64-bit Raspberry Pi OS from the beginning and try it.
* Install the Microphone drivers on a Fresh 32-bit install before ever running a sudo apt update/upgrade. This may require some script modification though and can be trickier.

See [https://github.com/adafruit/Raspberry-P ... issues/247](https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/issues/247) for more related details.`

https://forums.adafruit.com/viewtopic.php?t=200461

HinTak, 2023-04-13:
@mikeysklar I think you are mistaken. It isn't not a 32-bit vs 64-bit issue, but you seem to have gotten your 6.1.28 kernel from an unusual place? The official repo from which apt gets packages is: 
http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/  . I happened to have taken the 0405 package apart a few days ago, and it is 6.1.21 . While I only looked at the 32-bit one, I 'd assume the 64-bit one is the same kernel version, and the accompanying header package is available. My question is really, where did your 6.1.28 come from?

mikeysklar, 2023-04-13:
It doesn't help that I mistyped the kernel version. I'm running the same default update as everyone else on Bullseye which is the new 64-bit kernel on the 32-bit OS.

`$ uname -a
Linux basementpi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`

HinTak, 2023-04-13:
I think the question is if you manually the arm64 header package, does it work? If it does, then the pi people should get that to accompany the 64-bit kernel  package, instead of somehow having a 64-bit kernel but 32-bit kernel headers.

HinTak, 2023-04-13:
The x86 people have had it working okay for years, 64-bit kernel with bi-arch userspace. So if the pi switch to a 64-bit kernel, they should switch to 64-bit kernel headers too. Regardless of what userspace does.