respeaker/seeed-voicecard #333: [Bug]:  Seeed-Voiceard broken on latest RPIZ2W & 84 kernel released  a week ago

ramfish, 2023-01-14:
### Describe the bug

Getting: “failed to load overlay i2s-mmap” 

**To Reproduce**
Clean install the OS on RPIZ2W
Clone and install.sh
the install script will ask you to reboot to upgrade Kernel and another reboot to finish install
“failed to load overlay i2s-mmap”  will appear after boot + log files confirm a problem

I also posted on https://github.com/raspberrypi/firmware/issues/1771

Any suggestion on how resolve this?

### Expected behavior

Audio wouldnt play... Was working on an RPI OS image I made a month ago without an issue. After some investogation seems like the new clean install or RPI  and then SeeedVoiceCard is causing the fail to load I2S-MMAP error

### Platform

RPI Zero 2W
WM8690 Audio board

### Relevant log output

```shell
Let me know if you need specific log files
```

pelwell, 2023-01-14:
[ Duplicated from https://github.com/raspberrypi/linux/issues/5319 ]

Wow - that overlay (and the driver change that supports it) hasn't existed since rpi-4.4.y, i.e. around 2017. My understanding based on this comment (https://github.com/raspberrypi/linux/issues/1004#issuecomment-297010950) is that MMAP is enabled by default so there is no need for the overlay.

Having looked now, rpi-5.15.y still contains code in `dmaengine_pcm_set_runtime_hwparams` that sets `SNDRV_PCM_INFO_MMAP` and `SNDRV_PCM_INFO_MMAP_VALID`, so there should be no need for the overlay and driver patch.

ramfish, 2023-01-14:
Thanks. That said: (1) Before the upgarde to 83, didn't get the error message. On a system I built a month ago (Clean RPI on Z2W then Seeed install, sound as working). Now, after I install Seeed on a clean RPI I eee the error and sound is not working.
What log files ould help you figure the issue out? Give me some guidance and I will submit whatever debug info would help.

ramfish, 2023-01-15:
some more info in case it helps:
pi@raspberrypi:~ $ alsactl restore
alsactl: state_lock:125: file /var/lib/alsa/asound.state lock error: File exists
alsactl: load_state:1683: Cannot open /var/lib/alsa/asound.state for reading: File exists
alsa-lib main.c:1014:(snd_use_case_mgr_open) error: failed to import hw:0 use case configuration -2
Found hardware: "vc4-hdmi" "" "" "" ""
Hardware is initialized using a generic method
alsa-lib main.c:1014:(snd_use_case_mgr_open) error: failed to import hw:1 use case configuration -2
Found hardware: "simple-card" "" "" "" ""
Hardware is initialized using a generic method

**From /var/log**
Jan 14 16:03:27 raspberrypi seeed-voicecard[428]: + kernel_ver=5.15.84-v7+
Jan 14 16:03:27 raspberrypi seeed-voicecard[428]: + [[ 5.15.84-v7+ =~ ^4\.19.*$ ]]
Jan 14 16:03:27 raspberrypi seeed-voicecard[428]: + [[ 5.15.84-v7+ =~ ^5\.*$ ]]
Jan 14 16:03:27 raspberrypi seeed-voicecard[428]: + dtoverlay -d /boot/overlays seeed-2mic-voicecard
Jan 14 16:03:27 raspberrypi udisksd[416]: Failed to load the 'mdraid' libblockdev plugin
<ay removed, property: /soc/i2s@7e203000/status
Jan 14 16:03:28 raspberrypi kernel: [   15.245497] OF: overlay: WARNING: memory leak will occur if overlay rem>
Jan 14 16:03:28 raspberrypi kernel: [   15.245546] OF: overlay: WARNING: memory leak will occur if overlay rem>
Jan 14 16:03:28 raspberrypi kernel: [   15.245570] OF: overlay: WARNING: memory leak will occur if overlay rem>
Jan 14 16:03:28 raspberrypi kernel: [   15.245593] OF: overlay: WARNING: memory leak will occur if overlay rem>
Jan 14 16:03:28 raspberrypi kernel: [   15.245617] OF: overlay: WARNING: memory leak will occur if overlay rem>
Jan 14 16:03:28 raspberrypi kernel: [   15.245640] OF: overlay: WARNING: memory leak will occur if overlay rem>
Jan 14 16:03:28 raspberrypi kernel: [   15.245665] OF: overlay: WARNING: memory leak will occur if overlay rem>
Jan 14 16:03:28 raspberrypi seeed-voicecard[428]: + echo 'create seeed-2mic-voicecard asound configure file'
Jan 14 16:03:28 raspberrypi seeed-voicecard[428]: create seeed-2mic-voicecard asound configure file
Jan 14 16:03:28 raspberrypi seeed-voicecard[428]: + ln -s /etc/voicecard/asound_2mic.conf /etc/asound.conf

also from var/log:
Jan 14 16:03:35 raspberrypi kernel: [   23.176761] wm8960 1-001a: failed to configure clock
Jan 14 16:03:35 raspberrypi kernel: [   23.176787] wm8960 1-001a: ASoC: error at snd_soc_dai_hw_params on wm89>
Jan 14 16:03:35 raspberrypi kernel: [   23.176835]  bcm2835-i2s-wm8960-hifi: ASoC: error at soc_pcm_hw_params >
Jan 14 16:03:35 raspberrypi kernel: [   23.177305] wm8960 1-001a: failed to configure clock
Jan 14 16:03:35 raspberrypi kernel: [   23.177316] wm8960 1-001a: ASoC: error at snd_soc_dai_hw_params on wm89>
Jan 14 16:03:35 raspberrypi kernel: [   23.177336]  bcm2835-i2s-wm8960-hifi: ASoC: error at soc_pcm_hw_params >
Jan 14 16:03:35 raspberrypi kernel: [   23.177858] wm8960 1-001a: failed to configure clock
Jan 14 16:03:35 raspberrypi kernel: [   23.177869] wm8960 1-001a: ASoC: error at snd_soc_dai_hw_params on wm89>
Jan 14 16:03:35 raspberrypi kernel: [   23.177889]  bcm2835-i2s-wm8960-hifi: ASoC: error at soc_pcm_hw_params >
Jan 14 16:03:35 raspberrypi kernel: [   23.178324] wm8960 1-001a: failed to configure clock

And:
Jan 14 16:03:50 raspberrypi pulseaudio[703]: Failed to find a working profile.
Jan 14 16:03:50 raspberrypi pulseaudio[703]: Failed to load module "module-alsa-card" (argument: "device_id="1>
Jan 14 16:03:50 raspberrypi bluetoothd[629]: Endpoint registered: sender=:1.30 path=/MediaEndpoint/A2DPSource/>
Jan 14 16:03:50 raspberrypi pulseaudio[703]: Failed to find a working profile.
Jan 14 16:03:50 raspberrypi pulseaudio[703]: Failed to load module "module-alsa-card" (argument: "device_id="1>
Jan 14 16:03:50 raspberrypi pulseaudio[703]: Failed to find a working profile.
Jan 14 16:03:50 raspberrypi pulseaudio[703]: Failed to load module "module-alsa-card" (argument: "device_id="1>

ramfish, 2023-01-15:
After downgrading kernel - audio works:
Before:

uname -a
Linux raspberrypi 5.15.84-v7+ #1613 SMP Thu Jan 5 11:59:48 GMT 2023 armv7l GNU/Linux

sudo rpi-update a1658b86485225036eeaea7e0ad4438ed1ce3dd2
Linux raspberrypi 5.10.74-v7+ #1471 SMP Mon Oct 18 17:06:24 BST 2021 armv7l GNU/Linux

Now I need to go to 15.6X and try it... having problems with RPI-Update telling me the hash  I use are wrong

UmbrellaCodr, 2023-01-17:
➤  uname -a                                                                                      127 ↵
```
Linux hex 5.15.84-v8+ #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023 aarch64 GNU/Linux
```

So figured I would post on this thread in hopes to save the next person some time:

I was able to get this to build and install using the HinTak fork:
```
cd /var/lib/dkms/seeed-voicecard/0.3/source
vim dkms.conf
```

delete the following lines:
```
PATCH[0]="back-to-v4.19.diff"
PATCH[1]="back-to-v5.4.diff"
PATCH[2]="back-to-v5.8.diff"
PATCH_MATCH[0]="4\.19\.*"
PATCH_MATCH[1]="5\.4\.*"
PATCH_MATCH[2]="5\.8\.*"
```

Issue the follwing and reboot:
```
sudo dkms build -k $(uname -r) -m "seeed-voicecard" -v "0.3"
dkms install --force -k $(uname -r) -m "seeed-voicecard" -v "0.3"
```

HinTak, 2023-01-17:
Why do you need to delete those patch* lines? They are what they are, backport patches: if dkms detects you are running the older 4.19/5.4/5.8 kernels, they are applied. They are ignored with current kernels.

UmbrellaCodr, 2023-01-19:
I'm a newb to dkms; but the patch lines are improperly applying the patches to 5.15.84 kernel; so the kernel module fails to build because of the back-to-v5.8.diff reverts the mute_stream -> digital_mute.  The changes above allow the module to build as expected.   I am guessing the PATH_MATH lines are wrong but I would have to go readup on dkms to know what the contract is.

HinTak, 2023-01-19:
Argh , I see "5.15.84“ contains "5.8" in the middle! Yes, it is a small oversight.

ramfish, 2023-01-23:
Thanks guys... I am pretty new to all of this - can somebody edit the install script and commit a fix ?

HinTak, 2023-01-23:
Committed a small change which should fix the mismatch of 5.15.84 to 5.8 backport. Have a go and see if that works better.

ramfish, 2023-01-23:
Thanks HinTak. Can try to test it tonight. But embarrassed to say - which branch should I pull? Any instructions for a gray haired programmer who is more used to C then those modern Python/C+++++ or repositories  ;-)

HinTak, 2023-01-24:
Just read your `uname -r` - if it says 5.15.x, use the v5.15 branch on my repo. In general, I have a separate branch for each minor release (occasionally they are identical).

 The backport patch system is just a not very actively maintained mechanism for a later branch to support earlier kernels: basically if a later branch see that it is built against 5.8, it reverse-applies the difference. (The little bug was that `uname  -r` shows 5.15.84, and since it contains 5.8 in the middle, the difference was wrongly applied - the fix was to make sure we are matching from the beginning, instead of any part of it).

winne27, 2023-02-21:
The hint by UmbrellaCodr to delete some lines from dkms.conf worked for me. Thanks

HinTak, 2023-02-21:
The deletion shouldn't be needed anymore - it was just confused by 5.15.84 having a "5.8" fragment in the middle,  so it tried to change the code to be 5.8 compatible. I made some changes to avoid the confusion, so it should treat 5.15.84 as 5.15.x properly, rather than 5.8.x.

jloewe, 2023-03-14:
@HinTak thank you for the work in your fork! I tried to clone the v5.15 branch and noticed, that it doesn't exist anymore. Should I use another branch?

orcunkutlu, 2023-03-15:
> 

Hey,
My system looks yours(@UmbrellaCodr ) with cmd uname -r then try your solution after Hintaks install.sh cmd.
In var/lib/dkms/seeed-voicecard/0.3/source files copied its ok but all of them file size 0
i copied cloned folder to var/lib/dkms/seeed-voicecard/0.3/source
gives me module license error
then change module license("GPL V2") -> ("GPL")

Retry your command then all looking good but after reboot system halted. black screen then powered off

orcunkutlu, 2023-03-16:
Its impossible becuz its fresh raspi + driver only on 32gb sdcard. My problem is argonone case + respeaker i think. If i power on raspberry firstly then powerup respeaker with micro usb system works... but argone fan always on. when i try aplay or arecord cmd argone fan stop till cmd complete. I will try without argon case.