respeaker/seeed-voicecard #335: [Bug]: installation failed on bullseye with make error

bruesche, 2023-02-06:
### Describe the bug

Do the installation with sudo .install.sh. Refreshed the files from GitHub an get Error:

Reading state information... Done
E: Unable to locate package linux-raspi
E: Unable to locate package linux-headers-raspi
E: Unable to locate package linux-image-raspi

make -j4 KERNELRELEASE=5.15.84-v7l+ -C /lib/modules/5.15.84-v7l+/build M=/var/lib/dkms/seeed-voicecard/0.3/build....(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.15.84-v7l+ (armv7l)

The complet log (install log and Make.log) in the attachment.

The Rhasspy installation didn#t recognized the microphone...

Please help...
Thx

### Expected behavior

Installation without errors an a working microphone ;)

### Platform

- Device: Raspberry Pi 4
- OS: Bullseye
- Kernel 5.15.84-v7l+

### Relevant log output

```shell
Log from .install.sh:
pi@raspberrypi:~/seeed-voicecard $ sudo ./install.sh
Hit:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Hit:2 http://archive.raspberrypi.org/debian bullseye InRelease  
Hit:3 https://download.docker.com/linux/raspbian bullseye InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
raspberrypi-kernel is already the newest version (1:1.20230106-1).
raspberrypi-kernel-headers is already the newest version (1:1.20230106-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package linux-raspi
E: Unable to locate package linux-headers-raspi
E: Unable to locate package linux-image-raspi
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
dkms is already the newest version (2.8.4-3).
git is already the newest version (1:2.30.2-1+deb11u1).
i2c-tools is already the newest version (4.2-1+b1).
libasound2-plugins is already the newest version (1.2.2-2).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

------------------------------
Deleting module version: 0.3
completely from the DKMS tree.
------------------------------
Done.

Creating symlink /var/lib/dkms/seeed-voicecard/0.3/source ->
                 /usr/src/seeed-voicecard-0.3
[... 107 lines cut ...]
/var/lib/dkms/seeed-voicecard/0.3/build/ac108.c:1148:18: note: (near initialization for ‘ac108_dai_ops.bespoke_trigger’)
In file included from /var/lib/dkms/seeed-voicecard/0.3/build/wm8960.c:28:
/var/lib/dkms/seeed-voicecard/0.3/build/wm8960.c: In function ‘wm8960_configure_clocking’:
/var/lib/dkms/seeed-voicecard/0.3/build/sound-compatible-4.18.h:34:36: error: implicit declaration of function ‘snd_soc_component_read32’; did you mean ‘snd_soc_component_read’? [-Werror=implicit-function-declaration]
   34 | #define snd_soc_read               snd_soc_component_read32
      |                                    ^~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/seeed-voicecard/0.3/build/wm8960.c:619:15: note: in expansion of macro ‘snd_soc_read’
  619 |  u16 iface1 = snd_soc_read(codec, WM8960_IFACE1);
      |               ^~~~~~~~~~~~
/var/lib/dkms/seeed-voicecard/0.3/build/wm8960.c: At top level:
/var/lib/dkms/seeed-voicecard/0.3/build/wm8960.c:1239:3: error: ‘const struct snd_soc_dai_ops’ has no member named ‘digital_mute’
 1239 |  .digital_mute = wm8960_mute,
      |   ^~~~~~~~~~~~
/var/lib/dkms/seeed-voicecard/0.3/build/wm8960.c:1239:18: error: initialization of ‘int (*)(struct snd_pcm_substream *, struct snd_soc_dai *)’ from incompatible pointer type ‘int (*)(struct snd_soc_dai *, int)’ [-Werror=incompatible-pointer-types]
 1239 |  .digital_mute = wm8960_mute,
      |                  ^~~~~~~~~~~
/var/lib/dkms/seeed-voicecard/0.3/build/wm8960.c:1239:18: note: (near initialization for ‘wm8960_dai_ops.prepare’)
cc1: some warnings being treated as errors
make[1]: *** [scripts/Makefile.build:289: /var/lib/dkms/seeed-voicecard/0.3/build/ac108.o] Error 1
cc1: some warnings being treated as errors
make[1]: *** [scripts/Makefile.build:289: /var/lib/dkms/seeed-voicecard/0.3/build/wm8960.o] Error 1
make: *** [Makefile:1902: /var/lib/dkms/seeed-voicecard/0.3/build] Error 2
make: Leaving directory '/usr/src/linux-headers-5.15.84-v7l+'
(END)
```

HinTak, 2023-02-06:
See https://github.com/respeaker/seeed-voicecard/issues/332#issuecomment-1379439079

bruesche, 2023-02-06:
"After further playing around I noticed that the install script wasn't using the source from the git repo to build the module, but what was already present in /lib/modules/. So instead I just did a make, sudo make install and everything sprang back to life."

Didn't know what to do... can you give me advice?

HinTak, 2023-02-06:
It is as it says, make a fresh clone from my repo (HinTak/seed-voicecard instead of upstream), then type 'make' then "sudo make install" inside, and reboot.

bruesche, 2023-02-06:
tried. Sorry. I think I do something wrong. Here is what I do and what I get:

pi@raspberrypi:~ $ git clone https://github.com/respeaker/seeed-voicecard
Cloning into 'seeed-voicecard'...
remote: Enumerating objects: 1278, done.
remote: Counting objects: 100% (320/320), done.
remote: Compressing objects: 100% (20/20), done.
remote: Total 1278 (delta 308), reused 300 (delta 300), pack-reused 958
Receiving objects: 100% (1278/1278), 1.41 MiB | 1.23 MiB/s, done.
Resolving deltas: 100% (836/836), done.
pi@raspberrypi:~ $ cd seeed-voicecard/
pi@raspberrypi:~/seeed-voicecard $ make
make -C /lib/modules/5.15.84-v7l+/build M=/home/pi/seeed-voicecard modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.84-v7l+'
  CC [M]  /home/pi/seeed-voicecard/wm8960.o
  LD [M]  /home/pi/seeed-voicecard/snd-soc-wm8960.o
  CC [M]  /home/pi/seeed-voicecard/ac108.o
  CC [M]  /home/pi/seeed-voicecard/ac101.o
  LD [M]  /home/pi/seeed-voicecard/snd-soc-ac108.o
  CC [M]  /home/pi/seeed-voicecard/seeed-voicecard.o
/home/pi/seeed-voicecard/seeed-voicecard.c: In function ‘seeed_voice_card_dai_link_of’:
/home/pi/seeed-voicecard/seeed-voicecard.c:463:8: error: implicit declaration of function ‘asoc_simple_parse_cpu’; did you mean ‘asoc_simple_parse_clk’? [-Werror=implicit-function-declaration]
  463 |  ret = asoc_simple_parse_cpu(cpu, dai_link, &single_cpu);
      |        ^~~~~~~~~~~~~~~~~~~~~
      |        asoc_simple_parse_clk
/home/pi/seeed-voicecard/seeed-voicecard.c:468:8: error: implicit declaration of function ‘asoc_simple_parse_codec’; did you mean ‘asoc_simple_parse_dai’? [-Werror=implicit-function-declaration]
  468 |  ret = asoc_simple_parse_codec(codec, dai_link);
      |        ^~~~~~~~~~~~~~~~~~~~~~~
      |        asoc_simple_parse_dai
/home/pi/seeed-voicecard/seeed-voicecard.c:480:8: error: implicit declaration of function ‘asoc_simple_parse_platform’; did you mean ‘asoc_simple_parse_daifmt’? [-Werror=implicit-function-declaration]
  480 |  ret = asoc_simple_parse_platform(plat, dai_link);
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~~
      |        asoc_simple_parse_daifmt
/home/pi/seeed-voicecard/seeed-voicecard.c:505:8: error: implicit declaration of function ‘asoc_simple_parse_clk_cpu’; did you mean ‘asoc_simple_parse_clk’? [-Werror=implicit-function-declaration]
  505 |  ret = asoc_simple_parse_clk_cpu(dev, cpu, dai_link, cpu_dai);
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~
      |        asoc_simple_parse_clk
/home/pi/seeed-voicecard/seeed-voicecard.c:513:8: error: implicit declaration of function ‘asoc_simple_parse_clk_codec’; did you mean ‘asoc_simple_parse_clk’? [-Werror=implicit-function-declaration]
  513 |  ret = asoc_simple_parse_clk_codec(dev, codec, dai_link, codec_dai);
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |        asoc_simple_parse_clk
/home/pi/seeed-voicecard/seeed-voicecard.c:546:31: error: passing argument 1 of ‘asoc_simple_canonicalize_cpu’ from incompatible pointer type [-Werror=incompatible-pointer-types]
  546 |  asoc_simple_canonicalize_cpu(dai_link, single_cpu);
      |                               ^~~~~~~~
      |                               |
      |                               struct snd_soc_dai_link *
In file included from /home/pi/seeed-voicecard/seeed-voicecard.c:28:
./include/sound/simple_card_utils.h:158:70: note: expected ‘struct snd_soc_dai_link_component *’ but argument is of type ‘struct snd_soc_dai_link *’
  158 | void asoc_simple_canonicalize_cpu(struct snd_soc_dai_link_component *cpus,
      |                                   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~
/home/pi/seeed-voicecard/seeed-voicecard.c:548:36: error: passing argument 1 of ‘asoc_simple_canonicalize_platform’ from incompatible pointer type [-Werror=incompatible-pointer-types]
  548 |  asoc_simple_canonicalize_platform(dai_link);
      |                                    ^~~~~~~~
      |                                    |
      |                                    struct snd_soc_dai_link *
In file included from /home/pi/seeed-voicecard/seeed-voicecard.c:28:
./include/sound/simple_card_utils.h:156:75: note: expected ‘struct snd_soc_dai_link_component *’ but argument is of type ‘struct snd_soc_dai_link *’
  156 | void asoc_simple_canonicalize_platform(struct snd_soc_dai_link_component *platforms,
      |                                        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~
/home/pi/seeed-voicecard/seeed-voicecard.c:548:2: error: too few arguments to function ‘asoc_simple_canonicalize_platform’
  548 |  asoc_simple_canonicalize_platform(dai_link);
      |  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/pi/seeed-voicecard/seeed-voicecard.c:28:
./include/sound/simple_card_utils.h:156:6: note: declared here
  156 | void asoc_simple_canonicalize_platform(struct snd_soc_dai_link_component *platforms,
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
At top level:
/home/pi/seeed-voicecard/seeed-voicecard.c:263:12: warning: ‘asoc_simple_parse_dai’ defined but not used [-Wunused-function]
  263 | static int asoc_simple_parse_dai(struct device_node *node,
      |            ^~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:289: /home/pi/seeed-voicecard/seeed-voicecard.o] Error 1
make[1]: *** [Makefile:1902: /home/pi/seeed-voicecard] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.15.84-v7l+'
make: *** [Makefile:37: all] Error 2
pi@raspberrypi:~/seeed-voicecard $ sudo make install
sudo cp snd-soc-ac108.ko /lib/modules/5.15.84-v7l+/kernel/sound/soc/codecs/
cp: cannot stat 'snd-soc-ac108.ko': No such file or directory
make: *** [Makefile:43: install] Error 1
pi@raspberrypi:~/seeed-voicecard $

HinTak, 2023-02-06:
I said clone my fork, which is https://github.com/HinTak/seeed-voicecard

bruesche, 2023-02-07:
Sorry, old man, glasses... you know? ;)

Everything works fine now. 
Thanks for your help!

pauloesteban, 2023-03-09:
> I said clone my fork, which is https://github.com/HinTak/seeed-voicecard

```
$ git clone https://github.com/HinTak/seeed-voicecard
Cloning into 'seeed-voicecard'...
remote: Enumerating objects: 1363, done.
remote: Counting objects: 100% (387/387), done.
remote: Compressing objects: 100% (32/32), done.
remote: Total 1363 (delta 365), reused 366 (delta 355), pack-reused 976
Receiving objects: 100% (1363/1363), 1.16 MiB | 2.69 MiB/s, done.
Resolving deltas: 100% (897/897), done.

$ cd seeed-voicecard/

$ sudo make install
sudo cp snd-soc-ac108.ko /lib/modules/5.15.84-v7+/kernel/sound/soc/codecs/
cp: cannot stat 'snd-soc-ac108.ko': No such file or directory
make: *** [Makefile:43: install] Error 1

$ git status
On branch v5.9
Your branch is up to date with 'origin/v5.9'.

nothing to commit, working tree clean
```

The makefile error happened with HinTak repo

HinTak, 2023-03-09:
Doing "sudo make" between clone and "sudo make install" would be useful 😀

pauloesteban, 2023-03-09:
> Doing "sudo make" between clone and "sudo make install" would be useful 😀

Thanks for the suggestion, I am going to try a fresh installation.

FTR, I tried twice with `sudo ./install.sh` with the latest Raspberry OS image. It crashes the `lxpanel`, hence, I tried the method indicated above.

HinTak, 2023-03-09:
The lxpanel issue is generic to pulsesaudio : https://forums.raspberrypi.com/viewtopic.php?t=295641 and unrelated.

bruesche, 2023-04-11:
Hi, now have same problem. After kernel update the mic is not recognized by the pi. Did the same as last time.
Delete the directory speed-voicecard, clone new from github HinTak fork and start with make. 

Error:
pi@raspberrypi:~ $ rm -rf seeed-voicecard/
pi@raspberrypi:~ $ git clone https://github.com/HinTak/seeed-voicecard
Cloning into 'seeed-voicecard'...
remote: Enumerating objects: 1366, done.
remote: Counting objects: 100% (390/390), done.
remote: Compressing objects: 100% (34/34), done.
remote: Total 1366 (delta 366), reused 368 (delta 356), pack-reused 976
Receiving objects: 100% (1366/1366), 1.16 MiB | 3.98 MiB/s, done.
Resolving deltas: 100% (898/898), done.
pi@raspberrypi:~ $ cd seeed-voicecard/
pi@raspberrypi:~/seeed-voicecard $ make
make -C /lib/modules/6.1.21-v8+/build M=/home/pi/seeed-voicecard modules
make[1]: *** /lib/modules/6.1.21-v8+/build: No such file or directory.  Stop.
make: *** [Makefile:37: all] Error 2

sudo make gets same error. 

Tried to use install.sh

Error:
pi@raspberrypi:~/seeed-voicecard $ sudo ./install.sh
Get:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease [15.0 kB]
Get:2 http://archive.raspberrypi.org/debian bullseye InRelease [23.6 kB]                                                           
Hit:3 https://download.docker.com/linux/raspbian bullseye InRelease                                                                             
Get:4 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf Packages [13.2 MB]
Get:5 http://archive.raspberrypi.org/debian bullseye/main armhf Packages [316 kB]
Fetched 13.6 MB in 6s (2,103 kB/s)                                                                                                                                       
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
5 packages can be upgraded. Run 'apt list --upgradable' to see them.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
raspberrypi-kernel is already the newest version (1:1.20230405-1).
raspberrypi-kernel-headers is already the newest version (1:1.20230405-1).
0 upgraded, 0 newly installed, 0 to remove and 5 not upgraded.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package linux-raspi
E: Unable to locate package linux-headers-raspi
E: Unable to locate package linux-image-raspi
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
dkms is already the newest version (2.8.4-3).
git is already the newest version (1:2.30.2-1+deb11u2).
i2c-tools is already the newest version (4.2-1+b1).
libasound2-plugins is already the newest version (1.2.2-2).
0 upgraded, 0 newly installed, 0 to remove and 5 not upgraded.

------------------------------
Deleting module version: 0.3
completely from the DKMS tree.
------------------------------
Done.

Creating symlink /var/lib/dkms/seeed-voicecard/0.3/source ->
                 /usr/src/seeed-voicecard-0.3

DKMS: add completed.
Error! echo
Your kernel headers for kernel 6.1.21-v8+ cannot be found at
/lib/modules/6.1.21-v8+/build or /lib/modules/6.1.21-v8+/source.
You can use the --kernelsourcedir option to tell DKMS where it's located.
mkdir: cannot create directory ‘/etc/voicecard’: File exists
git init
Reinitialized existing Git repository in /etc/voicecard/.git/
git add --all
git commit -m "origin configures"
On branch master
nothing to commit, working tree clean
------------------------------------------------------
Please reboot your raspberry pi to apply all settings
Enjoy!

There is something wrong with the kernel version. Can you help me?

HinTak, 2023-04-11:
What does "ls -l /lib/modules/6.1.21-v8+/build" says, and does it points at a real place? It says no such file/ directory. You might want to try again?

bruesche, 2023-04-17:
Hier can I solve my Problem? Can anybody help me?

HinTak, 2023-04-17:
@bruesche it is not the same problem... but there is a workaround in https://github.com/raspberrypi/firmware/issues/1795 , adding one line in /boot/configuration.txt

Jeebus678, 2023-04-20:
@bruesche would like to update here that I had the exact same issue. Adding `arm_64bit=0` to `/boot/config.txt` and then reinstalling the driver fixed the issue.

HinTak, 2023-04-20:
FWIW, the initial report and current (since early April) are two different problems. The initial report is simply because upstream seeed studio has stopped updating beyond about 5.13, so won't build against newer kernels. The solution to that is to use one of my v5.x/v6.x branches. (It gets quite messy with a lot of "if kernel version <..." ">=" , so I make new branches when difference is too large for such constructs). 

The new problem since early April is that raspberrypi people decided that 64-bit kernel on 32-bit system is better than 32-bit kernel on 32-bit system. (Better compiler optimization and resource useage etc). But that specifically does not work for out-of-tree device  drivers. They are still working on it - https://github.com/raspberrypi/firmware/issues/1795 - but until they figure it out, you can add one line in/boot/config.txt to go back to a 32-bit kernel on 32-bit system setup.

bruesche, 2023-04-25:
Thanks a lot!

closed 2023-04-25 by bruesche