respeaker/seeed-voicecard #337: [Bug]: `install.sh` script throws multiple error. Sound card not available afterwards.

ngreve, 2023-04-10:
### Describe the bug

When running `install.sh` the script throws errors like:
```
E: Unable to locate package linux-raspi
E: Unable to locate package linux-headers-raspi
E: Unable to locate package linux-image-raspi
. . .
grep: /boot/kernel.img: No such file or directory

gzip: stdin: unexpected end of file
grep: /boot/kernel.img: No such file or directory

gzip: stdin: unexpected end of file
. . .

make -j4 KERNELRELEASE=6.1.21-v8+ -C /lib/modules/6.1.21-v8+/build M=/var/lib/dkms/seeed-voicecard/0.3/build....(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.1.21-v8+ (aarch64)
```

After rebooting, the soundcard is also not listed when running `arecord -L`

**To Reproduce**
```
sudo ./install.sh
```
```
arecord -L
```

### Expected behavior

No errors while installing and the sound card should be available after reboot.

### Platform

- Device: Raspberry Pi 4 (8GB)
- OS: Raspbian OS 64bit (Desktop) kernel version 6.1.21-v8+
- Version/commit number 6bf1dbd04e55d412abbba3832eccbcecb3f09810

### Relevant log output

```shell
sudo ./install.sh
Hit:1 http://deb.debian.org/debian bullseye InRelease
Hit:2 http://deb.debian.org/debian bullseye-updates InRelease                                                           
Hit:3 http://security.debian.org/debian-security bullseye-security InRelease                                            
Hit:4 http://archive.raspberrypi.org/debian bullseye InRelease                                                          
Hit:5 https://download.docker.com/linux/debian bullseye InRelease                                 
Reading package lists... Done                                               
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
raspberrypi-kernel is already the newest version (1:1.20230405-1).
raspberrypi-kernel-headers is already the newest version (1:1.20230405-1).
The following package was automatically installed and is no longer required:
  libfuse2
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package linux-raspi
E: Unable to locate package linux-headers-raspi
E: Unable to locate package linux-image-raspi
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libasound2-plugins is already the newest version (1.2.2-2).
dkms is already the newest version (2.8.4-3).
i2c-tools is already the newest version (4.2-1+b1).
git is already the newest version (1:2.30.2-1+deb11u2).
The following package was automatically installed and is no longer required:
  libfuse2
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
grep: /boot/kernel.img: No such file or directory

gzip: stdin: unexpected end of file
grep: /boot/kernel.img: No such file or directory

gzip: stdin: unexpected end of file

------------------------------
Deleting module version: 0.3
completely from the DKMS tree.
------------------------------
Done.

Creating symlink /var/lib/dkms/seeed-voicecard/0.3/source ->
                 /usr/src/seeed-voicecard-0.3

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=6.1.21-v8+ -C /lib/modules/6.1.21-v8+/build M=/var/lib/dkms/seeed-voicecard/0.3/build....(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.1.21-v8+ (aarch64)
Consult /var/lib/dkms/seeed-voicecard/0.3/build/make.log for more information.
mkdir: cannot create directory ‘/etc/voicecard’: File exists
git init
Reinitialized existing Git repository in /etc/voicecard/.git/
git add --all
git commit -m "origin configures"
On branch master
nothing to commit, working tree clean
------------------------------------------------------
Please reboot your raspberry pi to apply all settings
Enjoy!
------------------------------------------------------
```

```
arecord -L
null
    Discard all samples (playback) or generate zero samples (capture)
lavrate
    Rate Converter Plugin Using Libav/FFmpeg Library
samplerate
    Rate Converter Plugin Using Samplerate Library
speexrate
    Rate Converter Plugin Using Speex Resampler
jack
    JACK Audio Connection Kit
oss
    Open Sound System
pulse
    PulseAudio Sound Server
upmix
    Plugin for channel upmix (4,6,8)
vdownmix
    Plugin for channel downmix (stereo) with a simple spacialization
default
    Playback/recording through the PulseAudio sound server
playback
ac108
usbstream:CARD=Headphones
    bcm2835 Headphones
    USB Stream Output
usbstream:CARD=vc4hdmi0
    vc4-hdmi-0
    USB Stream Output
usbstream:CARD=vc4hdmi1
    vc4-hdmi-1
    USB Stream Output
```

Contents of `/var/lib/dkms/seeed-voicecard/0.3/build/make.log`:
```
DKMS make.log for seeed-voicecard-0.3 for kernel 6.1.21-v8+ (aarch64)
Mon 10 Apr 10:28:05 BST 2023
make: Entering directory '/usr/src/linux-headers-6.1.21-v8+'
  CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/wm8960.o
  CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/ac108.o
  CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/ac101.o
  CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.o
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c: In function ‘asoc_simple_init_dai_link_params’:
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c:367:27: error: ‘const struct snd_soc_component_driver’ has no member named ‘non_legacy_dai_naming’; did you mean ‘legacy_dai_naming’?
  367 |   if (!component->driver->non_legacy_dai_naming)
      |                           ^~~~~~~~~~~~~~~~~~~~~
      |                           legacy_dai_naming
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c: In function ‘seeed_voice_card_remove’:
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c:889:9: error: void value not ignored as it ought to be
  889 |  return asoc_simple_clean_reference(card);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c:890:1: error: control reaches end of non-void function [-Werror=return-type]
  890 | }
      | ^
cc1: some warnings being treated as errors
make[1]: *** [scripts/Makefile.build:250: /var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.o] Error 1
make[1]: *** Waiting for unfinished jobs....
/var/lib/dkms/seeed-voicecard/0.3/build/ac108.c:1332:3: error: ‘struct snd_soc_component_driver’ has no member named ‘non_legacy_dai_naming’; did you mean ‘legacy_dai_naming’?
 1332 |  .non_legacy_dai_naming  = 1,
      |   ^~~~~~~~~~~~~~~~~~~~~
      |   legacy_dai_naming
/var/lib/dkms/seeed-voicecard/0.3/build/wm8960.c:1294:3: error: ‘const struct snd_soc_component_driver’ has no member named ‘non_legacy_dai_naming’; did you mean ‘legacy_dai_naming’?
 1294 |  .non_legacy_dai_naming = 1,
      |   ^~~~~~~~~~~~~~~~~~~~~
      |   legacy_dai_naming
/var/lib/dkms/seeed-voicecard/0.3/build/wm8960.c:1412:14: error: initialization of ‘void (*)(struct i2c_client *)’ from incompatible pointer type ‘int (*)(struct i2c_client *)’ [-Werror=incompatible-pointer-types]
 1412 |  .remove =   wm8960_i2c_remove,
      |              ^~~~~~~~~~~~~~~~~
/var/lib/dkms/seeed-voicecard/0.3/build/wm8960.c:1412:14: note: (near initialization for ‘wm8960_i2c_driver.remove’)
/var/lib/dkms/seeed-voicecard/0.3/build/ac108.c:1551:14: error: initialization of ‘void (*)(struct i2c_client *)’ from incompatible pointer type ‘int (*)(struct i2c_client *)’ [-Werror=incompatible-pointer-types]
 1551 |  .remove =   ac108_i2c_remove,
      |              ^~~~~~~~~~~~~~~~
/var/lib/dkms/seeed-voicecard/0.3/build/ac108.c:1551:14: note: (near initialization for ‘ac108_i2c_driver.remove’)
cc1: some warnings being treated as errors
cc1: some warnings being treated as errors
make[1]: *** [scripts/Makefile.build:250: /var/lib/dkms/seeed-voicecard/0.3/build/wm8960.o] Error 1
make[1]: *** [scripts/Makefile.build:250: /var/lib/dkms/seeed-voicecard/0.3/build/ac108.o] Error 1
make: *** [Makefile:2012: /var/lib/dkms/seeed-voicecard/0.3/build] Error 2
make: Leaving directory '/usr/src/linux-headers-6.1.21-v8+'

```
```

HinTak, 2023-04-10:
See https://github.com/respeaker/seeed-voicecard/pull/323#issuecomment-1483811134 , and  possibly also the latest #336

lxne, 2023-04-15:
I have a Respeaker 4-mic linear array and can't get it to work in bullseye 64bit.

With the original repo I get the loop.
`Please reboot your machine AND THEN run this script again `

With HinTak's repo after the installation my menübar vanishes.
This is what I get in the terminal:

> pi@raspberrypi:~ $ git clone https://github.com/HinTak/seeed-voicecard.git
> Klone nach 'seeed-voicecard' ...
> remote: Enumerating objects: 1369, done.
> remote: Counting objects: 100% (393/393), done.
> remote: Compressing objects: 100% (37/37), done.
> remote: Total 1369 (delta 368), reused 370 (delta 356), pack-reused 976
> Empfange Objekte: 100% (1369/1369), 1.17 MiB | 1.68 MiB/s, fertig.
> Löse Unterschiede auf: 100% (900/900), fertig.
> 
> pi@raspberrypi:~ $ cd seeed-voicecard
> pi@raspberrypi:~/seeed-voicecard $ sudo ./install.sh
> OK:1 http://deb.debian.org/debian bullseye InRelease
> OK:2 http://deb.debian.org/debian bullseye-updates InRelease                   
> OK:3 http://security.debian.org/debian-security bullseye-security InRelease    
> OK:4 http://archive.raspberrypi.org/debian bullseye InRelease                  
> Paketlisten werden gelesen... Fertig
> Abhängigkeitsbaum wird aufgebaut... Fertig
> Statusinformationen werden eingelesen... Fertig
> Alle Pakete sind aktuell.
> Paketlisten werden gelesen... Fertig
> Abhängigkeitsbaum wird aufgebaut... Fertig
> Statusinformationen werden eingelesen... Fertig
> raspberrypi-kernel ist schon die neueste Version (1:1.20230405-1).
> raspberrypi-kernel-headers ist schon die neueste Version (1:1.20230405-1).
> Das folgende Paket wurde automatisch installiert und wird nicht mehr benötigt:
>   libfuse2
> Verwenden Sie »sudo apt autoremove«, um es zu entfernen.
> 0 aktualisiert, 0 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.
> Paketlisten werden gelesen... Fertig
> Abhängigkeitsbaum wird aufgebaut... Fertig
> Statusinformationen werden eingelesen... Fertig
> Hinweis: »gcc« wird an Stelle von »gcc-aarch64-linux-gnu« gewählt.
> gcc ist schon die neueste Version (4:10.2.1-1).
> Das folgende Paket wurde automatisch installiert und wird nicht mehr benötigt:
>   libfuse2
> Verwenden Sie »sudo apt autoremove«, um es zu entfernen.
> 0 aktualisiert, 0 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.
> Paketlisten werden gelesen... Fertig
> Abhängigkeitsbaum wird aufgebaut... Fertig
> Statusinformationen werden eingelesen... Fertig
> E: Paket linux-raspi kann nicht gefunden werden.
> E: Paket linux-headers-raspi kann nicht gefunden werden.
> E: Paket linux-image-raspi kann nicht gefunden werden.
> Paketlisten werden gelesen... Fertig
> Abhängigkeitsbaum wird aufgebaut... Fertig
> Statusinformationen werden eingelesen... Fertig
> libasound2-plugins ist schon die neueste Version (1.2.2-2).
> dkms ist schon die neueste Version (2.8.4-3).
> i2c-tools ist schon die neueste Version (4.2-1+b1).
> git ist schon die neueste Version (1:2.30.2-1+deb11u2).
> Das folgende Paket wurde automatisch installiert und wird nicht mehr benötigt:
>   libfuse2
> Verwenden Sie »sudo apt autoremove«, um es zu entfernen.
> 0 aktualisiert, 0 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.
> grep: /boot/kernel.img: No such file or directory
> 
> gzip: stdin: unexpected end of file
> grep: /boot/kernel.img: No such file or directory
> 
> gzip: stdin: unexpected end of file
> Error! The module/version combo: seeed-voicecard-0.3
> is not located in the DKMS tree.
> 
> Creating symlink /var/lib/dkms/seeed-voicecard/0.3/source ->
>                  /usr/src/seeed-voicecard-0.3
> 
> DKMS: add completed.
> 
> Kernel preparation unnecessary for this kernel.  Skipping...
> 
> Building module:
> cleaning build area...
> make -j4 KERNELRELEASE=6.1.21-v8+ -C /lib/modules/6.1.21-v8+/build M=/var/lib/dkms/seeed-voicecard/0.3/build.....
> cleaning build area...
> 
> DKMS: build completed.
> 
> snd-soc-wm8960.ko.xz:
> Running module version sanity check.
>  - Original module
>  - Installation
>    - Installing to /lib/modules/6.1.21-v8+/updates/dkms/
> 
> snd-soc-ac108.ko.xz:
> Running module version sanity check.
>  - Original module
>  - Installation
>    - Installing to /lib/modules/6.1.21-v8+/updates/dkms/
> 
> snd-soc-seeed-voicecard.ko.xz:
> Running module version sanity check.
>  - Original module
>  - Installation
>    - Installing to /lib/modules/6.1.21-v8+/updates/dkms/
> 
> depmod.....
> 
> DKMS: install completed.
> git init
> Hinweis: Als Name für den initialen Branch wurde 'master' benutzt. Dieser
> Hinweis: Standard-Branchname kann sich ändern. Um den Namen des initialen Branches
> Hinweis: zu konfigurieren, der in allen neuen Repositories verwendet werden soll und
> Hinweis: um diese Warnung zu unterdrücken, führen Sie aus:
> Hinweis: 
> Hinweis: 	git config --global init.defaultBranch <Name>
> Hinweis: 
> Hinweis: Häufig gewählte Namen statt 'master' sind 'main', 'trunk' und
> Hinweis: 'development'. Der gerade erstellte Branch kann mit diesem Befehl
> Hinweis: umbenannt werden:
> Hinweis: 
> Hinweis: 	git branch -m <Name>
> Leeres Git-Repository in /etc/voicecard/.git/ initialisiert
> git add --all
> git commit -m "origin configures"
> [master (Root-Commit) f422a29] origin configures
>  7 files changed, 1482 insertions(+)
>  create mode 100644 ac108_6mic.state
>  create mode 100644 ac108_asound.state
>  create mode 100644 asound_2mic.conf
>  create mode 100644 asound_4mic.conf
>  create mode 100644 asound_6mic.conf
>  create mode 100644 dkms.conf
>  create mode 100644 wm8960_asound.state
> Created symlink /etc/systemd/system/sysinit.target.wants/seeed-voicecard.service -> /lib/systemd/system/seeed-voicecard.service.
> ------------------------------------------------------
> Please reboot your raspberry pi to apply all settings
> Enjoy!
> ------------------------------------------------------
>

Any ideas would be appreciated.

HinTak, 2023-04-15:
@lxne I like to see the result of just "make ; sudo make install" ? That may be more useful at the moment.

lxne, 2023-04-16:
Hi @HinTak, for not messing things up, I started from scratch.
I instantly get an error when trying the make command (see at bottom).

**Hardware**
Raspberry Pi 400 Rev 1.1
Seeed Studio ReSpeaker 4-Mic Linear Array

**Raspberry Pi OS**
Preparing the SD card with Raspberry Pi Imager v1.7.4 Bullseye 64bit
First boot -> Updates -> reboot
_cat /etc/debian_version_
`11.6`
_uname -a_
`Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`

_sudo apt-get update_
```
Hit:1 http://security.debian.org/debian-security bullseye-security InRelease
Hit:2 http://deb.debian.org/debian bullseye InRelease            
Hit:3 http://archive.raspberrypi.org/debian bullseye InRelease   
Hit:4 http://deb.debian.org/debian bullseye-updates InRelease
Reading package lists... Done
```
_git clone https://github.com/HinTak/seeed-voicecard.git_
```
Cloning into 'seeed-voicecard'...
remote: Enumerating objects: 1369, done.
remote: Counting objects: 100% (393/393), done.
remote: Compressing objects: 100% (36/36), done.
remote: Total 1369 (delta 368), reused 371 (delta 357), pack-reused 976
Receiving objects: 100% (1369/1369), 1.17 MiB | 3.19 MiB/s, done.
Resolving deltas: 100% (900/900), done.
```
_cd seeed-voicecard/
make ; sudo make install_
```
make -C /lib/modules/6.1.21-v8+/build M=/home/pi/seeed-voicecard modules
make[1]: *** /lib/modules/6.1.21-v8+/build: No such file or directory.  Stop.
make: *** [Makefile:37: all] Error 2
sudo cp snd-soc-ac108.ko /lib/modules/6.1.21-v8+/kernel/sound/soc/codecs/
cp: cannot stat 'snd-soc-ac108.ko': No such file or directory
make: *** [Makefile:43: install] Error 1
```

lxne, 2023-04-16:
Then I tried install.sh and did a restart -> menu bar gone -> result of arecord -L at the bottom

_sudo ./install.sh_
```
Hit:1 http://deb.debian.org/debian bullseye InRelease
Hit:2 http://deb.debian.org/debian bullseye-updates InRelease
Hit:3 http://security.debian.org/debian-security bullseye-security InRelease
Hit:4 http://archive.raspberrypi.org/debian bullseye InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
raspberrypi-kernel is already the newest version (1:1.20230405-1).
The following package was automatically installed and is no longer required:
  libfuse2
Use 'sudo apt autoremove' to remove it.
The following NEW packages will be installed:
  raspberrypi-kernel-headers
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 10.2 MB of archives.
After this operation, 64.8 MB of additional disk space will be used.
Get:1 http://archive.raspberrypi.org/debian bullseye/main arm64 raspberrypi-kernel-headers arm64 1:1.20230405-1 [10.2 MB]
Fetched 10.2 MB in 4s (2,752 kB/s)                     
Selecting previously unselected package raspberrypi-kernel-headers.
(Reading database ... 96867 files and directories currently installed.)
Preparing to unpack .../raspberrypi-kernel-headers_1%3a1.20230405-1_arm64.deb ...
Unpacking raspberrypi-kernel-headers (1:1.20230405-1) ...
Setting up raspberrypi-kernel-headers (1:1.20230405-1) ...
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'gcc' instead of 'gcc-aarch64-linux-gnu'
gcc is already the newest version (4:10.2.1-1).
gcc set to manually installed.
The following package was automatically installed and is no longer required:
  libfuse2
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package linux-raspi
E: Unable to locate package linux-headers-raspi
E: Unable to locate package linux-image-raspi
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libasound2-plugins is already the newest version (1.2.2-2).
libasound2-plugins set to manually installed.
i2c-tools is already the newest version (4.2-1+b1).
i2c-tools set to manually installed.
git is already the newest version (1:2.30.2-1+deb11u2).
The following package was automatically installed and is no longer required:
  libfuse2
Use 'sudo apt autoremove' to remove it.
The following additional packages will be installed:
  dctrl-tools
Suggested packages:
  debtags menu
The following NEW packages will be installed:
  dctrl-tools dkms
0 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
Need to get 179 kB of archives.
After this operation, 720 kB of additional disk space will be used.
Get:1 http://deb.debian.org/debian bullseye/main arm64 dctrl-tools arm64 2.24-3 [101 kB]
Get:2 http://deb.debian.org/debian bullseye/main arm64 dkms all 2.8.4-3 [78.2 kB]
Fetched 179 kB in 1s (295 kB/s)
[... 109 lines cut ...]
ac108
dmixer
ac101
usbstream:CARD=vc4hdmi0
    vc4-hdmi-0
    USB Stream Output
usbstream:CARD=vc4hdmi1
    vc4-hdmi-1
    USB Stream Output
hw:CARD=seeed8micvoicec,DEV=0
    seeed-8mic-voicecard, bcm2835-i2s-ac10x-codec0 ac10x-codec0-0
    Direct hardware device without any conversions
plughw:CARD=seeed8micvoicec,DEV=0
    seeed-8mic-voicecard, bcm2835-i2s-ac10x-codec0 ac10x-codec0-0
    Hardware device with all software conversions
sysdefault:CARD=seeed8micvoicec
    seeed-8mic-voicecard, bcm2835-i2s-ac10x-codec0 ac10x-codec0-0
    Default Audio Device
dsnoop:CARD=seeed8micvoicec,DEV=0
    seeed-8mic-voicecard, bcm2835-i2s-ac10x-codec0 ac10x-codec0-0
    Direct sample snooping device
usbstream:CARD=seeed8micvoicec
    seeed-8mic-voicecard
    USB Stream Output
```

lxne, 2023-04-16:
Then I tried the make command again – this time no errors – and did a reboot -> everything stays the same (no menu bar). 

_make ; sudo make install_
```
make -C /lib/modules/6.1.21-v8+/build M=/home/pi/seeed-voicecard modules
make[1]: Entering directory '/usr/src/linux-headers-6.1.21-v8+'
  CC [M]  /home/pi/seeed-voicecard/wm8960.o
  LD [M]  /home/pi/seeed-voicecard/snd-soc-wm8960.o
  CC [M]  /home/pi/seeed-voicecard/ac108.o
  CC [M]  /home/pi/seeed-voicecard/ac101.o
  LD [M]  /home/pi/seeed-voicecard/snd-soc-ac108.o
  CC [M]  /home/pi/seeed-voicecard/seeed-voicecard.o
  LD [M]  /home/pi/seeed-voicecard/snd-soc-seeed-voicecard.o
  MODPOST /home/pi/seeed-voicecard/Module.symvers
  CC [M]  /home/pi/seeed-voicecard/snd-soc-ac108.mod.o
  LD [M]  /home/pi/seeed-voicecard/snd-soc-ac108.ko
  CC [M]  /home/pi/seeed-voicecard/snd-soc-seeed-voicecard.mod.o
  LD [M]  /home/pi/seeed-voicecard/snd-soc-seeed-voicecard.ko
  CC [M]  /home/pi/seeed-voicecard/snd-soc-wm8960.mod.o
  LD [M]  /home/pi/seeed-voicecard/snd-soc-wm8960.ko
make[1]: Leaving directory '/usr/src/linux-headers-6.1.21-v8+'
sudo cp snd-soc-ac108.ko /lib/modules/6.1.21-v8+/kernel/sound/soc/codecs/
sudo cp snd-soc-wm8960.ko /lib/modules/6.1.21-v8+/kernel/sound/soc/codecs/
sudo cp snd-soc-seeed-voicecard.ko /lib/modules/6.1.21-v8+/kernel/sound/soc/bcm/
sudo depmod -a
```

HinTak, 2023-04-17:
@lxne I posted elsewhere, The lxpanel issue (disappearing menu) is generic to pulsesaudio : https://forums.raspberrypi.com/viewtopic.php?t=295641 and somewhat unrelated; there are some tips there about restoring it.

lxne, 2023-04-17:
> @lxne I posted elsewhere, The lxpanel issue (disappearing menu) is generic to pulsesaudio : https://forums.raspberrypi.com/viewtopic.php?t=295641 and somewhat unrelated; there are some tips there about restoring it.

@HinTak Thank you for your reply. I looked into the link but I do not get the errors when I run pulseaudio manually and in /usr/share/alsa/alsa.conf there are 0 where they should be.

It is strange because I didn't do anything else with the fresh system then the tings documented above. Right now I am blank.

closed 2023-04-17 by ngreve

renatodex, 2024-11-11:
Just tested yesterday and the original respeaker repo didn't worked for me (https://github.com/respeaker/seeed-voicecard/issues/337)

But then I've tried this one -> https://github.com/HinTak/seeed-voicecard
And the ./install.sh worked !

I'm running aplay and I can see my devices now listed:

````
aplay --list-devices
**** List of PLAYBACK Hardware Devices ****
card 0: seeed2micvoicec [seeed-2mic-voicecard], device 0: 1f000a0000.i2s-wm8960-hifi wm8960-hifi-0 [1f000a0000.i2s-wm8960-hifi wm8960-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```