respeaker/seeed-voicecard #339: [Bug]: Respeaker 2-Mic Pi HAT不适用新的内核

3853824, 2023-09-11:
### Describe the bug

我的树莓派4B在运行sudo ./install_arm64.sh安装驱动的时候总会提示
WARNING Your loaded kernel version is 6.1.21 
Not matching the updated version

6.1.21
Kernel was updated, but	new kernel was not loaded yet	
Please reboot your machine AND THEN run this script again

这表明内核不匹配，而我的系统内核版本是6.1.21-v8+

### Expected behavior

有什么可以解决的办法，我不希望去重装早期的系统来去适配，希望官方能更新这个bug

### Platform

型号：Raspberry Pi 4B
系统：[Raspbian OS 64bit ]   发布时间2023.5.03

HinTak, 2023-09-11:
Use the v6.1 branch of https://github.com/HinTak/seeed-voicecard

mingxxming, 2024-11-08:
Building module:
cleaning build area...
make -j4 KERNELRELEASE=6.1.21-v8+ -C /lib/modules/6.1.21-v8+/build M=/var/lib/dkms/seeed-voicecard/0.3/build....(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.1.21-v8+ (aarch64)
Consult /var/lib/dkms/seeed-voicecard/0.3/build/make.log for more information.
.
run-parts: executing /etc/kernel/postinst.d/initramfs-tools 6.1.21-v8+ /boot/kernel8.img

------------------------------
Deleting module version: 0.3
completely from the DKMS tree.
------------------------------
Done.

Creating symlink /var/lib/dkms/seeed-voicecard/0.3/source ->
                 /usr/src/seeed-voicecard-0.3

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=6.1.21-v8+ -C /lib/modules/6.1.21-v8+/build M=/var/lib/dkms/seeed-voicecard/0.3/build....(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.1.21-v8+ (aarch64)
Consult /var/lib/dkms/seeed-voicecard/0.3/build/make.log for more information.
mkdir: cannot create directory ‘/etc/voicecard’: File exists
git init
Reinitialized existing Git repository in /etc/voicecard/.git/
git add --all
git commit -m "origin configures"
On branch master
nothing to commit, working tree clean
------------------------------------------------------
Please reboot your raspberry pi to apply all settings
Enjoy!
------------------------------------------------------
root@raspberrypi:/data/seeed-voicecard# ^C
root@raspberrypi:/data/seeed-voicecard# ^C
root@raspberrypi:/data/seeed-voicecard# cart ^C
root@raspberrypi:/data/seeed-voicecard# cat  /var/lib/dkms/seeed-voicecard/0.3/build/make.log
DKMS make.log for seeed-voicecard-0.3 for kernel 6.1.21-v8+ (aarch64)
Fri  8 Nov 18:48:00 GMT 2024
make: Entering directory '/root/linux-0afb5e98488aed7017b9bf321b575d0177feb7ed'
  CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/wm8960.o
  CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/ac108.o
  CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/ac101.o
  CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.o
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c:57:26: error: field ‘cpu_dai’ has incomplete type
   57 |   struct simple_util_dai cpu_dai;
      |                          ^~~~~~~
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c:58:26: error: field ‘codec_dai’ has incomplete type
   58 |   struct simple_util_dai codec_dai;
      |                          ^~~~~~~~~
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c:85:25: error: field ‘cpu_dai’ has incomplete type
   85 |  struct simple_util_dai cpu_dai;
      |                         ^~~~~~~
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c:86:25: error: field ‘codec_dai’ has incomplete type
   86 |  struct simple_util_dai codec_dai;
      |                         ^~~~~~~~~
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c: In function ‘seeed_voice_card_startup’:
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c:114:6: error: implicit declaration of function ‘snd_soc_rtd_to_cpu’; did you mean ‘asoc_rtd_to_cpu’? [-Werror=implicit-function-declaration]
  114 |  if (snd_soc_rtd_to_cpu(rtd, 0)->driver->playback.channels_min) {
      |      ^~~~~~~~~~~~~~~~~~
      |      asoc_rtd_to_cpu
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c:114:32: error: invalid type argument of ‘->’ (have ‘int’)
  114 |  if (snd_soc_rtd_to_cpu(rtd, 0)->driver->playback.channels_min) {
      |                                ^~
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c:115:63: error: invalid type argument of ‘->’ (have ‘int’)
  115 |   priv->channels_playback_default = snd_soc_rtd_to_cpu(rtd, 0)->driver->playback.channels_min;
[... 250 lines cut ...]
  574 |  simple_util_canonicalize_platform(dai_link->platforms, dai_link->cpus);
      |  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |  asoc_simple_canonicalize_platform
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c: In function ‘seeed_voice_card_parse_of’:
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c:673:8: error: implicit declaration of function ‘simple_util_parse_card_name’; did you mean ‘simple_util_parse_dai’? [-Werror=implicit-function-declaration]
  673 |  ret = simple_util_parse_card_name(&priv->snd_card, PREFIX);
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |        simple_util_parse_dai
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c: In function ‘seeed_voice_card_probe’:
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c:888:2: error: implicit declaration of function ‘simple_util_clean_reference’ [-Werror=implicit-function-declaration]
  888 |  simple_util_clean_reference(&priv->snd_card);
      |  ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c: At top level:
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c:918:12: error: initialization of ‘int (*)(struct platform_device *)’ from incompatible pointer type ‘void (*)(struct platform_device *)’ [-Werror=incompatible-pointer-types]
  918 |  .remove = seeed_voice_card_remove,
      |            ^~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c:918:12: note: (near initialization for ‘seeed_voice_card.remove’)
cc1: some warnings being treated as errors
make[1]: *** [scripts/Makefile.build:250: /var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.o] Error 1
cc1: some warnings being treated as errors
make[1]: *** [scripts/Makefile.build:250: /var/lib/dkms/seeed-voicecard/0.3/build/wm8960.o] Error 1
make: *** [Makefile:2012: /var/lib/dkms/seeed-voicecard/0.3/build] Error 2
make: Leaving directory '/root/linux-0afb5e98488aed7017b9bf321b575d0177feb7ed'

Still not working. 6.1.21-v8+

HinTak, 2024-11-09:
Told you already, wrong branch.
Use the v6.1 branch of https://github.com/HinTak/seeed-voicecard

mingxxming, 2024-11-09:
> Told you already, wrong branch. Use the v6.1 branch of https://github.com/HinTak/seeed-voicecard

It's v6.11 branch

root@raspberrypi:/data/seeed-voicecard# git branch
* v6.11
root@raspberrypi:/data/seeed-voicecard# git remote -v
origin  https://github.com/HinTak/seeed-voicecard.git (fetch)
origin  https://github.com/HinTak/seeed-voicecard.git (push)

mingxxming, 2024-11-09:
Downgrade to 2023-12-05-raspios-bookworm-arm64 resloved.

> Told you already, wrong branch. Use the v6.1 branch of https://github.com/HinTak/seeed-voicecard

I  am foo.  It should be "v.6.1"  but not "v6.1x"

dihan, 2025-01-15:
here is a little guide to help with the Pi 4 using HinTak's forked repo

https://github.com/dihan/pi-voice-assistant

tophiphi, 2025-03-24:
> here is a little guide to help with the Pi 4 using HinTak's forked repo
> 
> https://github.com/dihan/pi-voice-assistant

this doesn't refer to the problem at all.