respeaker/seeed-voicecard #340: [QUESTION]: Connecting Respeaker 6-Mic to Jetson Nano

Laura-VFA, 2023-09-21:
### Describe the bug

Hi there, I am trying to install the drivers to use a respeaker 6-mic array in a jetson nano, but it is not working. It is connected using the 40 pins.
I have tried to use these links:
https://wiki.seeedstudio.com/ReSpeaker_6-Mic_Circular_Array_kit_for_Raspberry_Pi/
https://wiki.seeedstudio.com/ReSpeaker_2_Mics_Pi_HAT_Jetson/

But it not seems to have support for the 6-mic array (although it has support for 2 and 4 mic respeakers). Is there any way to do this?

Thanks

### Expected behavior

Respeaker 6-mic working on Jetson Nano

### Platform

- Device: Jetson Nano
- OS: Ubuntu 18

HinTak, 2023-09-21:
I read that the 40-pin wiring of the jetson nano is slightly different from the pi? You may want to read up on it.

Disclaimer: just another customer, no relation to seeedstudio.

Laura-VFA, 2023-09-22:
> I read that the 40-pin wiring of the jetson nano is slightly different from the pi? You may want to read up on it.
> 
> Disclaimer: just another customer, no relation to seeedstudio.

But it does not make sense that Respeaker 2 Mics Arry is PINOUT compatible, but not with the 6 Mic array:
https://wiki.seeedstudio.com/ReSpeaker_2_Mics_Pi_HAT_Jetson/

[image]

In fact, if I use the command 
`sudo /opt/nvidia/jetson-io/config-by-hardware.py --list`
(that is mentioned in that seeedstudio post) the compatible devices are displayed, and the respeaker 4-mic array also appears.
How difficult would it be to add drivers for the 6 mic?

thanks for your quick response

HinTak, 2023-09-22:
It does make sense not all devices are compatible - not all pins are used in all cases. Most of the pins are the same, but not all.

ned-pcs, 2023-10-03:
@Laura-VFA you would have to add support for the second I2S channel that's used by the other codec chip that is connected to the other two microphones