respeaker/seeed-voicecard #345: [Bug]: Respeaker 2-Mic HAT on Alpine Linux

muellerj, 2024-10-09:
### Describe the bug

I tried to get the 2-mic HAT working on Alpine Linux using pointers in https://github.com/respeaker/seeed-voicecard/issues/174 and the `v6.6` branch from https://github.com/HinTak/seeed-voicecard. Most of the output looks ok, however the card is not listed properly and can't play/record. 

### Expected behavior

Working detection/interaction with `alsa`.

### Platform

- Device: Raspberry Pi Zero 2W
- OS: Linux 6.6.49-0-rpi #1-Alpine SMP PREEMPT Thu Sep  5 10:58:11 UTC 2024 aarch64 Linux
- https://github.com/HinTak/seeed-voicecard/tree/v6.6

### Relevant log output

```shell
$ dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 6.6.49-0-rpi (buildozer@build-3-20-aarch64) (gcc (Alpine 13.2.1_git20240309) 13.2.1 20240309, GNU ld (GNU Binutils) 2.42) #1-Alpine SMP PREEMPT Thu Sep  5 10:58:11 UTC 2024
[    0.000000] KASLR enabled
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi Zero 2 W Rev 1.0
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000017800000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x0000000017800000..0x000000001b7fffff (65536 KiB) map reusable linux,cma
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000001bffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001bffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000001bffffff]
[    0.000000] On node 0, zone DMA: 16384 pages in unavailable ranges
[    0.000000] percpu: Embedded 30 pages/cpu s85224 r8192 d29464 u122880
[    0.000000] pcpu-alloc: s85224 r8192 d29464 u122880 alloc=30*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] CPU features: detected: ARM erratum 843419
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_headphones=0 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:80:92:02 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  root=UUID=1d0abd4e-397d-466c-9f0b-cf23b2d54340 modules=sd-mod,usb-storage,ext4 quiet rootfstype=ext4
[    0.000000] Unknown kernel command line parameters "modules=sd-mod,usb-storage,ext4", will be passed to user space.
[    0.000000] Dentry cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 6, 262144 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 112896
[    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
[    0.000000] Memory: 353512K/458752K available (12608K kernel code, 2208K rwdata, 4252K rodata, 3456K init, 1073K bss, 39704K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: bcm2836_arm_irqchip_handle_irq
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000001] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000395] Console: colour dummy device 80x25
[    0.000412] printk: console [tty0] enabled
[    0.000517] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=76800)
[... 396 lines cut ...]
pcm.playback {
    type plug
    slave.pcm "dmixed"
}

pcm.capture {
    type plug
    slave.pcm "array"
}

pcm.dmixed {
    type dmix
    slave.pcm "hw:seeed2micvoicec"
    ipc_key 555555
}

pcm.array {
    type dsnoop
    slave {
        pcm "hw:seeed2micvoicec"
        channels 2
    }
    ipc_key 666666
}
```

HinTak, 2024-10-09:
Those sound like user / user-education issue - the 2-mic device is not capable of playback, and recording is sampled at specific rates. Anyway, there is no indication of a bug in the log - it is detected on i2c, so that confirms the device is responsive.

I am not going to comment any further, as I don't work for seeedstudio. Whatever I am willing to do, they are just provided as is, and I do not want to hear from people demanding more.

muellerj, 2024-10-10:
@HinTak: This abuse is completely uncalled for. I appreciate you not wanting to support hardware that you don't work for and I am not demanding any help. What I am demanding though, is manners in a public space.

sixtyfive, 2025-01-29:
Bro, that's thick. You opened an Issue on a repository of group A, mentioning your use of the code in the repository of person B. Possibly (I didn't check), your problem with either is PEBKAC-based. And then, when person B says they don't want to give any support, you accuse them of abusive behavior. I'd suggest you think over your position here and then rephrase your request for support a little if you want a chance at getting any.

muellerj, 2025-01-29:
Bro, you're 4 months late with your trolling attempt 😉

sixtyfive, 2025-01-29:
Maybe so. I'm not trying to troll you. I do note that no-one has responded with anything helpful (including me). Best of luck getting it to work!

jackhric, 2025-02-28:
Just got my RPi Zero 2W working with the audio output of the HAT. Don't know if this will solve your issue, but @muellerj did you set the HAT as the default ALSA output? Was running into your exact issue and that's what resolved it. Just had to add this to my /etc/asound.conf file. Set it to your card number for the HAT (looks like 0):
`defaults.pcm.card 1
defaults.ctl.card 1`

My aplay -l:
`name@rpi-zero2w:~$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: vc4hdmi [vc4-hdmi], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: seeed2micvoicec [seeed-2mic-voicecard], device 0: 3f203000.i2s-wm8960-hifi wm8960-hifi-0 [3f203000.i2s-wm8960-hifi wm8960-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0`

Here's my /etc/asound.conf:
`# The IPC key of dmix or dsnoop plugin must be unique
# If 555555 or 666666 is used by other processes, use another one

# use samplerate to resample as speexdsp resample is bad
defaults.pcm.rate_converter "samplerate"

defaults.pcm.card 1
defaults.ctl.card 1

pcm.!default {
    type asym
    playback.pcm "playback"
    capture.pcm "capture"
}

pcm.playback {
    type plug
    slave.pcm "dmixed"
}

pcm.capture {
    type plug
    slave.pcm "array"
}

pcm.dmixed {
    type dmix
    slave.pcm "hw:seeed2micvoicec"
    ipc_key 555555 
}

pcm.array {
    type dsnoop
    slave {
        pcm "hw:seeed2micvoicec"
        channels 2
    }
    ipc_key 666666
}`

muellerj, 2025-03-01:
>   jackhric  left a comment   (respeaker/seeed-voicecard#345) (https://github.com/respeaker/seeed-voicecard/issues/345#issuecomment-2689639282)

That does indeed fix my issue as well. Thanks a bunch for sharing your insight!

muellerj, 2025-03-01:
Closing this issue with reference to @jackhric s comment

closed 2025-03-01 by muellerj