respeaker/seeed-voicecard #346: [Bug]: Unable to install hw params

dihan, 2025-01-06:
### Describe the bug

Describe the bug
Install the latest Raspberry Pi 4 Os via Raspberry Pi Imager
Follow the instructions of this git and also checking out this: [wiki.seeedstudio.com/ReSpeaker_2_Mics_Pi_HAT_Raspberry](https://wiki.seeedstudio.com/ReSpeaker_2_Mics_Pi_HAT_Raspberry/)

I've reinstalled the software multiple times, but my microphone still won't work. Could the device itself be faulty? I've searched numerous support forums for similar issues, but none of the suggested fixes have helped me so far.

### Expected behavior

Expected behavior
Expecting the audio to record

### Platform

Platform
Device: Raspberry Pi 4
OS: Raspbian OS 64bit Linux raspberrypi4 6.6.51+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux
Pi Hat: ReSpeaker 2-Mics Pi HAT
Version/commit number latest version

### Relevant log output

```shell
pi4@raspberrypi4:~/seeed-voicecard $ arecord -D plughw:CARD=seeed2micvoicec,DEV=0 -r 16000 -c 1 -f S16_LE -t wav -d 5 test.wav
Recording WAVE 'test.wav' : Signed 16 bit Little Endian, Rate 16000 Hz, Mono
arecord: set_params:1416: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 16
CHANNELS: 1
RATE: 16000
PERIOD_TIME: 125000
PERIOD_SIZE: 2000
PERIOD_BYTES: 4000
PERIODS: 4
BUFFER_TIME: 500000
BUFFER_SIZE: 8000
BUFFER_BYTES: 16000
TICK_TIME: 0
```

lockpicker, 2025-01-12:
Actually, this does seem to be a bug. I have the exact same issue. 
The exact same hardware works just fine with the exact same sample rate when on your image of Buster provided in the readme link. 
But on bookworm results in the error above.

SoulAuctioneer, 2025-01-15:
Also had to downgrade from bookworm to bullseye.

dihan, 2025-01-17:
i manged to fix this using a forked repo on version 6.6 . ive created a repo with notes on how to install on a pi 4b if anyone else ran into this problem

https://github.com/dihan/pi-voice-assistant

closed 2025-01-17 by dihan

Fighter19, 2025-05-14:
@HinTak Shouldn't "plughw" resample as opposed to "hw"? To me it looks like the resampling just doesn't work.
I have a completely unrelated product here having the same problem after moving to bookworm from OpenEmbedded.

HinTak, 2025-05-14:
I think the plughw requires sime optional alsa plug-ins, as the name suggests. So it could depend on something else on your system, which may or may not be installed. Have a look at the available alsa plug-ins on your system, and try to install them. Sorry I not touched my pi for a while so can't help further or much on this.

wangminrui2022, 2025-06-15:
(base) administrator@raspberrypi:~/linux-sdk-python$ aplay -D "plughw:0,0" 123_pcm.wav
Playing WAVE '123_pcm.wav' : Signed 16 bit Little Endian, Rate 16000 Hz, Mono
aplay: set_params:1416: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 16
CHANNELS: 1
RATE: 16000
PERIOD_TIME: 125000
PERIOD_SIZE: 2000
PERIOD_BYTES: 4000
PERIODS: 4
BUFFER_TIME: 500000
BUFFER_SIZE: 8000
BUFFER_BYTES: 16000
TICK_TIME: 0
(base) administrator@raspberrypi:~/linux-sdk-python$ arecord -D "plughw:0,0" -f S16_LE -r 16000 -d 5 -t wav 123_pcm.wav
Recording WAVE '123_pcm.wav' : Signed 16 bit Little Endian, Rate 16000 Hz, Mono
arecord: set_params:1416: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 16
CHANNELS: 1
RATE: 16000
PERIOD_TIME: 125000
PERIOD_SIZE: 2000
PERIOD_BYTES: 4000
PERIODS: 4
BUFFER_TIME: 500000
BUFFER_SIZE: 8000
BUFFER_BYTES: 16000
TICK_TIME: 0

donburch888, 2025-10-06:
HinTak has updated the driver in his repository to work with the different versions of linux kernel, and it is important to **use the version for _your_ linux kernel** - unfortunately this is not even mentioned in the main installation procedure (which assumes the latest kernel version). 

The base (or kernel) of linux, which the various linux distrbutions are built on, is updated from time to time - usually with a new OS release, but sometimes as part of an update. To find your current OS details use the `uname` command.  In particular the -r option returns the kernel version, of which the first two numbers are important.  

In my case today `uname -r` returns 
```
6.12.47+rpt-rpi-v8
```
and so I will use the "6.12" branch in the "git clone" command which is given in the instructions. 

```
sudo apt install git
git clone -b v6.12 https://github.com/HinTak/seeed-voicecard
cd seeed-voicecard
sudo ./install.sh
sudo reboot
```

HinTak, 2025-10-06:
Not mentioning kernel-matching-branches, nor trying to get main/head to work with arbitrary older kernels, is really down to bitrot. Main/head did work with arbitrary older kernels via the dkms backport patching system (the bunch of patches inside, the `patches` directory) , but keeping it up and working correctly is a lot of work and time I don't have. Having kernel-matching branches is a simpler answer. (And even that, I am afraid, is hard to keep up, and hence the lack of v6.15 branch, despite 6.15 being current kernel series for a while)

donburch888, 2025-10-06:
Sorry Hintak but I am not a unix GOD so most of that flew over my head ... but I do appreciate the time and effort you have put in to keeping the reSpeaker 2-mic running. 

I used your driver 3 years ago setting up Rhasspy voice assistant satellites, and the answer at that time was manually selecting the branch.  I am now trying to update the same devices to current system and again manually selecting the 6.12 branch for Raspberry Pi OS Legacy (debian Bookworm) resolved the "set_params:1416: Unable to install hw params" error with the exact same arecord & aplay parameters from previous working configuration. 

I see there are still quite few issues reported which end up being fixed by using the matching branch for the users linux kernel ... and I suggest that the quick and easy fix is to mention it in the official documentation where most users are likely to see it. 

Hopefully that will reduce the number of issues, and let you spend more time on your current projects. Sorry to take so much of your time ... and thank you very much for keeping this hardware running.